Interbotix/interbotix_ros_manipulators #1: Installation failed due to missing ros-noetic packages

gad-imageinsight, 2020-12-08:
OS: `5.8.0-7630-generic #32~1607010078~20.04~383a644~dev-Ubuntu`
ROS Noetic

When running the xsarm_amd64_install.sh script, it failed with error:
`-- Could not find the required component 'dynamixel_sdk'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "dynamixel_sdk"
  with any of the following names:
`
Installing ros-neotic-dynamixel-sdk with apt solved this problem, but then there was a number of similar issues requiring installation of additional packages. 

N.B. if  anyone reading this runs into similar problems in future, you need to remove the interbotix_ws folder between each run of the script. 

In summary, the install script needs to also install `ros-neotic-dynamixel-sdk, ros-noetic-moveit, ros-noetic-joint-trajectory-controller, ros-noetic-moveit-visual-tools, ros-noetic-joy, ros-noetic-effort-controllers`

swiz23, 2020-12-09:
Yes, the installation script checks to see if the interbotix_ws directory already exists. If it does, it does not attempt to overwrite it. I figure some people probably edit or add files in that directory, and if they run the installation script again, they'd lose all those files...

Anyway, my guess is that you ran the installation file, then Ctrl-C'd after the workspace was already made for some reason. Then you ran the installation script again, but at that point, the installation script already assumed that all the needed dependencies have been installed - resulting in your error messages.

Besides for deleting the interbotix_ws directory if you want to redo the installation, you will want to go to the end of your ~/.bashrc file and delete duplicate environment sourcing commands as well.

All that aside, are you still having issues with the installation or any other questions?

gad-imageinsight, 2020-12-09:
Thanks for your response. The main point to my issue was in the last sentence listing the noetic components that were not automatically installed by the script. (The other business about the interbotix_ws directory was a side point I discovered in debugging the missing packages. I had to rerun the script after it halted because of the missing dependencies, and could not recover.) 

It's possible that these packages were missed because I installed Noetic before running your script, but I installed the most expansive version, so I'm not sure why that would be the case.

swiz23, 2020-12-09:
No, even if you installed ROS Noetic beforehand (desktop-full, or even just the core), the script should find and install any missing dependency packages. That's the purpose of the rosdep command in the installation script. I've tried both the RPI4 version of this script on Ubuntu Server 20.04 for the Raspberry Pi 4 and the AMD64 version of this script on Ubuntu 20.04 desktop, and did not have any issues.

Something I did notice that's different about the ROS Noetic installation procedure on the Wiki is that it doesn't tell you to install python3-rosdep, python3-rosinstall, python3-rosinstall-generator, python3-wstool, or build-essential like in the previous ROS versions (sustitute python3 with python of course). Maybe, Ubuntu has its own version of rosdep that's only somewhat compatible with ROS which is why it failed. Have you installed these tools?

Another thing I noticed is that the kernel you're using is 5.8.0-7630-generic, but I'm on 5.4.0-56-generic (seems to be the mainstream kernel currently). Don't know if that would make a difference.

Another thing could be that you haven't run the 'rosdep update' command in a while. I know for a fact that the noetic version of moveit-visual-tools wasn't available on apt until around November 23rd of this year.

gad-imageinsight, 2020-12-09:
Given that I hadn't heard of 'rosdep' until your comment above, no, I hadn't run it at all. In fact, it didn't even appear to have been installed. When I did run it, it deleted just about all the noetic packages and many related libraries. I don't know why. So I  deleted interbotix_ws again and ran your script. Which ran to completion and installed all the correct packages.  So it looks like all is good. (The only remaining question is which documentation had the reference to rosdep I missed.)

Thanks for the assistance.

closed 2020-12-09 by gad-imageinsight

swiz23, 2020-12-09:
That's just it. It doesn't mention on the Noetic install guide at http://wiki.ros.org/noetic/Installation/Ubuntu to install rosdep. However, if you go to the [rosdep ROS wiki](http://wiki.ros.org/rosdep), it'll tell you how to do it for ROS Noetic.

For reference, you can see that previous ROS wiki installation tutorials mentioned to install rosdep as shown [here](http://wiki.ros.org/melodic/Installation/Ubuntu).

ninja-hu, 2021-12-30:
Same packages were missing when installing the repo on Ubuntu 16.04 and ROS kinetic. It seems that since kinetic is EOL, rosdep skips it. To include it the following needs to be run - 
```
rosdep update --include-eol-distros
rosdep install --from-paths src --ignore-src -r -y --rosdistro kinetic
```