Interbotix/interbotix_ros_manipulators #2: Forearm-Roll Joint Limits on Wx250s

febert, 2021-01-14:
Hello,

I have a wx250s and the joint limits on the forearm_roll joint (in other words the 4th joint, or joint_angles[3]) are very restrictive. 
When the forearm_roll joint is in is neutral position (for example when grasping something from a table with a vertical gripper orientation) the forearm_roll is very close to one of its joint limits which are at +- 3.141.

Also I noticed that the initial joint position flips based on the position of the servo in the moment when I run arm_run.lauch. The neutral orientation of the joint can either correspond to -3.141 or +3.141 depending on how the joint is oriented when I run arm_run.launch.

Is there any way to relocate the joint limits and make the initial joint position the same for all initial robot confgurations?

Thanks!

I'm also posting this in the legacy repo, since it also has the same problem.

swiz23, 2021-01-14:
Hi @febert,

Can you include a picture of what you mean when say that when the forearm_roll joint is in a neutral position, it's very close to a limit? By default, the forearm_roll joint should be at 0 radians when in the position shown below...

[image]

About the size of the limits - it can't really be made any more than that without risking the cable connecting the servo to the elbow motor from becoming disconnected.

febert, 2021-01-15:
Hi @swiz23,

Thank you very much for this tip. My home configuration was flipped by 180 degrees in the forearm_roll joint resulting in this odd position close to the joint limits.

When setting the home configuration to the one you showed everything works out.

Thanks again!

swiz23, 2021-01-15:
No prob!

closed 2021-01-15 by swiz23