Interbotix/interbotix_ros_manipulators #3: Faster Inverse Kinematics

febert, 2021-01-29:
Hi,

I'm trying to use a widowx250s for teleoperation with a VR headset and it seems like the position update of `bot.arm.set_ee_pose_matrix(T_sd)` is very slow, between 0.1 and 0.6 seconds, making the motion jerky.

The reason for the slowness seems to be that the inverse kinematics used by the modern robotics repo are numerical and use a numpy matrix inverse which can be very slow.

Has anyone implemented faster inverse kinematics? 

Thanks!

swiz23, 2021-02-01:
Hi @febert,

You mentioned you're doing teleoperation. It's quite possible that you're seeing lag from the network (I assume your Python script is on one computer and your arm is connected to another computer).

Unless you're saying that you are measuring the time between consecutive calls of the set_ee_pose_matrix command, and it's giving you between 0.1 and 0.6 seconds?

The truth is that the modern_robotics library is not optimized for time. Even so, I've never seen it take that long to calculate IK. For example, the joystick control package does IK in real time, and it seems to run pretty well even at 40Hz.

Some things to think about...
-  one of the arguments in the set_ee_pose_matrix function is 'moving_time' and 'blocking'. By default, the moving_time is 2 seconds, and the blocking parameter is set to True. This means that the function won't return until after 2 seconds (or whatever value you set moving_time to be). If you want to do more real time control stuff, I'd recommend changing moving_time to be 0.2 seconds, accel_time to be 0.1 seconds, and set 'blocking' to False.
- Another thing you can do is provide a close initial guess. For realtime control, a good guess would be the previous IK output - this should help the algorithm converge faster. See the function arguments [here](https://github.com/Interbotix/interbotix_ros_toolboxes/blob/main/interbotix_xs_toolbox/interbotix_xs_modules/src/interbotix_xs_modules/arm.py#L164).

febert, 2021-02-02:
The problem seems to be with the thread I use to call set_ee_pose_matrix. I'm using a `rospy.Timer(rospy.Duration(0.02), func) ` function to call set_ee_pose_matrix, which takes between 0.1 and 0.6 seconds. When I call the set_ee_pose_matrix in a simple test script outside of my framework it only takes 0.02 seconds. 

It might be related to the global interpreter lock of python, or some other threading-related problem.

I'll try to restructure my code and see if that solves it.

swiz23, 2021-02-02:
Sounds good. If you're using a local threading.Lock() mutex in your `func` callback, that could be holding you up as well.

swiz23, 2021-02-09:
Closing due to lack of activity and since the issue is not related to the IK function. Feel free to reopen if you have any other questions!

closed 2021-02-09 by swiz23