Interbotix/interbotix_ros_manipulators #4: rviz pose does not match with robot pose (initial install)

seth814, 2021-02-07:
I got through installation easily enough.
ran `roslaunch interbotix_xsarm_control xsarm_control.launch robot_model:=rx200`
Is there a setting I need to change or did someone not test stuff? :thinking: 

Rviz screenshot

[image]

Reactor 200

[image]

I tried running the example scripts, and well... It proceeded to crash into things.

swiz23, 2021-02-08:
Ya,

The second thing you said - 'someone did not test stuff' :/

It looks like the shoulder motor ids need to be switched. In the picture you attached above, the shoulder motor that is visible should be ID'd Number 3. The shoulder motor on the other side (not viewable) should be ID'd Number 2. You can easily switch IDs using the Dynamixel Wizard 2.0 tool. See more at https://github.com/Interbotix/interbotix_ros_core/blob/main/interbotix_ros_xseries/interbotix_xs_sdk/TROUBLESHOOTING.md#dynamixel-wizard-20-tool.

seth814, 2021-02-09:
Switched the ids on the motors for 2 and 3. We're back in business!

[image]

closed 2021-02-09 by seth814