Interbotix/interbotix_ros_manipulators #8: 403 Forbidden Error

rkim87, 2021-05-27:
Hi,

I've been attempting to install the ros packages but have ran into cmake errors with the realsense packages. The following is what I get when I try to update the repositories:

`Err:6 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease`
`403  Forbidden [IP: 52.218.112.138 80]`
`Hit:7 https://librealsense.intel.com/Debian/apt-repo bionic InRelease`
`Err:8 https://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease`
`403  Forbidden [IP: 52.218.112.138 443]`
`Err:9 https://homerepo.oit.gatech.edu/saltstack/py3/ubuntu/18.04/amd64/3001 bionic InRelease`
`Could not connect to homerepo.oit.gatech.edu:443 (10.138.12.138), connection timed out`

rkim87, 2021-05-27:
The repository link for the "librealsense" library should be updated in xsarm_amd64_install.sh!
```diff
- deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo focal main

+ deb https://librealsense.intel.com/Debian/apt-repo focal 
```

swiz23, 2021-05-27:
Huh. I saw that Intel changed their server to the librealsense.intel.com one a couple months back, but the old one was still working too. I guess they finally took it down resulting in those errors. I'll get that fixed up soon.

swiz23, 2021-05-27:
Fixed

closed 2021-05-27 by swiz23

swiz23, 2021-06-14:
For anyone who is seeing this type of error, the way to fix it is as follows:
1. Navigate to the `/etc/apt/` directory and open the `sources.list` file by typing `sudo gedit sources.list`.
2. Find the two lines that say  `deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo focal main` (or similar) and delete them.
3. Save and close the file
4. In a terminal, type `sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u`
5. Now type `sudo apt update` and you should no longer see the error.