Interbotix/interbotix_ros_manipulators #9: CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):

rkim87, 2021-06-02:
Hi,

I've been trying to build the realsense_ws and the interbotix_ws and have been encountering cmake errors.

When trying to catkin_make the realsense_ws, this is the following error I get:

```
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "ddynamic_reconfigure" with any of the following names:

    ddynamic_reconfigureConfig.cmake
    ddynamic_reconfigure-config.cmake

  Add the installation prefix of "ddynamic_reconfigure" to CMAKE_PREFIX_PATH
  or set "ddynamic_reconfigure_DIR" to a directory containing one of the
  above files.  If "ddynamic_reconfigure" provides a separate development
  package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:11 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/rkim87/realsense_ws/build_isolated/realsense2_camera/CMakeFiles/CMakeOutput.log".
See also "/home/rkim87/realsense_ws/build_isolated/realsense2_camera/CMakeFiles/CMakeError.log".
<== Failed to process package 'realsense2_camera': 
  Command '['cmake', '/home/rkim87/realsense_ws/src/realsense-ros/realsense2_camera', '-DCATKIN_DEVEL_PREFIX=/home/rkim87/realsense_ws/devel_isolated/realsense2_camera', '-DCMAKE_INSTALL_PREFIX=/home/rkim87/realsense_ws/install_isolated', '-G', 'Unix Makefiles']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/rkim87/realsense_ws/build_isolated/realsense2_camera && cmake /home/rkim87/realsense_ws/src/realsense-ros/realsense2_camera -DCATKIN_DEVEL_PREFIX=/home/rkim87/realsense_ws/devel_isolated/realsense2_camera -DCMAKE_INSTALL_PREFIX=/home/rkim87/realsense_ws/install_isolated -G 'Unix Makefiles'

```

Running catkin_make on interbotix_ws, I get the following error:

```
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "realsense2_camera"
  with any of the following names:

    realsense2_cameraConfig.cmake
    realsense2_camera-config.cmake

  Add the installation prefix of "realsense2_camera" to CMAKE_PREFIX_PATH or
  set "realsense2_camera_DIR" to a directory containing one of the above
  files.  If "realsense2_camera" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  interbotix_ros_manipulators/interbotix_ros_xsarms/interbotix_xsarm_perception/CMakeLists.txt:10 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/rkim87/interbotix_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/rkim87/interbotix_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

```

Probably because I can't build the realsense_ws...

Any help?

swiz23, 2021-06-02:
If you'd like to use a realsense camera, I'd recommend answering 'y' or 'yes' to the question about installing the perception pipeline when running the installation script. It will install all dependency packages and catkin_make everything.

Since you've already run the installation script (I assume you did anyway), you should delete the interbotix_ws, the apriltag_ws, and the realsense_ws directories. Then open up the ~/.bashrc file and delete the following lines

```
source /home/softeng/realsense_ws/devel/setup.bash
source /home/softeng/apriltag_ws/devel_isolated/setup.bash
source /home/softeng/interbotix_ws/devel/setup.bash
export ROS_IP=$(echo `hostname -I | cut -d" " -f1`)
if [ -z "$ROS_IP" ]; then
	export ROS_IP=127.0.0.1
fi
```
Of course yours won't look exactly the same, but it will be similar. At this point, close and open a new terminal and rerun the install script.

OR...

just open a terminal and type _sudo apt install ros-melodic-ddynamic-reconfigure_. That should solve the dependency issue and you should be able to catkin_make successfully.

rkim87, 2021-06-02:
Hi,

Thanks for the feedback. I had to manually install the dependencies to fix the issue. I tried installing on a different laptop and didn't encounter this issue, which was odd.

However, on both laptops that I installed the packages, I was no longer able to access any camera under '/dev'. In fact, the '/dev/video*' option completely seems to be missing.

I tried updating/upgrading the repos just in case, and encountered the following issue:

```
rkim87@ME02L0464GRD09L:~$ sudo apt upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  librealsense2 librealsense2-dbg librealsense2-dev librealsense2-gl
  librealsense2-gl-dbg librealsense2-gl-dev librealsense2-net
  librealsense2-net-dbg librealsense2-net-dev librealsense2-udev-rules
  librealsense2-utils
0 upgraded, 0 newly installed, 0 to remove and 11 not upgraded.
```

I tried manually installing the dependencies and the following occurred:

```
rkim87@ME02L0464GRD09L:~$ sudo apt install librealsense2
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 librealsense2-dbg : Depends: librealsense2 (= 2.40.0-0~realsense0.3814) but 2.45.0-0~realsense0.4551 is to be installed
 librealsense2-dev : Depends: librealsense2 (= 2.40.0-0~realsense0.3814) but 2.45.0-0~realsense0.4551 is to be installed
 librealsense2-gl : Depends: librealsense2 (= 2.40.0-0~realsense0.3814) but 2.45.0-0~realsense0.4551 is to be installed
 librealsense2-net : Depends: librealsense2 (= 2.40.0-0~realsense0.3814) but 2.45.0-0~realsense0.4551 is to be installed
 librealsense2-utils : Depends: librealsense2 (= 2.40.0-0~realsense0.3814) but 2.45.0-0~realsense0.4551 is to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.

```

Do you happen to know what I can do to go about fixing this?

swiz23, 2021-06-02:
In the install script, I've purposely put a hold on the librealsense2 packages to prevent them from upgrading. This is because each version of librealsense2 corresponds to a specific version of realsense_ros. So if librealsense2 was allowed to update, then realsense_ros would complain that it cannot find the correct librealsense2 version. and stuff wouldn't work.

However, if you really want to upgrade, you can take off the hold by typing `sudo apt-mark unhold librealsense2*` in a terminal. Then you'll be able to install the latest version of librealsense. However, you'll need to update realsense_ros as well.

rkim87, 2021-06-02:
Do you happen to know how to resolve the issue of /dev/video going missing? I'm trying to utilize various usb cameras other than the realsense as well but am struggling to detect it.

Are you seeing a similar issue on your end?

swiz23, 2021-06-03:
Not sure. I have video0 to video9 showing up on my laptop just fine in the /dev directory.   If it helps at all, typing _rs-enumate-devices -S_ in a terminal with your realsense camera plugged in should show you a blurb with port info about the camera.

Other than that, my experience with usb cameras has pretty much been with the microsoft lifecam. To make sure my computer found it, i added the following rule to my udev rules.

`SUBSYSTEM=="video4linux", ATTRS{idVendor}=="045e", ATTRS{idProduct}=="0812", ATTR{index}=="0", SYMLINK+="lifecam"`

rkim87, 2021-06-03:
I'm wondering if the install file overwrites the udev rules and that might be causing the issue...

Does the install file overwrite udev rules? If so, do you know a way to add udev rules without overwriting the default one?

swiz23, 2021-06-03:
the only udev rule that i configure is 99-interbotix-udev.rules which goes into the /etc/udev/rules.d directory. Now, because i preface the filename with '99', my file takes precedence over udev rules with lower numbers. so it's possible that something in that udev rule is 'overwriting' a rule in a 'lower precedent' one.

rkim87, 2021-06-03:
I've tried changing the file to a lower number and even removing the udev rule and still no results :cry:

swiz23, 2021-06-03:
that is strange. i found an article at https://forums.developer.nvidia.com/t/usb-web-cam-does-not-show-up-in-dev/53155 where they seem to say it could be a driver related issue. 

That aside, I'm curious what would happen if you uninstalled all realsense related packages. Part of the realsense install process is to patch the linux kernel and maybe that affects things? See more at https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#uninstalling-the-packages

rkim87, 2021-06-03:
Purging all realsense related packages brought back the /dev/video* directories...Does this mean that I wouldn't be able to use realsense with my laptops?

swiz23, 2021-06-03:
Not sure - I have realsense installed on my laptop - yet i can still see the /dev/video ports. I was just about to recommend you post about this at librealsense's Github, then saw you already did, lol.

Making a link here to that issue so that others can follow it - https://github.com/IntelRealSense/librealsense/issues/9147

rkim87, 2021-06-03:
Thanks for all the help @swiz23 , I'll update this feed once I make any progress.

rkim87, 2021-06-03:
Booting to previous version of kernel fixed this issue! 😄 Thanks so much as always.

Best,

closed 2021-06-03 by rkim87