Interbotix/interbotix_ros_manipulators #10: Motors Overheating on WX250S

febert, 2021-06-04:
Hi,

we're using two WX250s robots and for both we're seeing issues of motors loosing power and blinking red, while getting quite warm after about 2 hours of continuous use.
 
This seems to be an overloading issue, but we're not applying any excessive force at the end-effector.
Here's an example of the motions we're performing:
https://photos.google.com/search/_tra_/photo/AF1QipP2wLZU0wOHl6L3m2jyELKghYH5LwEwk5Fva47P
We're using the standard position control mode at around 5Hz.

The errors appear most often for motor 2 and 5.

I tried sticking a few heat sinks on the motors of the "shoulder joint" but it didn't fully resolve the problem.

Thanks a lot for the help!

swiz23, 2021-06-04:
Did you use the Dynamixel Wizard 2.0 tool to determine if the issue was overheating or overloading?

Also - the link doesn't work - can you upload the photos directly?

If you're extending the arm past 70% of its max reach for long periods of time, that could cause the issue you are seeing. Alternatively, if you set a higher Position_P_Gain than the default, or set a non-zero Position_I_Gain, that could cause the issue as well.

Another thing that could be happening is internal 'fighting' for those two motors. As you know, motor 2 is one of two motors that drives the shoulder joint. Similarly, motor 5 is one of two motors that drives the elbow joint.

Essentially, the issue that can arise is that the encoder on each motor in a dual-joint can show slightly different positions despite having their horns physically aligned during robot assembly. As a result, if a goal position is commanded to that joint, one motor might reach the goal position and stop moving. However, the other motor's encoder might show that the goal position hasn't been reached yet and exert more effort to reach it. As a result, the first motor will exert more effort to keep holding its position. This leads to a vicious cycle that can ultimately lead to one or both motors overloading and torquing off. To prevent that from happening, a calibration step is done at the very beginning of the ROS node starting. It sets the Homing Offset register of the second motor in a dual joint setup so that the second motor's encoder shows the same value as the first motor's encoder.

Ideally, this should prevent the motors from heating up too much - but you may be able to get better results by adjusting the Homing Offset register yourself for the second motor in each dual joint using the Dynamixel Wizard 2.0 tool. Manually move the joint to various positions and hold it there. Then see what the Present Position registers say for both motors in a dual joint. If the second motor (the one with the higher ID) is still off, tune its Homing Offset register.

After tuning, head over to motor config file at https://github.com/Interbotix/interbotix_ros_manipulators/blob/834edcbe6b955497d10ebc20ea9242c3f61e8fd1/interbotix_ros_xsarms/interbotix_xsarm_control/config/wx250s.yaml#L21. Then set the 'calibrate' parameter for both the shoulder and elbow joints to 'false'. This ensures that the tuning you did via the Dynamixel Wizard 2.0 tool will not be overwritten.

febert, 2021-06-04:
Yes, we are using the robot at a decent amount of extension:

[image]

The position registers for motors 2 and 3 differ by about 0.2 degress, but the homing offset of motor 3 is set to 0.26 (which should compensate for that).

The positions for 4 and 5 differ between 0 and 0.3 degrees, the homing offset is set to 0.79.

These mismatches seem quite low, do you think they could be causing any of the overheats?

swiz23, 2021-06-06:
The Dynamixel Wizard 2.0 tool is not used with our ROS packages. It's a standalone debugging tool created by the motor manufacturers (Robotis). See how to use the tool [here](https://github.com/Interbotix/interbotix_ros_core/blob/main/interbotix_ros_xseries/interbotix_xs_sdk/TROUBLESHOOTING.md#dynamixel-wizard-20-tool). 

You'll specifically want to use the tool *after* the two motors you specified go into their error states while having the robot attempt doing your application. It's very important not to power cycle the arm after the motors go into their error states. Instead, shut down the ros packages (you should physically hold the arm while doing this as the motors will torque off when you shutdown). Then feel free to lower the arm so you don't have to hold it. Then start up the dynamixel wizard 2.0 tool. Select the motors that errored out. Look for the [Hardware Error Status](https://emanual.robotis.com/docs/en/dxl/x/xm430-w350/#hardware-error-status) register to see what the cause of the error is. You can also look at the Homing Offset register to get a better understanding of how that works.

Let me know what you find!

febert, 2021-06-07:
Here are screenshots right after motor 2 and 3 errored out:
[image]
[image]

The page for motor 2 looks like this:
[image]

swiz23, 2021-06-08:
So the fact that 'OL' is illuminated indicates that the motor experienced an overload error as opposed to overheating. The temperatures seem to be well below the max limit, and the difference in position also seems to be negligible.

Here are some options...

1) Verify that all Dynamixel Servos are up-to-date. Using the Wizard 2.0 tool, look on the left column where the motors are listed. If any of them have an asterisk by their name, the motor can be updated. See how to do that at https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/#firmware-update. Sometimes, updating the firmware will fix these kind of issues.

2) Try manually tuning the Homing Offset of ID:3 until its Present Position register matches the Present Position register of ID:2. Next, tune the Homing Offset of ID:5 until its Present Position register matches the Present Position register of ID:4. After doing this, make sure to set the ['calibrate' tag](https://github.com/Interbotix/interbotix_ros_manipulators/blob/a191e50611d53841e2f9b20642aff48cf6c0beaa/interbotix_ros_xsarms/interbotix_xsarm_control/config/wx250s.yaml#L21) for both the 'shoulder' and 'elbow' joints to false.

3) Move the arm closer to the application

4) Let the arm rest for a minute every hour and a half

Try doing 1 - 3 first and see if you get better results.

swiz23, 2021-06-17:
Closing due to lack of activity. Feel free to reopen if you still have questions.

closed 2021-06-17 by swiz23

tan-may16, 2023-12-04:
Hello, I am facing the same exact issue. I tried all the solutions but my joint motor experiences overload after some time. What could be done? Thank you.

abadithela, 2025-01-04:
Hi, I too am facing the same issue, but within the first trial or two, not after significant time (like 2 hours). 
After the error, we power cycled the motors based on the trouble shooting guidelines here: https://docs.trossenrobotics.com/interbotix_xsarms_docs/troubleshooting.html#common-issues

Would you recommend not doing this and killing the ros nodes?

I've tried diagnosing with the Dynamixel GUI, but when I run scan, it does not return anything.