Interbotix/interbotix_ros_manipulators #11: px150 failing to run interbotix_xsarm_diagnostic_tool on Ubuntu with Noetic

gib-gif, 2021-06-10:
I am very new to using ROS packages. I downloaded all the packages in the beginning and am able to run moveIt and whatnot, yet the diagnostic tool will not run and instead prompts the following error code:

"No such file or directory: /home/fionn/interbotix_ws/src/interbotix_ros_manipulators/interbotix_ros_xsarms/interbotix_xsarm_descriptions/urdf/.urdf.xacro [Errno 2] No such file or directory: '/home/fionn/interbotix_ws/src/interbotix_ros_manipulators/interbotix_ros_xsarms/interbotix_xsarm_descriptions/urdf/.urdf.xacro'
RLException: while processing /home/fionn/interbotix_ws/src/interbotix_ros_manipulators/interbotix_ros_xsarms/interbotix_xsarm_control/launch/xsarm_control.launch:
while processing /home/fionn/interbotix_ws/src/interbotix_ros_manipulators/interbotix_ros_xsarms/interbotix_xsarm_descriptions/launch/xsarm_description.launch:
Invalid <param> tag: Cannot load command parameter [/robot_description]: command [['xacro', '/home/fionn/interbotix_ws/src/interbotix_ros_manipulators/interbotix_ros_xsarms/interbotix_xsarm_descriptions/urdf/.urdf.xacro', 'robot_name:=', 'base_link_frame:=base_link', 'show_ar_tag:=false', 'show_gripper_bar:=true', 'show_gripper_fingers:=true', 'use_world_frame:=true', 'external_urdf_loc:=', 'load_gazebo_configs:=false']] returned with code [2]. 

Param xml is <param name="$(arg robot_name)/robot_description" command="xacro $(arg model)"/>
The traceback for the exception was written to the log file"

I checked through my files and I have down to urdf, but not .urdf.xacro. itself. Instead urdf contains px150.urdf.xacro. I tried copying and renaming the px150.urdf.xacro to urdf.xacro and while that solved the first part of the issue, the second part (processing the control launch) persisted.
Any help at all would be greatly appreciated

P.s I am new to posted here so apologies if this is outside the guidelines.

swiz23, 2021-06-10:
Hi @gib-gif,

It looks like you didn't set the 'robot_model' argument when you tried to run the package. For example, to run the interbotix_xsarm_control package, you would type...

```
roslaunch interbotix_xsarm_control xsarm_control.launch robot_model:=px150
```

Launching the diagnostic tool package follows a similar structure as well.

gib-gif, 2021-06-11:
That solved it. Thanks for you help.

closed 2021-06-11 by gib-gif