Interbotix/interbotix_ros_manipulators #12: Adjusting the interbotix_xsarm_diagnostic_tool

gib-gif, 2021-06-17:
I am looking to run a my own python movement script and retrieve data, from the joints of a px150, like temperature or current. I am wondering is there a way to use the diagnostic tool but instead of the default one joint rotating over and back motion in the package, I would like to include a python script with my own movements? Is this possible or will I have to write a script or package from scratch?
Any help is greatly appreciated, thanks.

LSinterbotix, 2021-06-18:
Hi @gib-gif! Unfortunately, there is no way to do what you're asking for given how the diagnostic tool is designed. Currently, the package takes a single parameter-specified joint through a sinusoidal trajectory while observing its position, velocity, effort, and temperature values.

If you want a purely observational diagnostic tool, you would have to make your own script. You are welcome to write one yourself and submit it as a PR. Using [xsarm_diagnostic_tool.cpp](https://github.com/Interbotix/interbotix_ros_manipulators/blob/a191e50611d53841e2f9b20642aff48cf6c0beaa/interbotix_ros_xsarms/examples/interbotix_xsarm_diagnostic_tool/src/xsarm_diagnostic_tool.cpp) as a reference, you could have something that listens to each joint state and temperature while your Python script is running.

Let us know if you have any other questions or require further guidance.

gib-gif, 2021-06-22:
Ok, I think I'll look into writing it from scratch so. Thanks 
for the help

LSinterbotix, 2021-06-22:
Great. Like I said above, if you have any questions regarding the capabilities of the current package, let us know. Otherwise, this issue is closed.

closed 2021-06-22 by LSinterbotix

gib-gif, 2021-06-30:
I am reading through the diagnostic_tool_node.cpp and am looking for the topic to subscribe to get back the temperature value. While I understand some C++, I am having trouble parsing this file. I believe that this line (line 24) subscribes to the topic I want:

  ros::Publisher pub_temperatures = n.advertise<interbotix_diagnostic_tool::MotorTemps>("joint_temperatures", 100); 

Am I correct in this? If so could someone explain to me how this line works and what the topic is? 
I tried looking through my own ROS system and while I could find the ROS msg used (interbotix_xsarm_diagnostic_tool/JointTemps),  I couldn't find the topic behind it.

LSinterbotix, 2021-06-30:
The line you reference creates a publisher that sends out messages of type `interbotix_diagnostic_tool::MotorTemps` to the topic `joint_temperatures`. Translated to Python, this would be: 
```Python
pub_temperatures = rospy.Publisher("joint_temperatures", interbotix_diagnostic_tool.MotorTemps, queue_size=100)
```

In the current iteration, we create publishers for group and single joints.
https://github.com/Interbotix/interbotix_ros_manipulators/blob/21683026ed7069cbb7aa6001e808a10f0f2d7670/interbotix_ros_xsarms/examples/interbotix_xsarm_diagnostic_tool/src/xsarm_diagnostic_tool.cpp#L29-L30

We also create the service client `srv_get_reg` to act on the service `get_motor_registers`. This service reads data from the motor registers to get information like current motor temperature.
https://github.com/Interbotix/interbotix_ros_manipulators/blob/21683026ed7069cbb7aa6001e808a10f0f2d7670/interbotix_ros_xsarms/examples/interbotix_xsarm_diagnostic_tool/src/xsarm_diagnostic_tool.cpp#L34

The code below is responsible for reading temperature values, specifically line 120. You can see that line 117 calls the service request and lines 123 and 124 publish the joint and group temperatures.
https://github.com/Interbotix/interbotix_ros_manipulators/blob/21683026ed7069cbb7aa6001e808a10f0f2d7670/interbotix_ros_xsarms/examples/interbotix_xsarm_diagnostic_tool/src/xsarm_diagnostic_tool.cpp#L109-L125

If you wanted to write your own tool, you would use these sections for listening to and publishing joint and group temperatures.

gib-gif, 2021-07-02:
Thanks for the breakdown. I am trying to create a node script to run these but am having issues with importing the messages and services. For example in the package.xml for the script  I added in all the depend lines for interbotix_xsarm_diagnostic_tool and at the top of the node script I have:

from interbotix_xsarm_diagnostic_tool import JointTemps

However when I try to run the node, I get the following error message:

ImportError: cannot import name 'JointTemps' from 'interbotix_xsarm_diagnostic_tool' (/home/user/interbotix_ws/devel/lib/python3/dist-packages/interbotix_xsarm_diagnostic_tool/__init__.py)

I am a bit lost with this as I definitely have the interbotix_xsarm_diagnostic_tool package installed and other imports like JointState from sensor_msgs.msg work fine. I tried following the address given in the error above and found _JointTemps.py there. 
Also, apologies for how long this thread is getting, I swear I'm not trying to just offload all my issues, I'm just quite lost.

LSinterbotix, 2021-07-02:
You would need to import the `JointTemps` message class from the `interbotix_xsarm_diagnostic_tool.msg` directory so your import statement should be:
```Python
from interbotix_xsarm_diagnostic_tool.msg import JointTemps
```

gib-gif, 2021-07-04:
Ok, thanks again

gib-gif, 2021-07-19:
First off, I want to say thanks. I got the temperature data back as I wanted thanks to your help and the example code. I do have two more questions that I can't seem to find answers to online and if you could help that would be great.

The first isn't a major issue, mainly my own curiousity. I am only using the /px150/get_motor_registers service. 

 `#!/usr/bin/env python3
import rospy
from interbotix_xs_sdk.srv import RegisterValues

def main():
    rospy.init_node('temp_monitor')
        
    rospy.wait_for_service('/px150/get_motor_registers')
    
    srv_get_reg = rospy.ServiceProxy('/px150/get_motor_registers',RegisterValues)
    
    rate = rospy.Rate(1.0)
    
    while not rospy.is_shutdown():
    
        #get_reg_msg = RegisterValues()
        #get_reg_msg.cmd_type = 'group'
        #get_reg_msg.name = 'arm'
        #get_reg_msg.reg = 'Present_Temperature'
        #get_reg_msg.value = None
    
        #srv_get_reg(get_reg_msg)
        get_reg_msg_response = srv_get_reg('group', 'arm', 'Present_Temperature', None)
        rospy.loginfo('temp = {}'.format(get_reg_msg_response))
      
        rate.sleep()    

    if __name__ == '__main__':
        main()`

In the diagnostic tool example code the get_reg_msg object is created and each attribute is filled out over three lines before the service is called, to fill the response attribute. I did the same above in the lines I have commented out in the while loop. However this won't run and instead I get the following error:

TypeError: Invalid number of arguments, args should be ['cmd_type', 'name', 'reg', 'value'] args are(<interbotix_xs_sdk.srv._RegisterValues.RegisterValues object at 0x7faffbfecd60>,)

Is this just a difference in the way c++ and python are designed or have I missed something? Again its not a huge issue as the problem can be solved by creating and calling the response in one line.

My second question is can the get_motor_registers service be used for obtaining other data beyond temperature, such as current or voltage in the motors? I can't seem to find any info about the service online and my own messing with the code hasn't turned anything up.

LSinterbotix, 2021-07-19:
1)
You're trying to define the request as the generic service object. You instead have to define it as a Request, i.e. a `RegisterValuesRequest` object. I have modified the script below. Note the additional import on line 3 and the change in object
type in line 16.
```Python
#!/usr/bin/env python3
import rospy
from interbotix_xs_sdk.srv import RegisterValues, RegisterValuesRequest

def main():
    rospy.init_node('temp_monitor')

    rospy.wait_for_service('/px150/get_motor_registers')

    srv_get_reg = rospy.ServiceProxy('/px150/get_motor_registers', RegisterValues)

    rate = rospy.Rate(1.0)

    while not rospy.is_shutdown():

        get_reg_msg = RegisterValuesRequest()
        get_reg_msg.cmd_type = 'group'
        get_reg_msg.name = 'arm'
        get_reg_msg.reg = 'Present_Temperature'
        get_reg_msg.value = 0

        print(srv_get_reg(get_reg_msg))
    
        rate.sleep()    

if __name__ == '__main__':
    main()
```

2)
You can get any set register value from the list [at this link](https://github.com/Interbotix/interbotix_ros_core/blob/a3fe00ee42f059df4f3f59722ecc23debbe8d0da/interbotix_ros_xseries/dynamixel_workbench_toolbox/src/dynamixel_workbench_toolbox/dynamixel_item.cpp#L27-L116). 

Yes, the actual value for the input voltage would actually be 12.2 since the request gives the voltage in units of 0.1V as documented in the [Robotis e-Manual](https://emanual.robotis.com/docs/en/dxl/x/xl430-w250/#control-table-of-ram-area).

gib-gif, 2021-07-21:
Yes, that solved it. Thanks! 
Finally, this may not be exactly the right place for this question, but in the diagnostic tool effort is also recorded. The unit given to effort in the tool is mA. Though effort comes from the joint_states topic, and the JointState message documentation lists effort as Nm [here](http://docs.ros.org/en/melodic/api/sensor_msgs/html/msg/JointState.html). Could you explain this difference?

LSinterbotix, 2021-07-21:
Effort can be interpreted as the torque applied to the joint or the current supplied to the motor since current and torque generally have a proportional relationship. You can see examples of effort reported as current in the Universal Robots driver repo [here](https://github.com/ros-industrial/ur_modern_driver/blob/fe9762e9d52b20bf6d1640bd8d046db27b0774d1/src/ur_ros_wrapper.cpp#L676) or the Husarion panther driver repo [here](https://github.com/husarion/panther_driver#encoders-motor-speed-motor-current).