Interbotix/interbotix_ros_manipulators #13: Launch file shutdown on motor error

Jendker, 2021-06-29:
Is there any way to tell the control node to shutdown or to at least signal the malfunction if any of the motors fails during operation e.g. due to overload error? Currently if we are running the robot and a motor is shut down the arm continues operation using IK but one of the motors does not react to the commands and it is difficult to figure out when the failure occurred.

swiz23, 2021-06-30:
Hi @Jendker,

Are you using the Interbotix Python API or working with your own custom ROS node?

LSinterbotix, 2021-06-30:
To answer your side question, programmatically rebooting motors is possible. See the code below from the [interbotix_xs_sdk](https://github.com/Interbotix/interbotix_ros_core/tree/main/interbotix_ros_xseries/interbotix_xs_sdk) for a ROS implementation using services:

Advertising reboot service:
https://github.com/Interbotix/interbotix_ros_core/blob/a3fe00ee42f059df4f3f59722ecc23debbe8d0da/interbotix_ros_xseries/interbotix_xs_sdk/src/xs_sdk_obj.cpp#L742

Reboot service:
https://github.com/Interbotix/interbotix_ros_core/blob/a3fe00ee42f059df4f3f59722ecc23debbe8d0da/interbotix_ros_xseries/interbotix_xs_sdk/src/xs_sdk_obj.cpp#L838-L846

Reboot service definition:
https://github.com/Interbotix/interbotix_ros_core/blob/a3fe00ee42f059df4f3f59722ecc23debbe8d0da/interbotix_ros_xseries/interbotix_xs_sdk/src/xs_sdk_obj.cpp#L144-L187

Jendker, 2021-06-30:
Thank you for the answers! @swiz23 we are using the Python API (`interbotix_xs_modules`).

Jendker, 2021-07-02:
I went over the file referenced by @LSinterbotix but I was not able to find any service which would give me a feedback about the servo errors. Is there anything I could use for it? Maybe the Python API provides such functionality? That would be optimal.

swiz23, 2021-07-04:
Hi @Jendker,

So there's a weird bug with the dynamixel_workebench_toolbox API created by Robotis (the motor manufacturers) which makes it impossible to successfully read any register from a motor in an error state. See more at https://github.com/ROBOTIS-GIT/dynamixel-workbench/issues/307. So if you want to know if a motor is probably in an error state, this is what you should do...

Create a thread that will do the following every second or so...
1. Call the [robot_get_motor_registers](https://github.com/Interbotix/interbotix_ros_toolboxes/blob/2fabcbf4b9f48ac917ef9ce4985419c0c4316763/interbotix_xs_toolbox/interbotix_xs_modules/src/interbotix_xs_modules/core.py#L84) function with cmd_type set to 'group', name set to 'all', and reg set to 'Hardware_Error_Status". Assuming your robot instance is called 'bot', this will look like `values = bot.dxl.robot_get_motor_registers("group", "all", "Hardware_Error_Status")`. 
2. Now, 'values' will be a list of what the current error state is for each motor in the 'all' group. As a refresher, the 'all' group includes every single motor defined under the [motors](https://github.com/Interbotix/interbotix_ros_manipulators/blob/21683026ed7069cbb7aa6001e808a10f0f2d7670/interbotix_ros_xsarms/interbotix_xsarm_control/config/wx250s.yaml#L31) param in the motor config file (I arbitrarily chose the wx250s one). Assuming we are working with the WX250 six dof, the 'all' group includes 9 motors. If no motor is in an error state, then 'values' should be a list of 9 zeros. Ideally, if a motor is in an error state, it should return a non-zero number. So theoretically, you should be able to look through 'values' to see if any are non-zero and if it is, reboot that motor. However, because of the bug, the ROS service responsible for reading the motor registers will just return early. So instead of returning a list of 9 values, you'll get a list of some number less than that - like a list of 8 zeros or 7 zeros, etc...
3. Long story short - you can tell if a motor is in an error state if the length of 'values' is less than 9. If so, you should call, `bot.dxl.robot_reboot_motors("group", "all", enable=True, smart_reboot=True)`. This will call a reboot ros service which will only reboot those motors that it tries to read from, but get's an error message instead. Alternatively, you could call the reboot function with the name of the motor that failed - for example, if the length of 'values' is 5, then you know there was an issue with the sixth motor in the 'all' group.
4. Now, when a reboot is performed, all the [RAM area registers](https://emanual.robotis.com/docs/en/dxl/x/xm430-w350/#control-table-of-ram-area) revert back to their default values. So you'll need to also change those back to what they should be. So if you customized any PID gains or the profile_velocity or profile_acceleration registers, you'll need to set those again.

At the end of the day - this process is a bit of a pain. Once and 'if' Robotis fixes this bug, then this process could become a whole lot simpler by having the reboot function defined at https://github.com/Interbotix/interbotix_ros_core/blob/a3fe00ee42f059df4f3f59722ecc23debbe8d0da/interbotix_ros_xseries/interbotix_xs_sdk/src/xs_sdk_obj.cpp#L149 to read all ram area registers before a reboot, and then write them back afterwards.

Jendker, 2021-07-04:
Thank you for the answer @swiz23 . Does 'interbotix_xsarm_control/launch/xsarm_control.launch' use the default values? We are not changing any registers after launching the roslaunch file. I think we should be fine then with the defaults?

swiz23, 2021-07-05:
The launch file uses the default values for all registers except for Profile_Velocity (default is 0 but the launch file sets to 2000), and Profile_Acceleration (default is 0, but the launch file sets to 300). The units of these registers are in milliseconds.

However, if you change the 'moving_time' or 'accel_time' parameters in your python script, then those values will change to be what you commanded them to be. So if you commanded a moving_time of 1.5 seconds, then profile_velocity will be changed to 1500 for all the arm motors).

Additionally, when using the set_ee_cartesian_trajectory funciton, those values will change again to the millisecond equivalent of the moving_time and accel_time parameters defined in that function (200 and 100 respectively).

So depending on where a motor fails, the profile_velocity and profile_acceleration registers might have different values.

Personally, I'd recommend that whenever an error occurs, you catch it with the method mentioned in my previous comment, immediately command the arm to go to its current joint states and exit the Python script.

Jendker, 2021-07-05:
Thank you, I will follow the recommendation!

closed 2021-07-05 by Jendker

yanlai00, 2021-07-09:
Hi @swiz23, I am trying to follow your instructions on a widowX250s robot with 9 motors. However, the `bot.dxl.robot_get_motor_registers("group", "all", "Hardware_Error_Status")` function only returns an array of length 7 for me, even when everything seems to be working completely fine. After I run the `bot.dxl.robot_reboot_motors("group", "all", enable=True, smart_reboot=True)` command, the two motors near the gripper also won't get rebooted. Any ideas on how to debug this?