Interbotix/interbotix_ros_manipulators #14: ROS2 release plans

Serafadam, 2021-07-01:
Hi, I'd like to ask if there are any plans on releasing a ROS2 version of this project?

LSinterbotix, 2021-07-01:
There are no immediate plans to port this to ROS2, though it may be something we think about in the future.

Serafadam, 2021-07-01:
I could try to help with that, since there is a possibility I will be working with one of the arms (though I think I'll start with some ros1bridge shenaningans). What would be the good point starting based https://ros-controls.github.io/control.ros.org/differences_to_ros1.html#robothardware-to-components?

swiz23, 2021-07-02:
Hi @Serafadam,

The link you mentioned is specifically for porting over ros_control from ROS1 to ROS2. I haven't really looked at the link, but I just wanted to let you know that ros_control is only used in this repo to work with the arm in MoveIt. However, many of the ROS packages (like the interbotix_xsarm_control, interbotix_xsarm_joy, interbotix_xsarm_dual, interbotix_xsarm_puppet) work either directly with the interbotix_xs_sdk or via our custom Python API - without ros_control.

The best way to start porting to ROS2 would be to get the interbotix_xs_sdk ROS package located in the interbotix_ros_core repo over to ROS2. It might be easiest to do this with the ROS1 bridge. Following that, you would want to port the interbotix_xsarm_descriptions package and the interbotix_xsarm_control package over to ROS2 (you could probably do this in ROS2 straight-up as opposed to using the bridge).

swiz23, 2021-07-16:
Hi @Serafadam,

If you ever do create a ROS2 port, feel free to post a link to it here - or send an email to trsupport@trossenrobotics.com with the link to let us know. We would be happy to check it out!

As the issue appears to be answered, it will be closed. Feel free to reopen if you have any questions!

closed 2021-07-16 by swiz23

Serafadam, 2021-08-03:
Hi @swiz23 , I've begun work on the port, starting from `interbotix_xs_sdk`, and I wanted to ask about the concept behind one-shot timers used for example here https://github.com/Interbotix/interbotix_ros_core/blob/a3fe00ee42f059df4f3f59722ecc23debbe8d0da/interbotix_ros_xseries/interbotix_xs_sdk/src/xs_sdk_obj.cpp#L957
I'm kinda unfamiliar with how this works, is this something that allows blocking of next message? I'm not sure what would be the best replacement here, since ROS2 timers don't have such functions (and I'm not sure whether this code is necessary).

Serafadam, 2021-08-25:
@swiz23 @LSinterbotix  I managed to build xs_sdk on Galactic, though I haven't been able to test it yet (I need to update robot_descriptions for that) I had to change the code a little bit :
- removed dynamixel_toolbox since I used cloned ros2 version from repository
- made a separate package for msg and srv files (xarm_msgs)
- changed how robot_description gets passed (urdf libs probably had some api changes)
- some cosmetic changes
- for xs_sdk_sim, urdf-parser-py also needs to be cloned into repository
- I didn't update the documentation yet, as probably there will be some more changes
I also started some work on uxarms, but I do not have that robot so I didn't commit my changes for that
Link: https://github.com/Serafadam/interbotix_ros_core/tree/ros2-galactic

Serafadam, 2021-08-30:
@swiz23 I managed to do some tests with the sdk on a real robot, and it seems like it's gonna work :slightly_smiling_face:
Although one thing that I'm thinking about is how trajectory execution works. Couldn't it be only activated after receiving the trajectory message? In other words, why not make it a regular function instead of a timer?

swiz23, 2021-09-03:
The way the trajectory execution currently works is that there is a [subscriber function](https://github.com/Interbotix/interbotix_ros_core/blob/a40446b42175d7cfaaaa5fa649e3b420b0397fd2/interbotix_ros_xseries/interbotix_xs_sdk/src/xs_sdk_obj.cpp#L790) that takes in a trajectory message. Assuming the trajectory message is valid, it calls the [one shot timer function](https://github.com/Interbotix/interbotix_ros_core/blob/a40446b42175d7cfaaaa5fa649e3b420b0397fd2/interbotix_ros_xseries/interbotix_xs_sdk/src/xs_sdk_obj.cpp#L944) to start it [here](https://github.com/Interbotix/interbotix_ros_core/blob/a40446b42175d7cfaaaa5fa649e3b420b0397fd2/interbotix_ros_xseries/interbotix_xs_sdk/src/xs_sdk_obj.cpp#L824-L825). For each pass through the one shot timer function, the one shot timer's period is reset so that it is called at the point in time that the next waypoint in the trajectory should be executed.

So to answer your question - it actually is only activated after receiving a trajectory message. As far as why I'm using a timer - that's because all the code is currently single threaded. So if I were to just have a 'for' loop within the trajectory subscriber function to parse through the trajectory, it would block other callbacks from being called. Of course there are ways around this - for example you could use an action or multi-threading. But I didn't want to go that route since I didn't want to over-complicate things.

Serafadam, 2021-10-04:
Hi @swiz23, I updated branches a little bit, and tried to use Moveit2 together with ros control to move the manipulator, though I am having some difficulties with that, could you help me a little bit with debugging?
Currently my setup is as follows:
A workspace with all repos mentioned [here](https://github.com/Serafadam/interbotix_ros_manipulators/blob/galactic/interbotix_ros_xsarms/install/interbotix.repos) downloaded into the workspace (some of them can be downloaded with rosdep, but I wanted to build them from source for debugging purposes).
After building the repos, I open 3 terminal windows:
In one of them, I run 
`ros2 launch interbotix_xsarm_control xsarm_control.launch.py robot_model:=rx200 robot_name:=rx200 use_sim:=true use_rviz:=false` (use_sim:=false also works, but more on that later)
In the other one, I run `
ros2 launch interbotix_xsarm_ros_control xsarm_ros_control.launch.py robot_model:=rx200 robot_name:=rx200 use_sim:=false` (If you want to use another robot you'll have to update the controller files, since I haven't figured out why default namespacing here does not work)
In the third terminal I launch 
`ros2 launch interbotix_xsarm_moveit move_group.launch.py robot_model:=rx200 robot_name:=rx200`
And Moveit window successfully launches.

Now, the main issues I face: 
- Robot state is not updated by Motion Planning scene, as if it does not get the state continously, after I restart the scene, the current robot state is correctly displayed. Strangely enough, TF seems to be correctly updated
[image]
In this image, you can see that the current robot state (black one) differs very much from the true one.
- After sending goals to the real arm, it moves very slowly, and does not reach the desired state, I'm not sure whether this is an issue related to the driver, or the dynamixel servos, it would be very helpful to test that on another robot, sadly I have only one at my disposal.
I'm happy to help with the setup