Interbotix/interbotix_ros_manipulators #19: [Bug]: Position-only-kinematics

m-elwin, 2021-10-30:
### What happened?

The lma_kinematics plugin seems to (almost?) always fail when attempting to do position-only kinematics.

### Robot Model

px100

### Operating System

Ubuntu 20.04

### Steps To Reproduce

1. Launch the px100 with moveit and the fake node
2. Use moveit to plan to a position target of (0.2, 0, 0.19). This will fail
3. Use moveit to plan to a pose target of (0.2, 0, 0.19) (0, 0, 0, 1). This will succeed
4. Editing kinematics.yaml and either
    1. Setting `position_only_ik: true` for the lma solver makes this work.
    2. Using the kdl_kinematics_solver rather than lma also works

### Anything Else

A comment in kinematics.yaml says that "position-only-ik and orientation-only-ik work automatically for the LMA plugin".
This does not seem to be true in my experience: adding `position_only_ik: true seems to fix the issue`.

LSinterbotix, 2021-11-01:
Hi @m-elwin! Thanks for catching that. It seems like `false` is the default for the parameter according to the [lma source](https://github.com/ros-planning/moveit/blob/d1465a7dd55ffead03b35f9d0d3d878fc4757eeb/moveit_kinematics/lma_kinematics_plugin/src/lma_kinematics_plugin.cpp#L129). We will update the configuration file and its comments accordingly.

For now, just manually specify `position_only_ik: true` under the lma plugin config, or use the kdl plugin.

swiz23, 2021-11-07:
I feel like I probably wrote that comment since MoveIt's documentation only says that position_only_ik is supported by the kdl kinematics plugin but doesn't mention it by the lma one. See http://docs.ros.org/en/kinetic/api/moveit_tutorials/html/doc/kinematics_configuration/kinematics_configuration_tutorial.html#position-only-ik.
There may have been another reason that I wrote that, but it has been a while and it's obviously incorrect - so good catch!

LSinterbotix, 2021-11-09:
Closing this issue since the comments and parameters in [kinematics.yml](https://github.com/Interbotix/interbotix_ros_manipulators/blob/bf59d608f8b17bedb57a43b3866b886035c3ec3d/interbotix_ros_xsarms/interbotix_xsarm_moveit/config/kinematics.yaml) were fixed in commit bf59d60. Feel free to reopen if needed.

closed 2021-11-09 by LSinterbotix