Interbotix/interbotix_ros_manipulators #21: [Question]:

autonomousTurtle, 2021-11-23:
### Question

Hello! 

First off, thank you for all of the great info you are publishing into this community! 

**My question:**
I am attempting to follow your youtube video [here](https://www.youtube.com/watch?v=Z9s3hRYcIp0) - Interbotix Tutorials: X-Series Arms | Using The Interbotix MoveIt Interface ROS Package

Using unmodified files, and am running the moveit interface with your cpp gui using: 

`roslaunch interbotix_xsarm_moveit_interface xsarm_moveit_interface.launch robot_model:=wx250s use_fake:=true dof:=6 use_cpp_interface:=true
`
however, every time I press the Plan Pose or Plan Orientation buttons on the GUI, the result is not successful. Only the plan position is successful. 

Output on the terminal: 

[ERROR] [1637678459.896249301]: interbotix_arm/interbotix_arm: Unable to sample any valid states for goal tree
[ INFO] [1637678459.896279412]: interbotix_arm/interbotix_arm: Created 1 states (1 start + 0 goal)
[ INFO] [1637678459.896315740]: No solution found after 5.008456 seconds
[ INFO] [1637678459.899706765]: Unable to solve the planning problem
[ WARN] [1637678459.899981409]: Fail: ABORTED: No motion plan found. No execution attempted.
[ INFO] [1637678459.900114777]: No planning scene passed into moveit_visual_tools, creating one.
[ INFO] [1637678459.903163485]: Loading robot model 'wx250s'...
[ INFO] [1637678459.903177667]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1637678459.925218288]: Using position only ik
[ INFO] [1637678460.057573819]: Publishing maintained planning scene on ''
[ WARN] [1637678461.059107885]: Skipping path because 0 points passed in.
[ INFO] [1637678461.059261588]: success: 0
[INFO] [1637678461.060084]: Planning EE pose was not successful.

Is there a step or config file I am missing? 

Thank you!

### Robot Model

wx250s, simulated

### Operating System

Ubuntu 20.04, ROS noetic

LSinterbotix, 2021-11-24:
Hi @autonomousTurtle! Try setting [this parameter](https://github.com/Interbotix/interbotix_ros_manipulators/blob/4a1b2ec08ad6a99d23bf5db8e66370d7ac439dbb/interbotix_ros_xsarms/interbotix_xsarm_moveit/config/kinematics.yaml#L26) to `false` or commenting it out.

autonomousTurtle, 2021-11-29:
> Hi @autonomousTurtle! Try setting [this parameter](https://github.com/Interbotix/interbotix_ros_manipulators/blob/4a1b2ec08ad6a99d23bf5db8e66370d7ac439dbb/interbotix_ros_xsarms/interbotix_xsarm_moveit/config/kinematics.yaml#L26) to `false` or commenting it out.

That did the trick. Thank you!

closed 2021-11-29 by autonomousTurtle