Interbotix/interbotix_ros_manipulators #25: [Question]: Interbotix_Xsarm_Perception

lcouso-byte, 2022-04-06:
### Question

Hello team Trossen Robotics,

I hope you are doing well. My name is Lazaro Couso and I am a student from the University of Florida doing a research project using your interbotix widow arm. I would appreciate it if you guys can help to set up the interbotix_xsarm perception package so I can launch it using this command: "roslaunch interbotix_​xsarm_perception xsarm_perception.launch robot_model:=wx250s" as one of your engineer explains the use of this command in this video, https://youtu.be/UesfMYM4qcc, at time 1:04.

When I installed the X-Series Arm Ros using the installation process from your docs, I noticed that in the directory build/interbotix_ros_manipulator/interbotix_ros_xsarms/, there is no interbotix_​xsarm_perception package. Because of this, I am not able to follow along in the tutorial video shared above.

Please let me know what I can do about this,

Lazaro Couso

### Robot Model

wx250s

### Operating System

Ubuntu 20.04

LSinterbotix, 2022-04-06:
Please check if the [`CATKIN_IGNORE`](https://github.com/Interbotix/interbotix_ros_manipulators/blob/9693e594c6d0ce51ff7080248d4de3c3fe205368/interbotix_ros_xsarms/interbotix_xsarm_perception/CATKIN_IGNORE) file in the xsarm_perception package's source exists, as well as the workspace directories `~/realsense_ws` and `~/apriltag_ws`. If the `CATKIN_IGNORE` file exists and the two workspaces do not, then you may not have accepted the perception option during the install.

To fix this, delete, rename, or move your `~/interbotix_ws` directory and re-run the [`xsarm_amd64_install.sh`](https://github.com/Interbotix/interbotix_ros_manipulators/blob/9693e594c6d0ce51ff7080248d4de3c3fe205368/interbotix_ros_xsarms/install/amd64/xsarm_amd64_install.sh) installation script. Make sure to say yes to the "Install the Perception Pipeline (includes RealSense and AprilTag packages)?" prompt. Check for any errors in the terminal when the script is running.

If this is not the case, let us know.

LSinterbotix, 2022-04-06:
Great! Feel free to close this if your problem has been solved.

closed 2022-04-07 by lcouso-byte