Interbotix/interbotix_ros_manipulators #28: [Question]: The install instruction for galactic wrapper

Zhuomin-Zhou, 2022-05-10:
### Question

Hi,

Is there an install instruction for the galactic wrapper?
I have tried to use **colcon build**  for the **interbotix_ros_core**, **interbotix_ros_manipulators**, and **interbotix_ros_toolboxes**, but that was not working.

Thanks in advance!
Min 

### Robot Model

wx250 6DOF

### Operating System

Ubuntu20.04

LSinterbotix, 2022-05-16:
We are in the middle of porting our packages to ROS2 galactic and do not have install instructions or an install script just yet. Check out how to use at least the core packages in [this issue](https://github.com/Interbotix/interbotix_ros_core/issues/19).

LSinterbotix, 2022-06-01:
@Zhuomin-Zhou Update: Follow [this guide](https://www.trossenrobotics.com/docs/interbotix_xsarms/ros_interface/ros2_software_setup.html) to install packages for ROS2 Galactic.