Interbotix/interbotix_ros_manipulators #32: [Question]: The position error?

Zhuomin-Zhou, 2022-06-09:
### Question

Hi, 

When I was trying to use **moveit** to do the path plan, the planned position did not match the goal position, which was symmetric with the goal.
When I did not give the goal position and then give it plan command, it would go to a specific position. 
But when I was trying to use Python API to control the arm, everything looks normal.

Could you have a look if are there some config file errors or something?
The screenshots are attached.

The symmetric with goal:
[image]

Plan Without goal:
[image]

The information in the terminal:

min@min:~/interbotix_ros2_ws$ ros2 launch interbotix_xsarm_moveit xsarm_moveit.launch.py robot_model:=wx250s hardware_type:=actual
[INFO] [launch]: All log files can be found below /home/min/.ros/log/2022-06-09-19-16-40-677573-min-36063
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [36083]
[INFO] [rviz2-2]: process started with pid [36085]
[INFO] [ros2_control_node-3]: process started with pid [36087]
[INFO] [spawner-4]: process started with pid [36089]
[INFO] [spawner-5]: process started with pid [36091]
[INFO] [xs_sdk-6]: process started with pid [36093]
[INFO] [robot_state_publisher-7]: process started with pid [36095]
[xs_sdk-6] [INFO] Using Interbotix X-Series Driver Version: 'v0.2.1'.
[xs_sdk-6] [INFO] Loaded mode configs from '/home/min/interbotix_ros2_ws/install/interbotix_xsarm_moveit/share/interbotix_xsarm_moveit/config/modes.yaml'.
[xs_sdk-6] [INFO] Loaded motor configs from '/home/min/interbotix_ros2_ws/install/interbotix_xsarm_control/share/interbotix_xsarm_control/config/wx250s.yaml'.
[move_group-1] [WARN] [1654766201.846310293] [move_group.move_group]: MoveGroup launched without ~default_planning_pipeline specifying the namespace for the default planning pipeline configuration
[move_group-1] [WARN] [1654766201.846358903] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-1] [INFO] [1654766201.848459874] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00197784 seconds
[move_group-1] [INFO] [1654766201.848499405] [moveit_robot_model.robot_model]: Loading robot model 'wx250s'...
[move_group-1] [INFO] [1654766201.848509392] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[move_group-1] Link wx250s/base_link had 1 children
[move_group-1] Link wx250s/shoulder_link had 1 children
[move_group-1] Link wx250s/upper_arm_link had 1 children
[move_group-1] Link wx250s/upper_forearm_link had 1 children
[move_group-1] Link wx250s/lower_forearm_link had 1 children
[move_group-1] Link wx250s/wrist_link had 1 children
[move_group-1] Link wx250s/gripper_link had 1 children
[move_group-1] Link wx250s/ee_arm_link had 2 children
[move_group-1] Link wx250s/gripper_prop_link had 0 children
[move_group-1] Link wx250s/gripper_bar_link had 1 children
[move_group-1] Link wx250s/fingers_link had 3 children
[move_group-1] Link wx250s/ee_gripper_link had 0 children
[move_group-1] Link wx250s/left_finger_link had 0 children
[move_group-1] Link wx250s/right_finger_link had 0 children
[xs_sdk-6] [INFO] Found Dynamixel ID : '8',	Model : 'XL430-W250',	Joint Name : 'wrist_rotate'
[move_group-1] [INFO] [1654766201.891336824] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-1] [INFO] [1654766201.891610603] [moveit.ros_planning_interface.moveit_cpp]: Listening to '/wx250s/joint_states' for joint states
[move_group-1] [INFO] [1654766201.891990331] [moveit_ros.current_state_monitor]: Listening to joint states on topic '/wx250s/joint_states'
[move_group-1] [INFO] [1654766201.892411405] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[move_group-1] [INFO] [1654766201.892435151] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-1] [INFO] [1654766201.893042004] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[move_group-1] [INFO] [1654766201.893076855] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-1] [INFO] [1654766201.893403713] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-1] [INFO] [1654766201.893658425] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-1] [WARN] [1654766201.893884362] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[move_group-1] [ERROR] [1654766201.893901823] [moveit.ros.occupancy_map_monitor.middleware_handle]: No sensor plugin specified for octomap updater ; ignoring.
[move_group-1] [INFO] [1654766201.893913207] [moveit.ros.occupancy_map_monitor.middleware_handle]: Skipping octomap updater plugin ''
[move_group-1] [INFO] [1654766201.896809464] [moveit.ros_planning_interface.moveit_cpp]: Loading planning pipeline 'move_group'
[xs_sdk-6] [INFO] Found Dynamixel ID : '7',	Model : 'XM430-W350',	Joint Name : 'wrist_angle'
[move_group-1] [INFO] [1654766201.916747604] [moveit.ros_planning.planning_pipeline]: Using planning interface 'OMPL'
[move_group-1] [INFO] [1654766201.918544276] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.path_tolerance' was not set. Using default value: 0.100000
[move_group-1] [INFO] [1654766201.918563902] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.resample_dt' was not set. Using default value: 0.100000
[move_group-1] [INFO] [1654766201.918569140] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.min_angle_change' was not set. Using default value: 0.001000
[... 144 lines cut ...]
[move_group-1] [INFO] [1654766631.308240337] [moveit_ros.plan_execution]: Planning attempt 1 of at most 1
[move_group-1] [INFO] [1654766631.308249207] [moveit_move_group_capabilities_base.move_group_capability]: Using planning pipeline 'move_group'
[move_group-1] [INFO] [1654766631.308594779] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'interbotix_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[move_group-1] [INFO] [1654766631.331728860] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1654766631.331840814] [moveit_ros.trajectory_execution_manager]: Validating trajectory with allowed_start_tolerance 0.01
[move_group-1] [INFO] [1654766631.338188858] [moveit_ros.trajectory_execution_manager]: Starting trajectory execution ...
[move_group-1] [INFO] [1654766631.338252762] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1654766631.338410184] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: sending trajectory to /wx250s/arm_controller
[ros2_control_node-3] [INFO] [1654766631.338667058] [wx250s.arm_controller]: Received new action goal
[ros2_control_node-3] [INFO] [1654766631.338716645] [wx250s.arm_controller]: Accepted new action goal
[move_group-1] [INFO] [1654766631.338811978] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: /wx250s/arm_controller started execution
[move_group-1] [INFO] [1654766631.338824480] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal request accepted!
[ros2_control_node-3] [INFO] [1654766636.471615880] [wx250s.arm_controller]: Goal reached, success!
[move_group-1] [INFO] [1654766636.471855714] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Controller '/wx250s/arm_controller' successfully finished
[move_group-1] [INFO] [1654766636.497953702] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status SUCCEEDED ...
[move_group-1] [INFO] [1654766636.498438607] [moveit_move_group_default_capabilities.move_action_capability]: Solution was found and executed.

### Robot Model

wx250s

### Operating System

Ubuntu20.04
ROS2 galactic

LSinterbotix, 2022-06-09:
This does seem to be an issue on our end - we'll look into it and get back to you with a fix. Thanks for reporting this!

Zhuomin-Zhou, 2022-06-10:
> This does seem to be an issue on our end - we'll look into it and get back to you with a fix. Thanks for reporting this!
Hi @LSinterbotix 

Thanks for your reply.
No worries.

Regards,
Min

LSinterbotix, 2022-07-02:
@Zhuomin-Zhou Thank you for your patience. I believe that the changes in 7fc83d8 solve this issue. Let me know if it works for you and we can close this issue.

Zhuomin-Zhou, 2022-07-03:
Hi @LSinterbotix,

Thanks for your help.
I have used the new urdf file, but the new errors are showing.
The motion can be planned but cannot be executed.
And the Python interface control works well. 
The logs of the terminal and screenshot are shown below.

Regards,
Min

Screenshot:
[image]

min@min:~/interbotix_ros2_ws$ ros2 launch interbotix_xsarm_moveit xsarm_moveit.launch.py robot_model:=wx250s hardware_type:=actual
[INFO] [launch]: All log files can be found below /home/min/.ros/log/2022-07-03-20-31-18-183066-min-99672
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [99692]
[INFO] [rviz2-2]: process started with pid [99694]
[INFO] [ros2_control_node-3]: process started with pid [99696]
[INFO] [spawner-4]: process started with pid [99698]
[INFO] [spawner-5]: process started with pid [99700]
[INFO] [xs_sdk-6]: process started with pid [99702]
[INFO] [robot_state_publisher-7]: process started with pid [99704]
[xs_sdk-6] [INFO] Using Interbotix X-Series Driver Version: 'v0.3.1'.
[xs_sdk-6] [INFO] Using logging level 'INFO'.
[xs_sdk-6] [INFO] Loaded mode configs from '/home/min/interbotix_ros2_ws/install/interbotix_xsarm_moveit/share/interbotix_xsarm_moveit/config/modes.yaml'.
[xs_sdk-6] [INFO] Loaded motor configs from '/home/min/interbotix_ros2_ws/install/interbotix_xsarm_control/share/interbotix_xsarm_control/config/wx250s.yaml'.
[ros2_control_node-3] [WARN] [1656844279.309434449] [wx250s.controller_manager]: 'update_rate' parameter not set, using default value.
[xs_sdk-6] [INFO] Pinging all motors specified in the motor_config file. (Attempt 1/3)
[move_group-1] [WARN] [1656844279.329099852] [move_group.move_group]: MoveGroup launched without ~default_planning_pipeline specifying the namespace for the default planning pipeline configuration
[move_group-1] [WARN] [1656844279.329151953] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-1] [INFO] [1656844279.331094916] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00180601 seconds
[move_group-1] [INFO] [1656844279.331122642] [moveit_robot_model.robot_model]: Loading robot model 'wx250s'...
[move_group-1] [INFO] [1656844279.331130744] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[move_group-1] Link wx250s/base_link had 1 children
[move_group-1] Link wx250s/shoulder_link had 1 children
[move_group-1] Link wx250s/upper_arm_link had 1 children
[move_group-1] Link wx250s/upper_forearm_link had 1 children
[move_group-1] Link wx250s/lower_forearm_link had 1 children
[move_group-1] Link wx250s/wrist_link had 1 children
[move_group-1] Link wx250s/gripper_link had 1 children
[move_group-1] Link wx250s/ee_arm_link had 2 children
[move_group-1] Link wx250s/gripper_prop_link had 0 children
[move_group-1] Link wx250s/gripper_bar_link had 1 children
[move_group-1] Link wx250s/fingers_link had 3 children
[move_group-1] Link wx250s/ee_gripper_link had 0 children
[move_group-1] Link wx250s/left_finger_link had 0 children
[move_group-1] Link wx250s/right_finger_link had 0 children
[xs_sdk-6] [INFO] 	Found DYNAMIXEL ID:  8, Model: 'XL430-W250', Joint Name: 'wrist_rotate'.
[move_group-1] [INFO] [1656844279.378051534] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-1] [INFO] [1656844279.378350661] [moveit.ros_planning_interface.moveit_cpp]: Listening to '/wx250s/joint_states' for joint states
[move_group-1] [INFO] [1656844279.378759856] [moveit_ros.current_state_monitor]: Listening to joint states on topic '/wx250s/joint_states'
[move_group-1] [INFO] [1656844279.379267317] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[move_group-1] [INFO] [1656844279.379292111] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-1] [INFO] [1656844279.379908034] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[move_group-1] [INFO] [1656844279.379934992] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-1] [INFO] [1656844279.380255141] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-1] [INFO] [1656844279.380811840] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-1] [WARN] [1656844279.381289129] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[move_group-1] [ERROR] [1656844279.381315389] [moveit.ros.occupancy_map_monitor.middleware_handle]: No sensor plugin specified for octomap updater ; ignoring.
[move_group-1] [INFO] [1656844279.381332431] [moveit.ros.occupancy_map_monitor.middleware_handle]: Skipping octomap updater plugin ''
[move_group-1] [INFO] [1656844279.384772423] [moveit.ros_planning_interface.moveit_cpp]: Loading planning pipeline 'move_group'
[xs_sdk-6] [INFO] 	Found DYNAMIXEL ID:  7, Model: 'XM430-W350', Joint Name: 'wrist_angle'.
[move_group-1] [INFO] [1656844279.401919728] [moveit.ros_planning.planning_pipeline]: Using planning interface 'OMPL'
[... cut ...]
[move_group-1] [ERROR] [1656844383.364433334] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Action client not connected to action server: /wx250s/arm_controller/follow_joint_trajectory
[move_group-1] [ERROR] [1656844383.364451213] [moveit_ros.trajectory_execution_manager]: Failed to send trajectory part 1 of 1 to controller /wx250s/arm_controller
[move_group-1] [INFO] [1656844383.364467905] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status ABORTED ...
[move_group-1] [INFO] [1656844383.374379604] [moveit_move_group_default_capabilities.move_action_capability]: Solution found but controller failed during execution

Zhuomin-Zhou, 2022-07-03:
Hi @LSinterbotix,

Could I ask how I can make the trajectory of the end effector have a solution when making the end effector go to the negative axis of the Z-direction in the python interface, like set_ee_pose_components(x=.1, y = .2, z=-.2)?

Thanks,
Min

Zhuomin-Zhou, 2022-07-03:
> 

Hi @LSinterbotix,

I have tried to delete some joint limit check code in the arm.py, which works.
Do you have a better method to skip the joint limit checking?
How can I change the joint limitation?

Regards,
Min

Zhuomin-Zhou, 2022-07-05:
> 

Hi @LSinterbotix,

The reason that the error happened before is that the galactic branch was used.
I did not notice that the new urdf was changed in the ros2-devel branch.
Thanks for your help!!

Regards,
Min

LSinterbotix, 2022-07-05:
Has everything been solved then or are you still experiencing the issue in your last comment? I would recommend opening a new issue if the problem has changed.

Zhuomin-Zhou, 2022-07-05:
The original issue is solved in the ros2-devel branch, But the new issue happened in galactic branch. Thanks.> Has everything been solved then or are you still experiencing the issue in your last comment? I would recommend opening a new issue if the problem has changed.

LSinterbotix, 2022-07-06:
Try the changes in commit https://github.com/Interbotix/interbotix_ros_manipulators/commit/195bd1573d5e660260d44e9b5125067b100149a7#diff-9e059c2ed3e88604af02c17c33bcd0fb8655e40c32cb0c9d174371d0f159786d for the wx250s controllers config.

LSinterbotix, 2022-07-13:
Do you have any further questions on this? Otherwise, this issue will be closed.

Zhuomin-Zhou, 2022-07-14:
Hi @LSinterbotix,

I don't have any further questions.
I appreciate your help!

Regards,
Min

commit 7fc83d81 (2022-07-02): [xsarm_descriptions] Fix control joint order

Move forearm_roll joint to its corrent sequence position.

Fixes #32
commit 9bc407b8 (2022-10-19): [xsarm_descriptions] Fix control joint order

Move forearm_roll joint to its corrent sequence position.

Fixes #32
closed 2022-07-15 by LSinterbotix