Interbotix/interbotix_ros_manipulators #33: [Bug]: joint_state_publisher_gui process has died

ManuelKochETH, 2022-06-10:
### What happened?

After installing the software according to "ROS2 Standard Software Setup" I went through the "ROS2 Quickstart Guide" and ran:
`ros2 launch interbotix_xsarm_descriptions xsarm_description.launch.py robot_model:=wx200 use_joint_pub_gui:=true`
I obtain an error saying that the joint_state_publisher_gui has died. There is no gui visible and the robot is not displayed properly (see screenshot)
[image]
The error in the visualization is probably due to missing transforms. And I assume this is correlated with the missing joint state publisher.
Running `ros2 run tf2_tools view_frames` gives me the following:
[frames.pdf](https://github.com/Interbotix/interbotix_ros_manipulators/files/8880053/frames.pdf)

### Robot Model

In reality wx250s, but I tried several in this simulation e.g. wx200

### Operating System

Ubuntu 20.04

### Steps To Reproduce

Run:
`ros2 launch interbotix_xsarm_descriptions xsarm_description.launch.py robot_model:=wx200 use_joint_pub_gui:=true`

### Relevant log output

```shell
mankoch@manuel:~$ ros2 launch interbotix_xsarm_descriptions xsarm_description.launch.py robot_model:=wx200 use_joint_pub_gui:=true
[INFO] [launch]: All log files can be found below /home/mankoch/.ros/log/2022-06-10-16-52-24-924161-manuel-vm-10900
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [10903]
[INFO] [joint_state_publisher_gui-2]: process started with pid [10905]
[INFO] [rviz2-3]: process started with pid [10907]
[ERROR] [joint_state_publisher_gui-2]: process has died [pid 10905, exit code 1, cmd '/opt/ros/galactic/lib/joint_state_publisher_gui/joint_state_publisher_gui --ros-args -r __ns:=/wx200 --params-file /tmp/launch_params_lnnt_my_'].
```

LSinterbotix, 2022-06-10:
Try commenting out this line:

https://github.com/Interbotix/interbotix_ros_manipulators/blob/8011074fc6bf9389382036757d1dfa577fcdd76c/interbotix_ros_xsarms/interbotix_xsarm_descriptions/launch/xsarm_description.launch.py#L90

The joint_state_publisher_gui node raises a ParameterAlreadyDeclaredException when provided the use_sim_time parameter.

<details>
<summary>Exception:</summary>

```console
~/ros2_ws $ ros2 launch interbotix_xsarm_descriptions xsarm_description.launch.py robot_model:=wx200 use_joint_pub_gui:=true
[INFO] [launch]: All log files can be found below /home/interbotix/.ros/log/2022-06-10-11-06-35-657398-interbotix-113286
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [113305]
[INFO] [joint_state_publisher_gui-2]: process started with pid [113307]
[INFO] [rviz2-3]: process started with pid [113309]
[joint_state_publisher_gui-2] Traceback (most recent call last):
[joint_state_publisher_gui-2]   File "/opt/ros/galactic/lib/joint_state_publisher_gui/joint_state_publisher_gui", line 11, in <module>
[joint_state_publisher_gui-2]     load_entry_point('joint-state-publisher-gui==2.2.0', 'console_scripts', 'joint_state_publisher_gui')()
[joint_state_publisher_gui-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/joint_state_publisher_gui/joint_state_publisher_gui.py", line 262, in main
[joint_state_publisher_gui-2]     JointStatePublisher(parsed_args.urdf_file))
[joint_state_publisher_gui-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/joint_state_publisher/joint_state_publisher.py", line 217, in __init__
[joint_state_publisher_gui-2]     super().__init__('joint_state_publisher', automatically_declare_parameters_from_overrides=True)
[joint_state_publisher_gui-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/node.py", line 216, in __init__
[joint_state_publisher_gui-2]     self.declare_parameters(
[joint_state_publisher_gui-2]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/node.py", line 487, in declare_parameters
[joint_state_publisher_gui-2]     raise ParameterAlreadyDeclaredException(parameters_already_declared)
[joint_state_publisher_gui-2] rclpy.exceptions.ParameterAlreadyDeclaredException: ('Parameter(s) already declared', ['use_sim_time'])
[ERROR] [joint_state_publisher_gui-2]: process has died [pid 113307, exit code 1, cmd '/opt/ros/galactic/lib/joint_state_publisher_gui/joint_state_publisher_gui --ros-args -r __ns:=/wx200 --params-file /tmp/launch_params_0vgmxe2t'].
```

</details>

ManuelKochETH, 2022-06-10:
Thanks a lot. 
I commented out this line and colcon built the interbotix_ws folder again. Now it works.

commit 5c32eb36 (2022-06-10): [xsarm_descriptions] Fix joint state pub gui

Removed the use_sim_time parameter from the joint_state_publisher_gui
node action. Fixes #33.

[skip ci]
closed 2022-06-10 by ManuelKochETH