Interbotix/interbotix_ros_manipulators #35: [Bug]: 2 motors detected but other 3 not detected

sbhambr1, 2022-06-16:
### What happened?

Hi,

Using the Dynamixel Wizard 2.0 Tool, I am able to detect 2 motors of XL430-W250 (ID 1 & 5 - firmware v46) but unable to detect other 3 motors (on firmware v45 - which I could read earlier when they were working fine). I tried updating the firmware and the firmware installation failed each time for motor IDs 2, 3 & 4.

Any help or support on this will be appreciated. Thanks!

### Robot Model

px100

### Operating System

Ubuntu 20.04
ROS Noetic

### Steps To Reproduce

Open Dynamixel Wizard 2.0 Tool
Scan for Protocol 2.0 on ttyUSB0 port and 1000000 bps baudrate for IDs 1-252.

LSinterbotix, 2022-06-16:
See if anything in the Troubleshooting Guide's ['Can't find Dynamixel ID' section](https://www.trossenrobotics.com/docs/interbotix_xsarms/troubleshooting/index.html#can-t-find-dynamixel-id) helps at all.

Have you changed anything about the physical configuration of the arm since you were last able to run it?

sbhambr1, 2022-06-16:
Hi, thanks a lot for your quick reply!
Yes, I tried all the steps from the mentioned section but couldn't resolve the issue.

I tried updating the firmware on the 3 motors that were on v45 and all three updates failed midway after which the motors were not being recognized.
Also, after I unplugged and plugged the arm back, none of the red LEDs of these 3 motors blink.

LSinterbotix, 2022-06-16:
Try doing a [firmware recovery](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/#firmware-recovery) on the servos. Follow the directions carefully. Try to only have a single servo connected to the U2D2 when running the recovery.

sbhambr1, 2022-06-17:
Thanks for the response!
To be sure that I am following correctly, I need to perform the firmware recovery on the two motors that are being recognized to v45 (then all motors would be on the same version). Request you to please confirm/correct so I can proceed further.

LSinterbotix, 2022-06-17:
You would be performing the firmware recovery on the servos that are **not recognized** by the Dynamixel Wizard 2.0 Tool. It's possible that the failed update broke something internally and their firmware needs to be restored to a working state.

sbhambr1, 2022-06-17:
Thanks for your response. Since I am unable to view any of the other 3 Dynamixel servos, I am not sure how to perform a firmware recovery on those servos.
I may be getting confused here, but is it the case that the firmware recovery happens for all servos when I follow the instructions as shown [here](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/#firmware-recovery)? If not, then how am I supposed to select the 3 non-recognized servos for firmware recovery?

LSinterbotix, 2022-06-17:
You would connect a single unresponsive servo to the U2D2, do the firmware recovery on that servo, and continue to the next unresponsive servo. The firmware recovery tool will (if it succeeds) be able to write to the servo as long as you choose the [correct model](https://www.trossenrobotics.com/docs/interbotix_xsarms/specifications/px100.html#default-servo-configurations) and select the right port.

sbhambr1, 2022-06-23:
Thanks for your response! And sorry for responding late to this.
I am trying to follow the steps as you mentioned, and will update here as soon as I have something!

closed 2023-01-05 by LSinterbotix