Interbotix/interbotix_ros_manipulators #38: [Question]: Python support for Gazebo

sbhambr1, 2022-06-22:
### Question

Hi, I wanted to ask if there is a way to run the python demos for simulating the robot in Gazebo too?

### Robot Model

px100

### Operating System

Ubuntu 20.04

### ROS Version

ROS1 Noetic

sbhambr1, 2022-06-22:
Okay perfect! Thanks for the quick response and guiding me into the right direction.

LSinterbotix, 2022-06-22:
Try adding the dynamics tag below to each non-fixed joint.

```xml
<dynamics friction="0.1" damping="0.1"/>
```

sbhambr1, 2022-06-22:
Thanks for your response! I tried but the same issue is happening again.

LSinterbotix, 2022-06-22:
I experienced the same problem when we ported our description files over to ROS2. We solved it by adding the dynamics tag, but we may have done something else as well. Try checking out the [px100 URDF](https://github.com/Interbotix/interbotix_ros_manipulators/blob/galactic/interbotix_ros_xsarms/interbotix_xsarm_descriptions/urdf/px100.urdf.xacro) on that branch.

Could you provide the contents of your world/SDF file so I can help troubleshoot?

sbhambr1, 2022-06-22:
Thanks for sharing the link! I will check that out.
I am attaching the world file (grasp_object.world) and the modified urdf file (px100.urdf.xacro) where I primarily made two changes: 1) updated the z coordinates for all links as per the table height of 1, and 2) added the dynamics tag as you shared above. Please let me know if I am making any other mistakes in these two files, once you are able to troubleshoot the issue
[Files.zip](https://github.com/Interbotix/interbotix_ros_manipulators/files/8960821/Files.zip)
.

LSinterbotix, 2022-06-22:
Seems like the issue is that the table and the robot are in collision. To fix this, you can spawn models in Gazebo in different poses when launching the spawn_model process. Change [these lines](https://github.com/Interbotix/interbotix_ros_manipulators/blob/27e76594d06e8fbc98262c424dd05732fe328f87/interbotix_ros_xsarms/interbotix_xsarm_gazebo/launch/xsarm_gazebo.launch#L100-L107) to the block below. Note the `-z` flag followed by the desired spawn z-position of 1.015m.
```xml
  <node
    name="urdf_spawner"
    pkg="gazebo_ros"
    type="spawn_model"
    respawn="false"
    output="screen"
    ns="$(arg robot_name)"
    args="-urdf -model $(arg robot_model) -param robot_description -z 1.015"/>
```

You should also revert your z origin changes in your URDF.

closed 2022-06-23 by sbhambr1