Interbotix/interbotix_ros_manipulators #39: [Question]: Actual robot not working with MoveIt Python Interface and PATH_TOLERANCE_VIOLATED warning.

sbhambr1, 2022-07-07:
### Question

Hi,
I am trying to use moveit (and Gazebo) using a python interface. I am successfully able to move the arm in the gazebo world, and the goal state is shown in MoveIt-RViz, but I receive the following warning (see attached terminal screenshot) and the real robot does not move. (see MoveIt and Gazebo screenshots). The following were obtained by planning for the robot to reach the 'Upright' pose.
[image]
[image]
[image]
Note: the MoveIt-RViz also shows the state of the actual robot.

### Robot Model

px100

### Operating System

Ubuntu 20.04

### ROS Version

ROS1 Noetic

sbhambr1, 2022-07-07:
After a few seconds in idle state, this happens in the Gazebo world simulation.
[image]
I am not sure what is causing this error everytime.

LSinterbotix, 2022-07-29:
I am unable to reproduce your first issue. Could you please provide more information on your Python interface? Is it possible that you are generating trajectories with an incorrect starting state?

I am not sure about the issue you face with Gazebo. I would guess that this is happening because the robot is not included in your world description file since you are spawning it in and calling the `reset_world` service may just reset the robot to the world's origin.

LSinterbotix, 2022-12-29:
Since this issue is not directly related to the Interbotix packages, it will be closed. I suggest you look for an answer on ROS Answers or the Gazebo forums if you have not already solved it.

closed 2022-12-29 by LSinterbotix