Interbotix/interbotix_ros_manipulators #40: [Question]: ROS software installation problem

helenvyle, 2022-07-19:
### Question

I've tried following the [setup guide](https://www.trossenrobotics.com/docs/interbotix_xsarms/ros_interface/software_setup.html) for installing the Trossen packages. I directed the install to a custom workspace, not the default interbotix_ws one. The install script ran successfully and said that it completed, however I can't find any of the interbotix packages when I run rospack list, nor can I find them in the workspace itself. I also can't run any of the example launches in the packages as ROS can't find them. 
I've already restarted the system and sourced my environments. 

Thanks for the help!

### Operating System

Ubuntu 20.04

### ROS Version

ROS1 Noetic

LSinterbotix, 2022-07-21:
What errors are you seeing when trying to run launch files?

The installation might not have caught a build error - try to run `catkin_make` in the workspace to see if it works properly.

helenvyle, 2022-07-21:
Just that neither the launch nor package could be found. I did run catkin_make and it built successfully, though I didn't note if it built the interbotix packages or not.
Anyway, I reinstalled and directed it to the default interbotix_ws workspace instead and that worked fine. There must have been an extra step with installing to a custom workspace that I missed.

closed 2022-07-21 by helenvyle