Interbotix/interbotix_ros_manipulators #44: [Question]: error:RLException: ERROR: unable to contact ROS master at [http://hostname.local:11311] The traceback for the exception was written to the log file

flamboyantflower, 2022-08-05:
### Question

when  I try to run 
roslaunch interbotix_xsarm_gazebo xsarm_gazebo.launch robot_model:=wx250 use_trajectory_controllers:=true

I get error 
RLException: ERROR: unable to contact ROS master at [http://hostname.local:11311]
The traceback for the exception was written to the log file

### Robot Model

rx150

### Operating System

Ubuntu 20.04

### ROS Version

ROS1 Noetic

LSinterbotix, 2022-08-05:
Looks like a bad ROS network setup. I see that your ROS Master is set to `hostname` — did you change this, or is your computer's name `hostname`?

If your machine is _not_ named `hostname` and you're running this on a local machine, edit your `~/.bashrc` file to export the `ROS_MASTER_URI` environment variable as below:

```bash
export ROS_MASTER_URI=http://localhost:11311
```

See the [ROS Network Setup Guide](https://wiki.ros.org/ROS/NetworkSetup) for more details.

flamboyantflower, 2022-08-05:
That export line was already in the bash file, I tried deleting and undeleting it hasnt worked

flamboyantflower, 2022-08-05:
Hey, so I got gazebo to run and robot appears in simulation but it gives a whole list of errors.

[ERROR] [1659738789.368240259]: Could not load controller 'arm_controller' because controller type 'effort_controllers/JointTrajectoryController' does not exist.
[ERROR] [1659738789.368266759]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
Error when loading 'arm_controller'
[ERROR] [1659738789.369746200]: Could not load controller 'gripper_controller' because controller type 'effort_controllers/JointTrajectoryController' does not exist.
[ERROR] [1659738789.369761675]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
Error when loading 'gripper_controller'
Loaded 'joint_state_controller'
[ERROR] [1659738789.375448151]: Could not start controller with name 'arm_controller' because no controller with this name exists
Error when starting ['arm_controller'] and stopping []
[ERROR] [1659738789.376621500]: Could not start controller with name 'gripper_controller' because no controller with this name exists

LSinterbotix, 2022-08-05:
> Solved: you have to comment out the export line AND restart the computer.

You only need to close/reopen the terminal or re-source when you make changes to the `~/.bashrc` file.

What roslaunch command did you use? Are you specifying a controller type i.e. `use_trajectory_controllers:=true`?

flamboyantflower, 2022-08-05:
I used this command 
roslaunch interbotix_xsarm_gazebo xsarm_gazebo.launch robot_model:=wx250 use_trajectory_controllers:=true

flamboyantflower, 2022-08-05:
Now I am trying to launch the robot in the rviz simulator with 
roslaunch interbotix_xsarm_control xsarm_control.launch robot_model:rx150 use_sim:=true

I am getting this back, I thought rx150 was a model?
Usage: roslaunch [options] [package] <filename> [arg_name:=value...]
       roslaunch [options] <filename> [<filename>...] [arg_name:=value...]

If <filename> is a single dash ('-'), launch XML is read from standard input.

roslaunch: error: The following input files do not exist: robot_model:rx150

LSinterbotix, 2022-08-08:
> I see that you guys don't have a python api for gazebo from another question, is this still true?

Yes, this is still true.

> roslaunch interbotix_xsarm_control xsarm_control.launch robot_model:rx150 use_sim:=true

You have a syntax error in your command. You have `robot_model:rx150`, it should be `robot_model:=rx150`.

LSinterbotix, 2022-08-08:
It's possible that you don't have the joint trajectory controller package installed. Run the following command to check if it is on your machine:

`rospack list | grep joint_trajectory_controller`

If the output of the above command is empty, install the package by running:

`sudo apt-get install ros-noetic-joint-trajectory-controller`

flamboyantflower, 2022-08-08:
Tried your solution, am getting
 python3-rosdep-modules : Depends: python3-rospkg-modules (>= 1.4.0) but it is not going to be installed
                          Depends: python3-catkin-pkg-modules (>= 0.4.0) but it is not going to be installed
 python3-rosdistro-modules : Depends: python3-catkin-pkg-modules but it is not going to be installed
                             Depends: python3-rospkg-modules but it is not going to be installed
 ros-noetic-rospack : Depends: python3-catkin-pkg-modules but it is not going to be installed
 ros-noetic-rqt-gui : Depends: python3-rospkg-modules but it is not going to be installed
 ros-noetic-rqt-robot-monitor : Depends: python3-rospkg-modules but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).

LSinterbotix, 2022-08-10:
Closing in favor of #45.

closed 2022-08-10 by LSinterbotix