Interbotix/interbotix_ros_manipulators #46: [Question]: Where do I put a python script file in order to execute it?

flamboyantflower, 2022-08-11:
### Question

So I am now able to send position commands to my gazebo simulated rx150, I would like to send position commands via a python script now. I wrote the script, and execute it in the terminal.

input to terminal:
rosrun rx150_control main.py

output:
[INFO] [1660247577.909809, 0.000000]: 1.0

but the simulated robot does not act on the command.

Is my script in the wrong location of the file tree?

### Robot Model

rx150

### Operating System

Ubuntu 20.04

### ROS Version

ROS1 Noetic

LSinterbotix, 2022-08-11:
It doesn't matter where you keep the Python script as long as you can execute it using some method (which you seem to be able to do using `rosrun`).

I would make sure that you're publishing to the correct topic. Use the `rqt_graph` tool or the `rostopic` command to check if your node is connected to the right topics coming from Gazebo.

I would be happy to help further if you provide more details about your script and what you're trying to do with it.

flamboyantflower, 2022-08-11:
hey thanks for looking at my question
here is the script
#!/usr/bin/env python3
import rospy
from std_msgs.msg import String

import rospy
from std_msgs.msg import Float64

def talker():
	pub = rospy.Publisher('/rx150/wrist_rotate_controller/command', Float64, queue_size=10)
	rospy.init_node('rx150_talker', anonymous=True)
	rate = rospy.Rate(10)
	while not rospy.is_shutdown():
		position = -1.0
		rospy.loginfo(position)
		pub.publish(position)
		rate.sleep()
		
if __name__ == '__main__':
	try:
		talker()
	except rospy.ROSInterruptException:
		pass

LSinterbotix, 2022-08-11:
Your script works for me. I modified it to cycle the wrist_rotate position command between 1.0 and -1.0 and got the expected result below:

[image]

[image]

Is your Gazebo unpaused when you're running this script? Are you setting `robot_name` when launching the xsarm_gazebo package, or leaving it as the default (`robot_model`)?

flamboyantflower, 2022-08-11:
I left the default robot model when launching and yes I did run rosservice call /gazebo/unpause_physics

flamboyantflower, 2022-08-12:
fixed, had to change code from wrist_rotate_controller to wrist_angle_controller

closed 2022-08-12 by LSinterbotix