Interbotix/interbotix_ros_manipulators #47: [Bug]: error during installation

cyberchucktx, 2022-08-12:
### What happened?

Am getting regular installation errors by following the Trossen Installation guide at https://www.trossenrobotics.com/docs/interbotix_xsarms/ros_interface/software_setup.html. 

I am trying to install on Ubuntu 22.04 (yes, I know it says 20.04). I modified the installation script a bit (added "22.04" as an option) but am still getting the error(s) on the AWENT section. 

Is there any way to install on 22.04? Is there a way to compile from source? 

### Robot Model

X250 manipulator

### Operating System

Ubuntu 22.04

### ROS Distro

ROS2 Galactic

### Steps To Reproduce

Run the installation process/sript. 

### Relevant log output

```shell
Starting >>> interbotix_common_sim
--- stderr: dynamixel_workbench_toolbox
CMake Error at CMakeLists.txt:18 (find_package):
  By not providing "Findament_cmake.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "ament_cmake", but CMake did not find one.
  Could not find a package configuration file provided by "ament_cmake" with
  any of the following names:
    ament_cmakeConfig.cmake
    ament_cmake-config.cmake
  Add the installation prefix of "ament_cmake" to CMAKE_PREFIX_PATH or set
  "ament_cmake_DIR" to a directory containing one of the above files.  If
  "ament_cmake" provides a separate development package or SDK, be sure it
  has been installed.
---
Failed   <<< dynamixel_workbench_toolbox [0.67s, exited with code 1]
Aborted  <<< interbotix_xs_msgs [0.66s]
Aborted  <<< interbotix_common_sim [0.99s]
Aborted  <<< interbotix_common_modules [1.06s]
Summary: 0 packages finished [1.31s]
  1 package failed: dynamixel_workbench_toolbox
  3 packages aborted: interbotix_common_modules interbotix_common_sim 
interbotix_xs_msgs
  3 packages had stderr output: dynamixel_workbench_toolbox 
interbotix_common_sim interbotix_xs_msgs
  20 packages not processed
[ERROR] Failed to build Interbotix Arm ROS Packages.
[ERROR] Interbotix Installation Failed!
```

### Anything Else

Nothing else at present

LSinterbotix, 2022-08-12:
```cmake
  By not providing "Findament_cmake.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "ament_cmake", but CMake did not find one.
```

This error leads me to think that something went wrong when installing ROS 2 Galactic or that sourcing the Galactic setup script didn't work properly. The `ament_cmake` package is provided by the ROS 2 standard install and you should have it on your CMake path if everything worked properly.

You may be able to go step-by-step through the script. I would pay close attention to the ROS 2 section. Because I don't know what went wrong when installing ROS 2, I can't help much without more details.

Another way might be to install Humble instead of Galactic, though we haven't tested our packages on that distro yet.

Because we do not support either ROS 2 Humble or Ubunutu 22.04, this issue will be closed. Feel free to continue to ask questions or reopen if necessary.

closed 2022-08-12 by LSinterbotix

cyberchucktx, 2022-08-12:
What is "Humble"? If that is an ROS distro is isn't listed in the installation guide(s) I've found so far. 

I guess if I *have* to I can just wipe the laptop and install Ubuntu 20.04. I'd rather *not* but if that is the only way, then .... 

I'll try a step-by-step. 

As I mentioned, I had to do some basic editing on the install script anyway to allow 22.04 (it checks for Ubuntu version).

LSinterbotix, 2022-08-12:
Yes, [Humble](https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html) is the latest ROS 2 distribution and targets Ubuntu 22.04.

cyberchucktx, 2022-08-12:
COOL! I'll be a "Beta Tester" ... how do I get the installation script for "Humble"? 

As mentioned, I got the Trossen Installation guide at

 https://www.trossenrobotics.com/docs/interbotix_xsarms/ros_interface/software_setup.html.

cyberchucktx, 2022-08-12:
NM (Never Mind). I followed the HUMBLE link in your response and got to the installation page(s). 

I'm new to the ROS website & world ... wouldn't have been able to easily find the Humble stuff. 

Will try using these new installation instructions for HUMBLE today on my 22.04 installation. 

THANKS! I bet it works now ..