Interbotix/interbotix_ros_manipulators #49: [Question]: When i install  the X-Series Arm packages , a problem occurred: [ERROR] Something went wrong. [ERROR] Interbotix Installation Failed!

stuliujiayu, 2022-10-06:
### Question

ROS version: Melodic
Ubuntu version:18.04
 When i install  the X-Series Arm packages following the scripe below:
``` 
 sudo apt install curl
 curl 'https://raw.githubusercontent.com/Interbotix/interbotix_ros_manipulators/main/interbotix_ros_xsarms/install/amd64/xsarm_amd64_install.sh' > xsarm_amd64_install.sh
 chmod +x xsarm_amd64_install.sh
 ./xsarm_amd64_install.sh melodic
```
A problem occurred:
```
Creating config file /etc/ssh/sshd_config with new version
Creating SSH2 RSA key; this may take some time ...
2048 SHA256:y+Edo13n2uc6iu15cYNmqST8VAhVWMSfW8/zoA6Hmsc root@leona (RSA)
Creating SSH2 ECDSA key; this may take some time ...
256 SHA256:mKkvuV+BXq1iQchGgynp60NGPCQCIyoAEGFHaCny/nw root@leona (ECDSA)
Creating SSH2 ED25519 key; this may take some time ...
256 SHA256:nsdTkm5Az4hnEF5KHkw6tn8Iiz/lUzrTwb6e8Lm6SHk root@leona (ED25519)
Created symlink /etc/systemd/system/sshd.service → /lib/systemd/system/ssh.service.
Created symlink /etc/systemd/system/multi-user.target.wants/ssh.service → /lib/systemd/system/ssh.service.
Processing triggers for man-db (2.8.3-2ubuntu0.1) ...
Processing triggers for ufw (0.36-0ubuntu0.18.04.2) ...
Processing triggers for ureadahead (0.100.0-21) ...
ureadahead will be reprofiled on next reboot
Processing triggers for systemd (237-3ubuntu10.56) ...
./xsarm_amd64_install.sh: line 159: [: ==: unary operator expected
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python-pip is already the newest version (9.0.1-2.3~ubuntu1.18.04.5).
0 to upgrade, 0 to newly install, 0 to remove and 0 not to upgrade.
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://mirrors.ustc.edu.cn/pypi/web/simple
Collecting modern_robotics
  Downloading https://mirrors.bfsu.edu.cn/pypi/web/packages/a0/78/18a10bb636b0f1ae101efdb8dfe65f5dce231e00de95ae1a14af32c49249/modern_robotics-1.1.0.tar.gz (14 kB)
Requirement already satisfied: numpy in /usr/lib/python2.7/dist-packages (from modern_robotics) (1.13.3)
Building wheels for collected packages: modern-robotics
  Building wheel for modern-robotics (setup.py) ... done
  Created wheel for modern-robotics: filename=modern_robotics-1.1.0-py2-none-any.whl size=16550 sha256=c3643b0781a507d2701b4f409a061d79223b738a32a1ff07ccd8dfb7d1900fa2
  Stored in directory: /home/ljy/.cache/pip/wheels/6e/58/d5/016df3eece29c0560d3cb0c886499ea5e990a68987babdd239
Successfully built modern-robotics
Installing collected packages: modern-robotics
Successfully installed modern-robotics-1.1.0
[ERROR] Something went wrong.
[ERROR] Interbotix Installation Failed!
```
What's more, I can't find the file named interbotix_ws. 
Could anyone tell me why? Thanks!!

### Operating System

Ubuntu 18.04

### ROS Version

ROS1 Melodic

LSinterbotix, 2022-10-06:
To install the Interbotix Arm packages for Melodic, you can either provide no arguments while running on Ubuntu 18.04, or specify the distribution using the `-d` flag, i.e.  
```
./xsarm_amd64_install.sh -d melodic
```

closed 2022-10-06 by LSinterbotix

stuliujiayu, 2022-10-07:
Very helpful. Thank you!!