Interbotix/interbotix_ros_manipulators #51: [Bug]:

kirstyellis, 2022-11-03:
### What happened?

I followed the installation instructions, but when I connect the USB cable to my computer, there is nothing to indicate that the connection has been detected, no change in /dev and no log in dmesg. This command returns nothing: ls /dev | grep ttyDXL

When I powered the robot, a red light on each motor flashed once, I am not sure if this is expected.

I also installed Dynamixel wizard and the scan doesn't return any results. 

Any ideas what could be wrong?

### Robot Model

Rx150

### Operating System

Ubuntu 20.04

### ROS Distro

ROS1 Noetic

LSinterbotix, 2022-11-04:
I would first check the [Can't Find DYNAMIXEL ID](https://www.trossenrobotics.com/docs/interbotix_xsarms/troubleshooting/index.html#can-t-find-dynamixel-id) troubleshooting guide. Then see the steps below:

- It's possible that there's something wrong with the connection between the Power Hub and U2D2 or the U2D2 and your computer - check these connections, maybe even see if a different USB->MicroUSB cable fixes the issue.
- Does the U2D2 light up red when connected to your computer?

kirstyellis, 2022-11-04:
Hi @LSinterbotix, thank you for your reply. I only tried with the cable that was shipped with the robot, I will try another. When I connect the robot to my computer there is no LED action on the U2D2.

LSinterbotix, 2022-11-04:
No LED on the U2D2 would indicate a U2D2 failure or a bad connection to the computer. Let us know if the former is the case - we can ship you a replacement.

kirstyellis, 2022-11-09:
I can confirm that after trying a different USB cable, the light on the U2D2 is now on and I can see the device in /dev.

Thank you for assisting with this!

closed 2022-11-09 by kirstyellis