Interbotix/interbotix_ros_manipulators #53: Perception pipeline not working with RealSense D405

hcdiekmann, 2022-11-09:
### What happened?

Using the Intel D405 RealSense camera gives no image in RViz.

[image]

### Robot Model

px100

### Operating System

Ubuntu 20.04

### ROS Distro

ROS2 Galactic

### Steps To Reproduce

ros2 launch interbotix_xsarm_perception xsarm_perception.launch.py robot_model:=px100 use_pointcloud_tuner_gui:=true use_armtag_tuner_gui:=true

### Anything Else

The interest is to use the D405 for short distance perception (7- 50cm)

hcdiekmann, 2022-11-09:
changing the image topic to [/camera/color/image_rect_raw] displays an image feed, however causes the ArmTag Tuner to crash when "Snap Pose" is pressed.
[image]

LSinterbotix, 2022-11-09:
The D405 may not be publishing image messages to the `camera/color/image_raw` topic. It seems like the ArmTag Tuner GUI may be trying to pull images from a topic that doesn't exist based on the fact that the RViz Image display isn't showing anything and you had to change its topic. I would see if setting the GUI's `camera_color_topic` launch argument to `/camera/color/image_rect_raw` in the xsarm_perception launch file fixes this.

https://github.com/Interbotix/interbotix_ros_manipulators/blob/9cf7353a5901fd07a4af0fa5c4aac0328af4d2eb/interbotix_ros_xsarms/interbotix_xsarm_perception/launch/xsarm_perception.launch.py#L374-L378

hcdiekmann, 2022-11-14:
Changing the the launch argument to `/camera/color/image_rect_raw` has no effect, as seen in the output from the by the picture-snapper node, which still refers to the `camera/color/image_raw` topic.
[image]

hcdiekmann, 2022-11-14:
This is the traceback when pressing the "Snap Pose" nevertheless 
[image]

LSinterbotix, 2022-11-14:
The commits Interbotix/interbotix_ros_toolboxes@ce479a9 and Interbotix/interbotix_ros_toolboxes@21a88d4 should fix the launch argument problems. The commits Interbotix/interbotix_ros_toolboxes@cecfef1 and Interbotix/interbotix_ros_toolboxes@449eb30 should prevent or handle errors you may experience when using the ArmTag Tuner GUI or the Picture Snapper.

LSinterbotix, 2022-11-16:
I'm not able to replicate the issue with the defaults. Could you share the contents of your `xsarm_perception/config/filter_params.yaml` file? Is your x_filter_min `'-0.25'` instead of `-0.25`? Are you manually setting these somewhere other than the config file?

hcdiekmann, 2022-11-16:
The contents of my filter_params.yaml file should all be default, as I did a new clean install of galactic packages and I am not setting them anywhere else. 
`/**:
  ros__parameters:
    cluster_max_size: 1000
    cluster_min_size: 50
    cluster_tol: 0.02
    plane_dist_thresh: 0.005
    plane_max_iter: 50
    ror_min_neighbors: 5
    ror_radius_search: 0.01
    voxel_leaf_size: 0.004
    x_filter_max: 0.25
    x_filter_min: -0.25
    y_filter_max: 0.25
    y_filter_min: -0.25
    z_filter_max: 0.75
    z_filter_min: 0.25`

hcdiekmann, 2022-11-16:
I tried reinstalling with the install script from both, main and galactic branch. Unfortunately the error remains.
The picture below shows my python ros packages versions, rclpy being 1.9.2 .

[image]

hcdiekmann, 2022-11-17:
No CropBox or pc_markers are visible in rviz either. Inspecting the camera node with rqt also shows that no data is published to the `/camera/depth/color/points` topic. However when I launch the realsense2_camera package independently with 
`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=1280x720x30 pointcloud.enable:=true` 
, the node does publish point_cloud data to that topic.
[image]

LSinterbotix, 2022-11-19:
The pointcloud issue should be fixed in 1c40529dd5d673f5eef924970f37a0d5d82d7c6c.
The parameter type error may be fixed in Interbotix/interbotix_ros_toolboxes@a41e8b8fdf77fddd118ef3a40a4aa77ac39fc4ac.

The Cropbox and other perception displays will not show up if there is no TF between the robot and the camera - i.e. that you have successfully run the ArmTag calibration. I assume that you have already done this. These will also not show up if the pipeline does not receive any pointcloud messages to trigger the callback - I believe that this is your issue. Try applying the updates mentioned above to see if this fixes it.

To the filter parameter issue - I'm still unsure why you're getting these errors, but I added hard type conversions to the get/set methods anyways. This may fix the InvalidParameterTypeException.

hcdiekmann, 2022-11-23:
Thanks, the pointcloud data and cropbox is now displayed. However only when launching with the `use_pointcloud_tuner_gui:=true ` argument. The pc tuner GUI then still exits with an InvalidParameterTypeException. Interestingly, when I remove the `filter_params.yaml ` file, the GUI is displayed, but all parameters are shown with a comma as decimal separator instead of a dot.

hcdiekmann, 2022-12-20:
The InvalidParameterException when reading the `filter_params.yaml` file is resolved by changing the numeric locale to en_US.UTF-8.
``` bash
sudo locale-gen en_US en_US.UTF-8
sudo update-locale LC_NUMERIC=en_US.UTF-8
export LC_NUMERIC=en_US.UTF-8
```

closed 2022-12-20 by hcdiekmann