Interbotix/interbotix_ros_manipulators #54: [Enhancement]: close range perception with camera mounted near base

hcdiekmann, 2022-11-14:
### Describe the solution you'd like

The specifications of the [RealSense D405 ](https://www.intelrealsense.com/depth-camera-d405/)allow for depth sensing from 7cm up 50cm, in a smaller form factor. This makes it optimal for mounting the camera near the robot arm base when the arm is mounted on a platform or AMR. It would therefor be useful to get the camera_link TF with the AprilTag mounted on the inside of the arm. 

### Platform

xarms

### Suggested code

Changing the urdf.xacro file of the ar_tag

LSinterbotix, 2022-11-14:
Feel free to fork this repo and change the ar_tag URDF yourself according to how you modify your arm and AR tag mount, specifically these lines:
 https://github.com/Interbotix/interbotix_ros_manipulators/blob/681faeafb7ffe94f2e5b0c5e1a8e5dffea81f915/interbotix_ros_xsarms/interbotix_xsarm_descriptions/urdf/ar_tag.urdf.xacro#L7-L16

Let me know if you have and questions on how to modify your URDF files.

closed 2022-11-14 by LSinterbotix

hcdiekmann, 2022-11-15:
I adjusted the URDF of the AR tag and added a custom STL for the AR tag. The TF of the camera_link seems to be in the correct direction, but a bit to far from the tag link. I assume this is caused by the smaller size of the AR Tag? 
I changed the tag size from the original 47x47 to 30x30mm to prevent collision with the forearm servo. Is there a way to accommodate for this size change in the ArmTag_tuner code?

[image]

hcdiekmann, 2022-11-15:
In the AprilTag user guide I see the option to the add a [tag size](https://github.com/AprilRobotics/apriltag/wiki/AprilTag-User-Guide#pose-estimation) parameter. Where would I add that?

LSinterbotix, 2022-11-15:
The tag's size parameter is configured in this file: https://github.com/Interbotix/interbotix_ros_toolboxes/blob/galactic/interbotix_perception_toolbox/interbotix_perception_modules/config/tags.yaml

and given to the node in this launch file:
https://github.com/Interbotix/interbotix_ros_toolboxes/blob/galactic/interbotix_perception_toolbox/interbotix_perception_modules/launch/apriltag.launch.py

hcdiekmann, 2022-11-16:
Thank you. Is the `size` parameter an arbitrary unit or which unit of measurement? I cant find much about this in the AprilTag documentation.

LSinterbotix, 2022-11-16:
The `size` parameter is in meters.