Interbotix/interbotix_ros_manipulators #55: [Bug]: No planning Library loaded

hcdiekmann, 2022-11-17:
### What happened?

When launching the MoveIt package, no planning library is loaded in RViz and no robot model is shown. 
[image]

### Robot Model

px100

### Operating System

Ubuntu 20.04

### ROS Distro

ROS2 Galactic

### Steps To Reproduce

ros2 launch interbotix_xsarm_moveit xsarm_moveit.launch.py robot_model:=px100 hardware_type:=actual

### Relevant log output

```shell
[INFO] [launch]: All log files can be found below /home/aaeon/.ros/log/2022-11-17-11-22-47-883840-aaeon-UPX-TGL01-17749
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [17758]
[INFO] [rviz2-2]: process started with pid [17760]
[INFO] [ros2_control_node-3]: process started with pid [17762]
[INFO] [spawner-4]: process started with pid [17764]
[INFO] [spawner-5]: process started with pid [17767]
[INFO] [xs_sdk-6]: process started with pid [17769]
[INFO] [robot_state_publisher-7]: process started with pid [17780]
[move_group-1] [WARN] [1668680568.904112951] [move_group.move_group]: MoveGroup launched without ~default_planning_pipeline specifying the namespace for the default planning pipeline configuration
[move_group-1] [WARN] [1668680568.905039455] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-1] [INFO] [1668680568.912702955] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00315058 seconds
[move_group-1] [INFO] [1668680568.912751817] [moveit_robot_model.robot_model]: Loading robot model 'px100'...
[move_group-1] [INFO] [1668680568.912767520] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[xs_sdk-6] [INFO] Using Interbotix X-Series Driver Version: 'v0.3.3'.
[xs_sdk-6] [INFO] Using logging level 'INFO'.
[xs_sdk-6] [INFO] Loaded mode configs from '/home/aaeon/interbotix_ws/install/interbotix_xsarm_moveit/share/interbotix_xsarm_moveit/config/modes.yaml'.
[xs_sdk-6] [INFO] Loaded motor configs from '/home/aaeon/interbotix_ws/install/interbotix_xsarm_control/share/interbotix_xsarm_control/config/px100.yaml'.
[xs_sdk-6] [INFO] Pinging all motors specified in the motor_config file. (Attempt 1/3)
[move_group-1] Link px100/base_link had 1 children
[move_group-1] Link px100/shoulder_link had 1 children
[move_group-1] Link px100/upper_arm_link had 1 children
[move_group-1] Link px100/forearm_link had 1 children
[move_group-1] Link px100/gripper_link had 1 children
[move_group-1] Link px100/ee_arm_link had 2 children
[move_group-1] Link px100/gripper_prop_link had 0 children
[move_group-1] Link px100/gripper_bar_link had 1 children
[move_group-1] Link px100/fingers_link had 3 children
[move_group-1] Link px100/ee_gripper_link had 0 children
[move_group-1] Link px100/left_finger_link had 0 children
[move_group-1] Link px100/right_finger_link had 0 children
[xs_sdk-6] [INFO] 	Found DYNAMIXEL ID:  4, Model: 'XL430-W250', Joint Name: 'wrist_angle'.
[move_group-1] [INFO] [1668680568.993406975] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-1] [INFO] [1668680568.993710937] [moveit.ros_planning_interface.moveit_cpp]: Listening to '/px100/joint_states' for joint states
[move_group-1] [INFO] [1668680568.994400669] [moveit_ros.current_state_monitor]: Listening to joint states on topic '/px100/joint_states'
[move_group-1] [INFO] [1668680568.995002777] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[move_group-1] [INFO] [1668680568.995043452] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-1] [INFO] [1668680568.996053022] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[move_group-1] [INFO] [1668680568.996083576] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-1] [INFO] [1668680568.996659001] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-1] [INFO] [1668680568.997182819] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-1] [WARN] [1668680568.997622414] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[move_group-1] [ERROR] [1668680568.997655254] [moveit.ros.occupancy_map_monitor.middleware_handle]: No sensor plugin specified for octomap updater ; ignoring.
[move_group-1] [INFO] [1668680568.997672957] [moveit.ros.occupancy_map_monitor.middleware_handle]: Skipping octomap updater plugin ''
[move_group-1] [INFO] [1668680569.001161246] [moveit.ros_planning_interface.moveit_cpp]: Loading planning pipeline 'move_group'
[xs_sdk-6] [INFO] 	Found DYNAMIXEL ID:  3, Model: 'XL430-W250', Joint Name: 'elbow'.
[... 59 lines cut ...]
[xs_sdk-6] [INFO] The operating mode for the 'arm' group was changed to 'position' with profile type 'velocity'.
[xs_sdk-6] [INFO] The operating mode for the 'gripper' joint was changed to 'linear_position' with profile type 'velocity'.
[xs_sdk-6] [WARN] Could not get 'Goal_Current' Item Info. This message can be ignored if none of the robot's motors support current control.
[xs_sdk-6] [INFO] Interbotix X-Series Driver is up!
[ros2_control_node-3] [INFO] [1668680569.899574147] [px100.xs_hardware_interface]: Joint states received.
[ros2_control_node-3] [INFO] [1668680569.909743900] [px100.controller_manager]: update rate is 10 Hz
[ros2_control_node-3] [INFO] [1668680569.910490916] [px100.controller_manager]: RT kernel is recommended for better performance
[ros2_control_node-3] [INFO] [1668680569.944569276] [px100.controller_manager]: Loading controller 'gripper_controller'
[xs_sdk-6] [INFO] [1668680569.992436158] [interbotix_xs_sdk.xs_sdk]: InterbotixRobotXS is up!
[spawner-5] [INFO] [1668680570.012059753] [px100.gripper_controller_spawner]: Loaded gripper_controller
[ros2_control_node-3] [INFO] [1668680570.015211254] [px100.controller_manager]: Loading controller 'arm_controller'
[ros2_control_node-3] [INFO] [1668680570.111051881] [px100.controller_manager]: Configuring controller 'gripper_controller'
[ros2_control_node-3] [INFO] [1668680570.112088957] [px100.gripper_controller]: Command interfaces are [position] and and state interfaces are [position].
[spawner-4] [INFO] [1668680570.112449330] [px100.arm_controller_spawner]: Loaded arm_controller
[ros2_control_node-3] [INFO] [1668680570.115103536] [px100.gripper_controller]: Controller state will be published at 50.00 Hz.
[ros2_control_node-3] [INFO] [1668680570.117438723] [px100.gripper_controller]: Action status changes will be monitored at 20.00 Hz.
[ros2_control_node-3] [INFO] [1668680570.124911132] [px100.controller_manager]: Configuring controller 'arm_controller'
[ros2_control_node-3] [INFO] [1668680570.125032145] [px100.arm_controller]: Command interfaces are [position] and and state interfaces are [position].
[ros2_control_node-3] [INFO] [1668680570.125697364] [px100.arm_controller]: Controller state will be published at 50.00 Hz.
[ros2_control_node-3] [INFO] [1668680570.126482616] [px100.arm_controller]: Action status changes will be monitored at 20.00 Hz.
[spawner-4] [INFO] [1668680570.312600840] [px100.arm_controller_spawner]: Configured and started arm_controller
[spawner-5] [INFO] [1668680570.512254823] [px100.gripper_controller_spawner]: Configured and started gripper_controller
[INFO] [spawner-4]: process has finished cleanly [pid 17764]
[INFO] [spawner-5]: process has finished cleanly [pid 17767]
```

LSinterbotix, 2022-12-29:
This issue does not occur on a fresh install. Was this resolved after changing your locale, as mentioned in https://github.com/Interbotix/interbotix_ros_manipulators/issues/53#issuecomment-1359276795?

hcdiekmann, 2023-01-03:
Yes, changing the locale as stated in https://github.com/Interbotix/interbotix_ros_manipulators/issues/53#issuecomment-1359276795 fixes this.

closed 2023-01-03 by hcdiekmann