Interbotix/interbotix_ros_manipulators #56: [Question]: Waist joint works in rviz but not on pysical robot

flamboyantflower, 2022-12-14:
### Question

I am trying to use the joint current control python file on the robot, shoulder and elbow joints work but not waist joint. not sure why.

### Robot Model

rx150

### Operating System

Ubuntu 20.04

### ROS Version

ROS1 Melodic

LSinterbotix, 2022-12-14:
Could you provide some more information like console output, the contents of your script, your launch command(s), etc.?

flamboyantflower, 2022-12-14:
here is script
from interbotix_xs_modules.arm import InterbotixManipulatorXS

# This script commands currents [mA] to the arm joints
#
# To get started, open a terminal and type 'roslaunch interbotix_xsarm_control xsarm_control.launch robot_model:=vx250'
# Then change to this directory and type 'python joint_current_control.py'

def main():
    joint_currents = [300, 0, 500, 0, 0]
    bot = InterbotixManipulatorXS(robot_model="rx150")
    bot.dxl.robot_set_operating_modes("group", "arm", "current")
    bot.dxl.robot_write_commands("arm", joint_currents)
    x = bot.dxl.robot_get_joint_states()
    print("x is", x)

if __name__=='__main__':
    main()

flamboyantflower, 2022-12-14:
here is my terminal command
roslaunch interbotix_xsarm_control xsarm_control.launch robot_model:=rx150 
and python file launch
python3 joint_current_control.py

flamboyantflower, 2022-12-15:
console out put is only [INFO] [1671132111.323780]: [xs_sdk] The operating mode for the 'arm' group was changed to current.

flamboyantflower, 2022-12-16:
I'm sending currents to the waist motor, the robot behaves as predicted in the rviz simulation but, the physical robot's waist motor is not responding, to the currents I send it. If I use use_sim:= true in the terminal the simulation works, if I don't use that line the rviz simulation and physical robot behave the the same, just moving elbow and shoulder joint but not waist joint.

LSinterbotix, 2022-12-16:
Are you sure the waist joint isn't getting a current command? Is the arm's cradle or another part of the robot blocking its movement?

You can read the waist joint's effort (current) with the following command:

```
$ rostopic echo /<robot_name>/joint_states/effort[0]
303.9700012207031
```

Note that the default joint_current_control.py script doesn't have a command for the waist because it wouldn't move due to the cradle being in the way. The reason why the waist moves in the DYNAMIXEL simulator is because it doesn't take collisions into account and the collision between the cradle and the arm is not considered.

flamboyantflower, 2022-12-19:
works now, is there a way to make the robot less collision sensitive?

LSinterbotix, 2022-12-19:
Great to hear you got it working!

The robot isn't exactly "sensitive" to collisions since it doesn't take collisions into account when moving or planning (if not using MoveIt). It also does not have any built-in collision detection and will try to achieve the commanded goal up to a certain point. Could you elaborate on what you are looking to do?

closed 2023-01-03 by LSinterbotix