Interbotix/interbotix_ros_manipulators #57: [Question]: Robot model parameter not found! Did you remap 'robot_description'?

YXGuan, 2022-12-14:
### Question

Got errors when running: 

`roslaunch interbotix_xsarm_moveit_interface xsarm_moveit_interface.launch robot_model:=px100 dof:=4`

Errors:

```
[ERROR] [1671028802.387172531]: Robot model parameter not found! Did you remap 'robot_description'?
[ERROR] [1671028802.414057565]: Robot model not loaded
[ERROR] [1671028802.421722550]: Planning scene not configured
[FATAL] [1671028802.421755327]: Unable to configure planning scene monitor
terminate called after throwing an instance of 'std::runtime_error'
  what():  Unable to configure planning scene monitor
[ INFO] [1671028802.469466311]: rviz version 1.14.19
[ INFO] [1671028802.469519355]: compiled against Qt version 5.12.8
[ INFO] [1671028802.469534416]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1671028802.478350490]: Forcing OpenGl version 0.
[ INFO] [1671028802.684999526]: Stereo is NOT SUPPORTED
[ INFO] [1671028802.685062268]: OpenGL device: Mesa Intel(R) UHD Graphics (CML GT2)
[ INFO] [1671028802.685086055]: OpenGl version: 4.6 (GLSL 4.6) limited to GLSL 1.4 on Mesa system.
[px100/move_group-2] process has died [pid 232372, exit code -6, cmd /opt/ros/noetic/lib/moveit_ros_move_group/move_group __name:=move_group __log:=/home/yuxiang/.ros/log/309eda92-7bbd-11ed-ba76-5ffe69aaeb6d/px100-move_group-2.log].
log file: /home/yuxiang/.ros/log/309eda92-7bbd-11ed-ba76-5ffe69aaeb6d/px100-move_group-2*.log
[ERROR] [1671028805.986774154]: Robot model parameter not found! Did you remap 'robot_description'?
[ERROR] [1671028805.988701465]: Robot model not loaded
```

It turns out /px100/robot_description is not in the rosparam list
```

yuxiang@yuxiang:~/interbotix_ws/src/interbotix_ros_manipulators/interbotix_ros_xsarms/examples/interbotix_xsarm_moveit_interface$ rosparam list | grep robot_description

/px100/robot_description_kinematics/interbotix_arm/kinematics_solver
/px100/robot_description_kinematics/interbotix_arm/kinematics_solver_search_resolution
/px100/robot_description_kinematics/interbotix_arm/kinematics_solver_timeout
/px100/robot_description_kinematics/interbotix_arm/position_only_ik
/px100/robot_description_planning/default_acceleration_scaling_factor
/px100/robot_description_planning/default_velocity_scaling_factor
/px100/robot_description_planning/joint_limits/elbow/has_acceleration_limits
/px100/robot_description_planning/joint_limits/elbow/has_velocity_limits
/px100/robot_description_planning/joint_limits/elbow/max_acceleration
/px100/robot_description_planning/joint_limits/elbow/max_velocity
/px100/robot_description_planning/joint_limits/left_finger/has_acceleration_limits
/px100/robot_description_planning/joint_limits/left_finger/has_velocity_limits
/px100/robot_description_planning/joint_limits/left_finger/max_acceleration
/px100/robot_description_planning/joint_limits/left_finger/max_velocity
/px100/robot_description_planning/joint_limits/right_finger/has_acceleration_limits
/px100/robot_description_planning/joint_limits/right_finger/has_velocity_limits
/px100/robot_description_planning/joint_limits/right_finger/max_acceleration
/px100/robot_description_planning/joint_limits/right_finger/max_velocity
/px100/robot_description_planning/joint_limits/shoulder/has_acceleration_limits
/px100/robot_description_planning/joint_limits/shoulder/has_velocity_limits
/px100/robot_description_planning/joint_limits/shoulder/max_acceleration
/px100/robot_description_planning/joint_limits/shoulder/max_velocity
/px100/robot_description_planning/joint_limits/waist/has_acceleration_limits
/px100/robot_description_planning/joint_limits/waist/has_velocity_limits
/px100/robot_description_planning/joint_limits/waist/max_acceleration
/px100/robot_description_planning/joint_limits/waist/max_velocity
/px100/robot_description_planning/joint_limits/wrist_angle/has_acceleration_limits
/px100/robot_description_planning/joint_limits/wrist_angle/has_velocity_limits
/px100/robot_description_planning/joint_limits/wrist_angle/max_acceleration

/px100/robot_description_planning/joint_limits/wrist_angle/max_velocity
/px100/robot_description_semantic

```

[image]

Everything else works fine.

Looking forward to hearing from you.

### Operating System

Ubuntu 20.04

### ROS Version

ROS1 Noetic

### Anything Else

robot_model:=px100

YXGuan, 2022-12-14:
similar issue: https://github.com/Interbotix/interbotix_ros_arms/issues/3

LSinterbotix, 2022-12-14:
The `<robot_name>/robot_description` parameter is not loaded if you do not specify a hardware type (`use_fake`, `use_actual`, `use_gazebo`).

You need to set one of these to true (`use_fake:=true`, `use_actual:=true`, or `use_gazebo:=true`) in your launch command:

```console
$ roslaunch interbotix_xsarm_moveit_interface xsarm_moveit_interface.launch robot_model:=px100 dof:=4 use_actual:=true
```

See the [interbotix_xsarm_moveit package usage documentation](https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros1_packages/moveit_motion_planning_configuration.html#usage) for details.

closed 2022-12-14 by LSinterbotix

YXGuan, 2022-12-14:
Awesome. Thanks Luke!