Interbotix/interbotix_ros_manipulators #58: [Question]: Faster moving time

rouzinho, 2022-12-15:
### Question

Hi !

I am trying to make my pincherX150 follow a joint trajectory. I tried using moveit and publish a interbotix_xs_msgs/JointTrajectoryCommand but the robot didn't finish its motion. At the moment, I am using the InterbotixManipulatorXS interface and just use bot.arm.set_joint_positions(joints_cmd) in a loop. The trajectory is accurate and smooth but is very slow. I was wondering if it was possible to speed things up or if you maybe know another way to make the robot follow a trajectory in joints space fast enough.

Thanks for the tips !

### Robot Model

px150

### Operating System

Ubuntu 20.04

### ROS Version

ROS 1 Noetic

LSinterbotix, 2022-12-16:
# Interbotix Python-ROS Interface

You can control the speed of movements using the different [timing parameters](https://docs.trossenrobotics.com/interbotix_xsarms_docs/python_ros_interface.html#timing-parameters). See the [source code](https://github.com/Interbotix/interbotix_ros_toolboxes/blob/main/interbotix_xs_toolbox/interbotix_xs_modules/src/interbotix_xs_modules/arm.py#L134-L145) for more details. Look at the `moving_time` and `accel_time` parameters in each movement command method specifically.

# MoveIt

I'm not sure if there's a way to set movement timing for trajectory executions, but you can control each joint's velocity and acceleration limits in the relevant joint_limits.yaml file:

https://github.com/Interbotix/interbotix_ros_manipulators/blob/2a6c1f5458a1c8ef7571928ce787fa6ad81ec4b8/interbotix_ros_xsarms/interbotix_xsarm_moveit/config/joint_limits/5dof_joint_limits.yaml#L8-L43

Note that you can also change the maximum velocity of the servos in the arm group in the modes.yaml config file:

https://github.com/Interbotix/interbotix_ros_manipulators/blob/2a6c1f5458a1c8ef7571928ce787fa6ad81ec4b8/interbotix_ros_xsarms/interbotix_xsarm_moveit/config/modes.yaml#L1-L7

See the [Profile Velocity](https://emanual.robotis.com/docs/en/dxl/x/xl430-w250/#profile-velocity) and [Profile Acceleration](https://emanual.robotis.com/docs/en/dxl/x/xl430-w250/#profile-acceleration) sections in the servo documentation for details.

rouzinho, 2022-12-16:
Thanks for the answer, I will try that soon !

closed 2022-12-16 by rouzinho