Interbotix/interbotix_ros_manipulators #59: Package 'interbotix_xsarm_descriptions' not found

deepaksinghcv, 2022-12-21:
### What happened?

I have a Viper-X 300 (6dof). I followed the instructions to install ROS 2 Standard Software Setup(https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros_interface/ros2/software_setup.html)

I have successfully installed the packages.
```
...
ros-galactic-desktop-full is already installed!
Apriltag ROS Wrapper already installed!
Interbotix Arm ROS packages already installed!
Environment variables already set!
Installation complete, took 29 seconds in total.
```
I restarted the laptop after the installation.

I'm able to detect the U2D2 interface:
```
ls /dev | grep ttyDXL
ttyDXL
```

But when i execute:
```
$ ros2 launch interbotix_xsarm_descriptions xsarm_description.launch.py robot_model:=vx300s use_joint_pub_gui:=true
```
It is throwing the following error:
```
`$ ros2 launch interbotix_xsarm_descriptions xsarm_description.launch.py robot_model:=vx300s use_joint_pub_gui:=true
Package 'interbotix_xsarm_descriptions' not found: "package 'interbotix_xsarm_descriptions' not found, searching: ['/opt/ros/galactic']"
```

OS info:
Ubuntu 20.04.5 LTS

### Robot Model

vx300s

### Operating System

Ubuntu 20.04

### ROS Distro

ROS 2 Galactic

### Steps To Reproduce

I followed the installation instruction specified at : https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros_interface/ros2/software_setup.html

deepaksinghcv, 2022-12-21:
I also noticed that the build for galactic has failed.

LSinterbotix, 2022-12-21:
Did you source the workspace setup script before trying to run an Interbotix package?

The ROS 2 installation script does not configure your bash terminal to source the setup script for the Interbotix workspace. You need to source it each time you open a new terminal.

```
source ~/interbotix_ws/install/setup.bash
```

deepaksinghcv, 2022-12-24:
The issue is resolved.

Cause 1: I was not able to connect to the hostname raw.githubusercontent.com sometimes. The mentioned installation script(https://raw.githubusercontent.com/Interbotix/interbotix_ros_manipulators/main/interbotix_ros_xsarms/install/amd64/xsarm_amd64_install.sh) has many other resources hosted on that domain. 
Solution 1: I manually added the DNS lookup address in /etc/hosts filename

The installation script needs further checks. Even on partial installation the script prints a positive message stating that the Installation is successful.

Cause 2: After the installation, the port ttyDXL was not being shown.
Solution 2: This was due to faulty microUSB cable; upon changing, the U2D2 interface was detected.

Closing the issue.
Thanks @LSinterbotix

closed 2022-12-24 by deepaksinghcv