Interbotix/interbotix_ros_manipulators #62: [Question]: Eye in hand calibration

YXGuan, 2022-12-29:
### Question

Hi Luke,

**Motivation:**
apply hand-in-eye calibration with MoveIt

**Reproduce procedure:**
1. cd interbotix_ws/src
2. git clone git@github.com:ros-planning/moveit_calibration.git
3.  catkin_make (succeeded without any error)
4.  roslaunch interbotix_xsarm_moveit xsarm_moveit.launch robot_model:=px100 use_actual:=true dof:=4
5. add handeyecalibration gui in RVIZ
[image]

**Issue**
once clicked on adding , Rviz would shut down abruptly. 
```
You can start planning now!

[px100/controller_spawner-3] process has finished cleanly
log file: /home/yuxiang/.ros/log/e30f87dc-87b4-11ed-be52-d1a0a89d201f/px100-controller_spawner-3*.log
[ INFO] [1672352301.818421804]: Loading robot model 'px100'...
[ INFO] [1672352301.818471286]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1672352301.913259206]: Starting planning scene monitor
[ INFO] [1672352301.915449531]: Listening to '/px100/move_group/monitored_planning_scene'
[ INFO] [1672352301.958513467]: Constructing new MoveGroup connection for group 'interbotix_arm' in namespace ''
[ INFO] [1672352303.124808289]: Ready to take commands for planning group interbotix_arm.
[ INFO] [1672352344.638096036]: TFVisualTools Ready.
[ INFO] [1672352344.641796431]: Loading robot model 'px100'...
[ INFO] [1672352344.641824910]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1672352344.673140354]: Loading robot model 'px100'...
[ INFO] [1672352344.673195251]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1672352344.703026765]: Loading robot model 'px100'...
[ INFO] [1672352344.703063587]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1672352344.732081976]: Loading robot model 'px100'...
[ INFO] [1672352344.732122978]: No root/virtual joint specified in SRDF. Assuming fixed joint
[px100/rviz_yuxiang_251668_3751733079435275489-6] process has died [pid 251737, exit code -11, cmd /opt/ros/noetic/lib/rviz/rviz -f world -d /home/yuxiang/interbotix_ws/src/interbotix_ros_manipulators/interbotix_ros_xsarms/interbotix_xsarm_moveit/launch/moveit.rviz __name:=rviz_yuxiang_251668_3751733079435275489 __log:=/home/yuxiang/.ros/log/e30f87dc-87b4-11ed-be52-d1a0a89d201f/px100-rviz_yuxiang_251668_3751733079435275489-6.log].
log file: /home/yuxiang/.ros/log/e30f87dc-87b4-11ed-be52-d1a0a89d201f/px100-rviz_yuxiang_251668_3751733079435275489-6*.log

```

[image]

### Robot Model

px100

### Operating System

Ubuntu 20.04

### ROS Version

ROS 1 Noetic

LSinterbotix, 2022-12-29:
1. If you haven't already, make sure you've installed the required dependencies:
```
cd ~/interbotix_ws && rosdep install -y --from-paths . --ignore-src --rosdistro noetic
```
2. Please provide the RViz logs from the ROS session where you experienced this issue.
3. I'd suggest that you post this issue to the [moveit_calibration](https://github.com/ros-planning/moveit_calibration/issues) repo instead - you might find better help there.

YXGuan, 2022-12-30:
Yes I have already installed all the dependencies required

I Could not find the following log file: /home/yuxiang/.ros/log/e30f87dc-87b4-11ed-be52-d1a0a89d201f/px100-rviz_yuxiang_358426_638358878339853344-6*.log

None of the log files I found are related to RViz. 

[image]

LSinterbotix, 2023-01-03:
This issue seems similar to https://github.com/ros-planning/moveit_calibration/issues/127. I would suggest asking about it there.

Closing this as it is unrelated to the Interbotix packages.

closed 2023-01-03 by LSinterbotix