Interbotix/interbotix_ros_manipulators #63: [Bug]: Object cluster position incorrect when more than one object is found

hcdiekmann, 2023-01-03:
### What happened?

When running the pick_place.py script I get wrong cluster positions when more than one object is placed in the cameras FOV. When only one object is present the position is correct and the script runs successfully.
The two incorrect positions are also exactly the same [x,y,z]  coordinates as shown in RViz below.
[image]

### Robot Model

px100

### Operating System

Ubuntu 20.04

### ROS Distro

ROS 2 Galactic

### Steps To Reproduce

ros2 launch interbotix_xsarm_perception xsarm_perception.launch.py robot_model:=px100
python3 pick_place.py 

### Relevant log output

```shell
[INFO] [1672735566.159837130] [px100.robot_manipulation]: 
        Robot Name: px100
        Robot Model: px100
[INFO] [1672735566.160209831] [px100.robot_manipulation]: Initialized InterbotixRobotXSCore!
[INFO] [1672735566.166547184] [px100.robot_manipulation]: 
        Arm Group Name: arm
        Moving Time: 2.00 seconds
        Acceleration Time: 0.30 seconds
        Drive Mode: Time-Based-Profile
[INFO] [1672735566.167063651] [px100.robot_manipulation]: Initialized InterbotixArmXSInterface!
[INFO] [1672735566.671349422] [px100.robot_manipulation]: 
        Gripper Name: gripper
        Gripper Pressure: 50.0%
[INFO] [1672735566.671799933] [px100.robot_manipulation]: Initialized InterbotixGripperXSInterface!
[INFO] [1672735566.679337653] [px100.robot_manipulation]: FilterParam filepath: '/home/aaeon/interbotix_ws/install/interbotix_perception_modules/share/interbotix_perception_modules/config/filter_params.yaml'
[INFO] [1672735566.701446924] [px100.robot_manipulation]: Initialized InterbotixPointCloudInterface.
[INFO] [1672735567.212264043] [px100.robot_manipulation]: Initialized InterbotixAprilTagInterface!
[INFO] [1672735567.218694289] [px100.robot_manipulation]: Initialized InterbotixArmTagInterface!
-0.11102592947214768 -0.2668080120287048 -0.18490427368824908
[WARN] [1672735570.880857415] [px100.robot_manipulation]: No valid pose could be found. Will not execute
[WARN] [1672735571.004454439] [px100.robot_manipulation]: No valid pose could be found. Will not execute
[WARN] [1672735572.099685035] [px100.robot_manipulation]: No valid pose could be found. Will not execute
-0.11102592947214768 -0.2668080120287048 -0.18490427368824908
[WARN] [1672735572.234343508] [px100.robot_manipulation]: No valid pose could be found. Will not execute
[WARN] [1672735572.423319602] [px100.robot_manipulation]: No valid pose could be found. Will not execute
[WARN] [1672735573.515864356] [px100.robot_manipulation]: No valid pose could be found. Will not execute
```

### Anything Else

Not sure whether this can be seen as a bug/question, as I am  using a modified setup with the camera mounted at the base and the AR tag on the inside of the arm. However this setup is working with identifying and picking up one object.

hcdiekmann, 2023-01-03:
I should mention that I am using the following filter parameters and the object markers in RViz are steady and not flickering.
```yml
/**:
  ros__parameters:
    cluster_max_size: 1000
    cluster_min_size: 152
    cluster_tol: 0.02
    plane_dist_thresh: 0.015
    plane_max_iter: 50
    ror_min_neighbors: 10
    ror_radius_search: 0.01
    voxel_leaf_size: 0.002
    x_filter_max: 0.19999999999999996
    x_filter_min: -0.2
    y_filter_max: 0.19999999999999996
    y_filter_min: -0.2
    z_filter_max: 0.25
    z_filter_min: 0.05
```

hcdiekmann, 2023-01-24:
The problem is in the pointcloud.py file. In the [get_cluster_positions()](https://github.com/Interbotix/interbotix_ros_toolboxes/blob/3130b7d93bc9de52ad092bc1bfbb11162640c292/interbotix_perception_toolbox/interbotix_perception_modules/interbotix_perception_modules/pointcloud.py#L493) function, the for loop which is intended to calculate the average positions of each cluster, is using the first cluster's data to update all other clusters positions. This is because the variable avg_clusters is initialized with a list of ClusterInfo() with the length of the total number of clusters, and then the for loop iterates through all the clusters, updating the position of each cluster in avg_clusters with the position of the current cluster in the iteration. This is incorrect, as each cluster should have its own unique position. 

The solution is to initialize avg_clusters as an empty list and then for each cluster in the first iteration, append a new object of type ClusterInfo() to avg_clusters. 

``` python
        avg_clusters = []
        for i in range(num_clusters):
            avg_clusters.append(ClusterInfo())
        for _ in range(num_samples):
            clusters: List[ClusterInfo] = self.srv_get_cluster_positions.call_async(
                ClusterInfoArray.Request()
            )
            self.spin_until_future_complete(future_get_cluster_pos)
            clusters: List[ClusterInfo] = future_get_cluster_pos.result().clusters
            if len(clusters) != num_clusters:
                self.get_logger().warning(
                    f'Found {len(clusters)} clusters instead of {num_clusters} clusters...'
                )
                return False, []
```

LSinterbotix, 2023-01-26:
Thanks for proposing this fix - this works on our end as well. We would be happy to accept a PR  to the `rolling` branch in the [interbotix_ros_toolboxes](https://github.com/Interbotix/interbotix_ros_toolboxes/pulls) repo if you wish to contribute.

hcdiekmann, 2023-01-31:
I have created a [PR](https://github.com/Interbotix/interbotix_ros_toolboxes/pull/42) in the [interbotix_ros_toolboxes](https://github.com/Interbotix/interbotix_ros_toolboxes/pulls) repo. I will also close this issue with this comment.

closed 2023-01-31 by hcdiekmann