Interbotix/interbotix_ros_manipulators #65: [Question]: ImportError: No module named interbotix_xs_modules.arm.py

Chuckbartow, 2023-01-05:
### Question

Hi.
I am trying to run the bartender.py script on the wx250s model. I am following the guide in the pyhon demos but i get an import error upon trying to execute the script. Neither simulation or real execution on the robot works.

ImportError: No module named interbotix_xs_modules.arm.py

I have checked that the module is inside the interbotix_ros_toolbox
(this path: /interbotix_ws/src/interbotix_ros_toolboxes/interbotix_xs_toolbox/interbotix_xs_modules/src/interbotix_xs_modules)

Could it be some path problems?

### Robot Model

wx250s

### Operating System

Ubuntu 18.04

### ROS Version

ROS 1 Melodic

LSinterbotix, 2023-01-05:
Is your workspace built? Did you source your workspace's setup file before running the example? Check that you get similar results to the command below:

```bash
$ printenv PYTHONPATH
/home/user/interbotix_ws/devel/lib/python2.7/dist-packages:/opt/ros/melodic/lib/python2.7/dist-packages
```

Chuckbartow, 2023-01-05:
Yes i suppose the workspace is build (i did this with catkin build).
I get the following output from the terminal when running the command, so i suppose it is okay?:

user@ubuntu:~$ printenv PYTHONPATH
/opt/ros/melodic/lib/python2.7/dist-packages

LSinterbotix, 2023-01-05:
Your `PYTHONPATH` should also contain `/home/user/interbotix_ws/devel/lib/python2.7/dist-packages`. Source your workspace setup file to append this.

```
$ source ~/interbotix_ws/devel/setup.bash
```

See the ROS Wiki [Installing and Configuring Your ROS Environment](https://wiki.ros.org/action/fullsearch/ROS/Tutorials/InstallingandConfiguringROSEnvironment) for details.

To avoid having to configure your environment every time you open a new terminal, you could append the command above to your `~/.bashrc` file. This is done automatically when using our install scripts.

Chuckbartow, 2023-01-05:
Please see my edit in the last comment, i did source

LSinterbotix, 2023-01-05:
Add these lines to the end of your `~/.bashrc` file, close and reopen your terminal, and try again.

```
source /opt/ros/melodic/setup.bash
source ~/interbotix_ws/devel/setup.bash
```

Chuckbartow, 2023-01-05:
Thank you! This was in fact the problem, as the the source commands were reversed in bashcr, so that /opt/ros/melodic/setup.bash was sourced last.

closed 2023-01-05 by Chuckbartow