Interbotix/interbotix_ros_manipulators #66: [Question]: Max velocity of dynamixel motors on wx250s?

Chuckbartow, 2023-01-05:
### Question

Hello there

What is the maximum value of the Velocity_Limit parameter in the .yaml file for the wx250s?
The default value is set to 131 and i got no clue how this translates into real usage. 

And is this even the right way to make the robot move its joints faster?

### Robot Model

wx250s

### Operating System

Ubuntu 18.04

### ROS Version

ROS 1 Melodic

LSinterbotix, 2023-01-05:
You can see more details about the ROS 1 Interface motor configurations [here](https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros_interface/ros1/config.html#motor-configs). Expand the Motor Configs Template dropdown.

The Velocity_Limit config is:

> Defines the max speed of the motor. A value of '131' corresponds to a max speed of PI rad/s

You can find even more info on the [ROBOTIS eManual page](https://emanual.robotis.com/docs/en/dxl/x/xm430-w350/#velocity-limit) for any servo in the wx250s.

To change the movement timing, you must modify the profile using the modes.yaml config file. See the Mode Configs Template dropdown [here](https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros_interface/ros1/config.html#mode-configs) and the [Operating Mode service definition](https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros_interface/ros1/overview/xs_msgs.html#operatingmodes) for more details.

LSinterbotix, 2023-01-06:
You may also want to look at https://github.com/Interbotix/interbotix_ros_manipulators/issues/58#issuecomment-1353916882.

Chuckbartow, 2023-01-06:
What i am trying to do is a single maneuver at a very fast speed (im trying to throw a ping pong ball), while the rest of the time, the robot should move at moderate speed. 

I have adjusted the profile_velocity and profile_acceleration values to lower values (in time mode) in the modes.yaml file in interbotix_xsarm_control/config/
The idea was to change the general movement time of the robot to be lower in the modes file, then call set_trajectory_time() in my main python file to a more reasonable value. For the fast maneuver, the movement time and accel. time is then set much lower in the joint movement function call.  

By changing the profile type from time to velocity, the robot was much faster, but this also caused ALL movement to be fast instead of the one maneuver.

I have also tried to edit the velocity limits in wx250s.yaml and in the motor_configs_template.yaml.

LSinterbotix, 2023-01-06:
If you're using the Interbotix Python-ROS API you can set an individual movement's time profile by setting the `moving_time`/`accel_time` arguments in any command. You would then just set them back to the original values afterward.

If using anything else, you can call the **xs_sdk**'s [`set_motor_registers`](https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros_interface/ros1/overview/xs_sdk.html#set-register-values-service) service for the `Profile_Velocity`/`Profile_Acceleration` registers before commanding a movement and call it again afterward to reset to their original values.

Note that this is all assuming that you're in the position operating mode with the profile type set to time.

Chuckbartow, 2023-01-06:
Im using the Python-ROS API and changing moving_time and accel_time accordingly. The problem is that theres no real difference bereden using a moving_time of 0.5s and 0.1s, so i am assuming the servos are limited in some other file?

Chuckbartow, 2023-01-09:
Are the servos set to maximum speed at default? I.e. does a velocity limit of 131 PI rad/s correspond to the maximum speed?

LSinterbotix, 2023-01-09:
No, they are set to 3.14 rad/s by default but the velocity limit value can be anything between 0-1023 * 0.229rpm as stated in the Velocity Limit section of the [XM430-W350 eManual page](https://emanual.robotis.com/docs/en/dxl/x/xm430-w350/#velocity-limit). The Velocity_Limit value in the motor configuration file controls this.

Chuckbartow, 2023-01-09:
Just to be sure, then you mean the motor_configs_template.yaml file in the following path?
user@ubuntu:~/interbotix_ws/src/interbotix_ros_core/interbotix_ros_xseries/interbotix_xs_sdk/config/motor_configs_template.yaml.
I have tried to change som values, but nothing seems to work so far. Do i put this file somewhere else in my workspace for it to take effect?

closed 2023-01-26 by LSinterbotix