Interbotix/interbotix_ros_manipulators #67: [Question]: bartender.py on ros2 humble No module named 'interbotix_xs_modules.arm'

sekisushai, 2023-02-14:
### Question

Hi there,
I am trying to run the bartender.py script on the wx250s. I have seen the issue #65 and I am having the same error.

```ImportError: No module named interbotix_xs_modules.arm.py```

In my case I don't have the path `devel` in `~/interbotix_ws`.

Should I use galactic instead of humble ?

### Robot Model

wx250s

### Operating System

Ubuntu 22.04

### ROS Version

ROS 2 Humble

LSinterbotix, 2023-02-14:
Hello @sekisushai. Did you go through the ROS 2 installation process as described [here](https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros_interface/ros2/software_setup.html#amd64-architecture)? Are you sourcing the `~/interbotix_ws/install/setup.bash` file before trying to run any of the Interbotix packages?

sekisushai, 2023-02-20:
Hello @LSinterbotix . Yes I've followed all the step you in the documentation you mentioned.
However, when choosing `humble` for the distro, I don't have the `python_demos` folder. I've retrieved it from github.

closed 2023-02-20 by sekisushai

LSinterbotix, 2023-02-20:
The path to your Python Demos directory 

```
interbotix_ros_manipulators/interbotix_ros_xsarms/examples/python_demos/bartender.py
```

exists only in the ROS 1 branches. In the ROS 2 branches, the path is 

```
https://github.com/Interbotix/interbotix_ros_manipulators/tree/galactic/interbotix_ros_xsarms/interbotix_xsarm_control/demos/python_ros2_api
```

The cause of your issue may have been that you cloned from a ROS 1 branch. See the [Humble bartender script](https://github.com/Interbotix/interbotix_ros_manipulators/blob/humble/interbotix_ros_xsarms/interbotix_xsarm_control/demos/python_ros2_api/bartender.py) for details.