Interbotix/interbotix_ros_manipulators #68: [Question]:  Unable to start simulation in ROS 2 Humble

prateek-ari, 2023-02-28:
### Question

Getting this output while trying to run the simulation example on 
Ubuntu 22.04 ROS 2 Humble
```
[spawner-6] [INFO] [1677586917.742791248] [wx250.joint_state_broadcaster_spawner]: Waiting for '/wx250/wx250/controller_manager' node to exist
[spawner-6] [INFO] [1677586919.748938559] [wx250.joint_state_broadcaster_spawner]: Waiting for '/wx250/wx250/controller_manager' node to exist
```
All dependencies(ros2 control, controllers etc) are installed.

### Operating System

Ubuntu 22.04

### ROS Version

ROS 2 Humble

LSinterbotix, 2023-02-28:
A few things:

1. Could you share the command you're using to launch the simulation?
2. Which guide are you following if any?
3. Could you share the entire contents of your terminal with the issue you're experiencing?

prateek-ari, 2023-02-28:
1.` ros2 launch interbotix_xsarm_sim xsarm_gz_classic.launch.py robot_model:=wx250 verbose:=true` (added verbose so you have more details)
2. Following the software installation procedure here `https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros_interface/ros2/software_setup.html`  
3.  ```
ros2 launch interbotix_xsarm_sim xsarm_gz_classic.launch.py robot_model:=wx250 verbose:=true
[INFO] [launch]: All log files can be found below /home/ty-desktop-22/.ros/log/2023-02-28-21-57-23-624598-ty-desktop-22-5487
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [5489]
[INFO] [gzclient-2]: process started with pid [5491]
[INFO] [spawn_entity.py-3]: process started with pid [5493]
[INFO] [robot_state_publisher-4]: process started with pid [5495]
[INFO] [rviz2-5]: process started with pid [5497]
[gzclient-2] Gazebo multi-robot simulator, version 11.10.2
[gzclient-2] Copyright (C) 2012 Open Source Robotics Foundation.
[gzclient-2] Released under the Apache 2 License.
[gzclient-2] http://gazebosim.org
[gzclient-2] 
[gzserver-1] Gazebo multi-robot simulator, version 11.10.2
[gzserver-1] Copyright (C) 2012 Open Source Robotics Foundation.
[gzserver-1] Released under the Apache 2 License.
[gzserver-1] http://gazebosim.org
[gzserver-1] 
[gzserver-1] [INFO] [1677601646.940084057] [gazebo_ros2_control]: Loading gazebo_ros2_control plugin
[gzserver-1] [INFO] [1677601646.941849599] [wx250.gazebo_ros2_control]: Starting gazebo_ros2_control plugin in namespace: /wx250
[gzserver-1] [INFO] [1677601646.941974254] [wx250.gazebo_ros2_control]: Starting gazebo_ros2_control plugin in ros 2 node: gazebo_ros2_control
[gzserver-1] [INFO] [1677601646.942588643] [wx250.gazebo_ros2_control]: Loading parameter files /home/ty-desktop-22/interbotix_ws/install/interbotix_xsarm_sim/share/interbotix_xsarm_sim/config/trajectory_controllers/wx250_trajectory_controllers.yaml
[gzserver-1] [INFO] [1677601646.944444857] [wx250.gazebo_ros2_control]: connected to service!! robot_state_publisher
[gzserver-1] [INFO] [1677601646.946983852] [wx250.gazebo_ros2_control]: Recieved urdf from param server, parsing...
[gzserver-1] [INFO] [1677601646.960689646] [wx250.gazebo_ros2_control]: Loading joint: waist
[gzserver-1] [INFO] [1677601646.960814086] [wx250.gazebo_ros2_control]: 	State:
[gzserver-1] [INFO] [1677601646.960840492] [wx250.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1677601646.961035024] [wx250.gazebo_ros2_control]: 	Command:
[gzserver-1] [INFO] [1677601646.961043366] [wx250.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1677601646.961321210] [wx250.gazebo_ros2_control]: Loading joint: shoulder
[gzserver-1] [INFO] [1677601646.961332834] [wx250.gazebo_ros2_control]: 	State:
[gzserver-1] [INFO] [1677601646.961339205] [wx250.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1677601646.961413555] [wx250.gazebo_ros2_control]: 	Command:
[gzserver-1] [INFO] [1677601646.961421081] [wx250.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1677601646.961503829] [wx250.gazebo_ros2_control]: Loading joint: elbow
[gzserver-1] [INFO] [1677601646.961513127] [wx250.gazebo_ros2_control]: 	State:
[gzserver-1] [INFO] [1677601646.961518977] [wx250.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1677601646.961526336] [wx250.gazebo_ros2_control]: 	Command:
[gzserver-1] [INFO] [1677601646.961532300] [wx250.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1677601646.961545948] [wx250.gazebo_ros2_control]: Loading joint: wrist_angle
[gzserver-1] [INFO] [1677601646.961553675] [wx250.gazebo_ros2_control]: 	State:
[gzserver-1] [INFO] [1677601646.961559356] [wx250.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1677601646.961565311] [wx250.gazebo_ros2_control]: 	Command:
[gzserver-1] [INFO] [1677601646.961570848] [wx250.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1677601646.961583170] [wx250.gazebo_ros2_control]: Loading joint: wrist_rotate
[gzserver-1] [INFO] [1677601646.961590561] [wx250.gazebo_ros2_control]: 	State:
[gzserver-1] [INFO] [1677601646.961596158] [wx250.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1677601646.961603020] [wx250.gazebo_ros2_control]: 	Command:
[gzserver-1] [INFO] [1677601646.961608760] [wx250.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1677601646.961621056] [wx250.gazebo_ros2_control]: Loading joint: gripper
[gzserver-1] [INFO] [1677601646.961628560] [wx250.gazebo_ros2_control]: 	State:
[gzserver-1] [INFO] [1677601646.961634280] [wx250.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1677601646.961640938] [wx250.gazebo_ros2_control]: 	Command:
[gzserver-1] [INFO] [1677601646.961646571] [wx250.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1677601646.961655305] [wx250.gazebo_ros2_control]: Loading joint: left_finger
[gzserver-1] [INFO] [1677601646.961662192] [wx250.gazebo_ros2_control]: 	State:
[gzserver-1] [INFO] [1677601646.961667766] [wx250.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1677601646.961673504] [wx250.gazebo_ros2_control]: 	Command:
[gzserver-1] [INFO] [1677601646.961678908] [wx250.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1677601646.961701797] [wx250.gazebo_ros2_control]: Loading joint: right_finger
[gzserver-1] [INFO] [1677601646.961709776] [wx250.gazebo_ros2_control]: 	State:
[gzserver-1] [INFO] [1677601646.961715179] [wx250.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1677601646.961720873] [wx250.gazebo_ros2_control]: 	Command:
[... cut ...]
[spawner-6] [INFO] [1677601651.281437239] [wx250.joint_state_broadcaster_spawner]: Waiting for '/wx250/wx250/controller_manager' node to exist
[spawner-6] [INFO] [1677601653.287901870] [wx250.joint_state_broadcaster_spawner]: Waiting for '/wx250/wx250/controller_manager' node to exist
[spawner-6] [INFO] [1677601655.295314073] [wx250.joint_state_broadcaster_spawner]: Waiting for '/wx250/wx250/controller_manager' node to exist

```
Have tried installing on a fresh ubuntu system as well but get the exact same issue.

LSinterbotix, 2023-04-02:
Could you see if the changes in https://github.com/Interbotix/interbotix_ros_manipulators/commit/3ba0e5e55edabb4a116396e9af96e9228409ac88 fix this issue?