Interbotix/interbotix_ros_manipulators #69: [Question]: Failed to launch ros2 interface

esspress0, 2023-03-05:
### Question

I have followed all steps in "ROS2 Interface" documents but I always meet the error 'Package 'interbotix_xsarm_descriptions' not found' . I have tried these on three PCs but I met this same error ;(

### Robot Model

px150

### Operating System

Ubuntu 22.04

### ROS Version

ROS 2 Humble

### Additional Info

```
jinyong2@jinyong-Surface-Pro-4:~$ sudo apt install curl
[sudo] password for jinyong2: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
curl is already the newest version (7.81.0-1ubuntu1.8).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
jinyong2@jinyong-Surface-Pro-4:~$ curl 'https://raw.githubusercontent.com/Interbotix/interbotix_ros_manipulators/main/interbotix_ros_xsarms/install/amd64/xsarm_amd64_install.sh' > xsarm_amd64_install.sh
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 16709  100 16709    0     0  20147      0 --:--:-- --:--:-- --:--:-- 20131
jinyong2@jinyong-Surface-Pro-4:~$ chmod +x xsarm_amd64_install.sh
jinyong2@jinyong-Surface-Pro-4:~$ ./xsarm_amd64_install.sh -d humble
Chosen Version: ROS2 humble
Install the Interbotix Perception packages? This will include the RealSense and AprilTag packages as dependencies.
> yes
Install the MATLAB-ROS API?
> yes
INSTALLATION SUMMARY:
	ROS Distribution:           ROS 2 humble
	Install Perception Modules: true
	Install MATLAB Modules:     true
	Installation path:          /home/jinyong2/interbotix_ws

Is this correct?
> yes

**********************************************

            Starting installation!            
   This process may take around 15 Minutes!   

**********************************************

Hit:1 https://dl.google.com/linux/chrome/deb stable InRelease                  
Get:2 http://security.ubuntu.com/ubuntu jammy-security InRelease [110 kB]      
Hit:3 http://kr.archive.ubuntu.com/ubuntu jammy InRelease                
Get:4 http://kr.archive.ubuntu.com/ubuntu jammy-updates InRelease [119 kB]
Hit:5 http://packages.ros.org/ros2/ubuntu jammy InRelease                      
Get:6 http://kr.archive.ubuntu.com/ubuntu jammy-backports InRelease [107 kB]
Hit:7 http://repo.ros2.org/ubuntu/main jammy InRelease
Fetched 336 kB in 3s (98.6 kB/s)                
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.
W: http://repo.ros2.org/ubuntu/main/dists/jammy/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
Get more security updates through Ubuntu Pro with 'esm-apps' enabled:
  libjs-jquery-ui libopenexr-dev libopenexr25
Learn more about Ubuntu Pro at https://ubuntu.com/pro
The following packages have been kept back:
  tcpdump
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
curl is already the newest version (7.81.0-1ubuntu1.8).
openssh-server is already the newest version (1:8.9p1-3ubuntu0.1).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
Requirement already satisfied: transforms3d in /usr/local/lib/python3.10/dist-packages (0.4.1)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-pip is already the newest version (22.0.2+dfsg-1ubuntu0.2).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
Defaulting to user installation because normal site-packages is not writeable
Collecting modern_robotics
  Downloading modern_robotics-1.1.1-py3-none-any.whl (15 kB)
Requirement already satisfied: numpy in /usr/lib/python3/dist-packages (from modern_robotics) (1.21.5)
Installing collected packages: modern_robotics
Successfully installed modern_robotics-1.1.1
ros-humble-desktop-full is already installed!
Installing Apriltag ROS Wrapper...
./xsarm_amd64_install.sh: line 267: git: command not found

ERROR: your rosdep installation has not been initialized yet.  Please run:

    sudo rosdep init
    rosdep update

                     
Summary: 0 packages finished [0.21s]
Apriltag ROS Wrapper built successfully!
Installing ROS packages for the Interbotix Arm...
./xsarm_amd64_install.sh: line 288: git: command not found
./xsarm_amd64_install.sh: line 289: git: command not found
./xsarm_amd64_install.sh: line 290: git: command not found
./xsarm_amd64_install.sh: line 292: git: command not found
rm: cannot remove 'interbotix_ros_manipulators/interbotix_ros_xsarms/interbotix_xsarm_perception/COLCON_IGNORE': No such file or directory
rm: cannot remove 'interbotix_ros_toolboxes/interbotix_perception_toolbox/COLCON_IGNORE': No such file or directory
rm: cannot remove 'interbotix_ros_toolboxes/interbotix_common_toolbox/interbotix_moveit_interface/COLCON_IGNORE': No such file or directory
rm: cannot remove 'interbotix_ros_toolboxes/interbotix_common_toolbox/interbotix_moveit_interface_msgs/COLCON_IGNORE': No such file or directory
./xsarm_amd64_install.sh: line 299: cd: interbotix_ros_core: No such file or directory
./xsarm_amd64_install.sh: line 300: git: command not found
./xsarm_amd64_install.sh: line 301: git: command not found
./xsarm_amd64_install.sh: line 304: cd: interbotix_ros_toolboxes: No such file or directory
./xsarm_amd64_install.sh: line 305: git: command not found
./xsarm_amd64_install.sh: line 308: cd: interbotix_ros_core/interbotix_ros_xseries/interbotix_xs_sdk: No such file or directory
cp: cannot stat '99-interbotix-udev.rules': No such file or directory

ERROR: your rosdep installation has not been initialized yet.  Please run:

    sudo rosdep init
    rosdep update

                     
Summary: 0 packages finished [0.19s]
Interbotix Arm ROS Packages built successfully!
Setting up Environment Variables...
Installation complete, took 15 seconds in total.
NOTE: Remember to reboot the computer before using the robot!
jinyong2@jinyong-Surface-Pro-4:~$ source ~/interbotix_ws/install/setup.bash
jinyong2@jinyong-Surface-Pro-4:~$ ros2 launch interbotix_xsarm_descriptions xsarm_description.launch.py robot_model:=px150 use_joint_pub_gui:=true
Package 'interbotix_xsarm_descriptions' not found: "package 'interbotix_xsarm_descriptions' not found, searching: ['/opt/ros/humble']"

```

I followed below documents to install 'humble', 

I met error codes running 'xsarm_amd64_install.sh' because of 'rosdep' and 'colcon'.
so I installed them with
$ sudo apt-get install python3-rosdep2
$ sudo apt install python3-colcon-common-extensions

esspress0, 2023-03-06:
Ok I found that I haven't installed git. The green message successfully faked me.

but I still have an error code that 
`[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught exception when trying to load file of format [py]: No module named 'apriltag_ros'`
after running 
`ros2 launch interbotix_xsarm_descriptions xsarm_description.launch.py robot_model:=px150 use_joint_pub_gui:=true`

LSinterbotix, 2023-03-07:
Delete the `~/apriltag_ws` directory and try to run the script again.

esspress0, 2023-03-08:
> Delete the `~/apriltag_ws` directory and try to run the script again.

THANKS FOR YOUR HELP!!

The interface and All codes I ran worked well.
but one I'm worrying is that after I ran the code
`ros2 service call /px150/set_motor_registers interbotix_xs_msgs/srv/RegisterValues "{cmd_type: "single, name: "elbow", reg: "Position_P_Gain", value: 1500}"`
the red light on one of the motors of the elbow is blinking. What's the matter on it?

LSinterbotix, 2023-03-08:
See the "[Red LED Flashing on motor(s)](https://www.trossenrobotics.com/docs/interbotix_xsarms/troubleshooting/index.html#red-led-flashing-on-motor-s)" section of our troubleshooting guide. This is likely caused by overloading of the elbow motor.

esspress0, 2023-03-08:
I think something's wrong on Elbow motor 

```

jinyong2@jinyong-Surface-Pro-4:~$ ros2 launch interbotix_xsarm_control xsarm_control.launch.py  robot_model:=px150
[INFO] [launch]: All log files can be found below /home/jinyong2/.ros/log/2023-03-09-02-20-31-461249-jinyong-Surface-Pro-4-3084
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [xs_sdk-1]: process started with pid [3086]
[INFO] [robot_state_publisher-2]: process started with pid [3088]
[INFO] [rviz2-3]: process started with pid [3090]
[xs_sdk-1] [INFO] Using Interbotix X-Series Driver Version: 'v0.3.3'.
[xs_sdk-1] [INFO] Using logging level 'INFO'.
[xs_sdk-1] [INFO] Loaded mode configs from '/home/jinyong2/interbotix_ws/install/interbotix_xsarm_control/share/interbotix_xsarm_control/config/modes.yaml'.
[xs_sdk-1] [INFO] Loaded motor configs from '/home/jinyong2/interbotix_ws/install/interbotix_xsarm_control/share/interbotix_xsarm_control/config/px150.yaml'.
[xs_sdk-1] [INFO] Pinging all motors specified in the motor_config file. (Attempt 1/3)
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  7, Model: 'XL430-W250', Joint Name: 'wrist_rotate'.
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  6, Model: 'XL430-W250', Joint Name: 'wrist_angle'.
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  5, Model: 'XL430-W250', Joint Name: 'elbow_shadow'.
[xs_sdk-1] [ERROR] 	Can't find DYNAMIXEL ID:  4, Joint Name: 'elbow':
[xs_sdk-1] 		  '[RxPacketError] Hardware error occurred. Check the error at Control Table (Hardware Error Status)!'
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  2, Model: 'XL430-W250', Joint Name: 'shoulder'.
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  8, Model: 'XL430-W250', Joint Name: 'gripper'.
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  3, Model: 'XL430-W250', Joint Name: 'shoulder_shadow'.
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  1, Model: 'XL430-W250', Joint Name: 'waist'.
[xs_sdk-1] [INFO] Pinging all motors specified in the motor_config file. (Attempt 2/3)
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  7, Model: 'XL430-W250', Joint Name: 'wrist_rotate'.
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  6, Model: 'XL430-W250', Joint Name: 'wrist_angle'.
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  5, Model: 'XL430-W250', Joint Name: 'elbow_shadow'.
[xs_sdk-1] [ERROR] 	Can't find DYNAMIXEL ID:  4, Joint Name: 'elbow':
[xs_sdk-1] 		  '[RxPacketError] Hardware error occurred. Check the error at Control Table (Hardware Error Status)!'
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  2, Model: 'XL430-W250', Joint Name: 'shoulder'.
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  8, Model: 'XL430-W250', Joint Name: 'gripper'.
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  3, Model: 'XL430-W250', Joint Name: 'shoulder_shadow'.
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  1, Model: 'XL430-W250', Joint Name: 'waist'.
[xs_sdk-1] [INFO] Pinging all motors specified in the motor_config file. (Attempt 3/3)
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  7, Model: 'XL430-W250', Joint Name: 'wrist_rotate'.
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  6, Model: 'XL430-W250', Joint Name: 'wrist_angle'.
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  5, Model: 'XL430-W250', Joint Name: 'elbow_shadow'.
[xs_sdk-1] [ERROR] 	Can't find DYNAMIXEL ID:  4, Joint Name: 'elbow':
[xs_sdk-1] 		  '[RxPacketError] Hardware error occurred. Check the error at Control Table (Hardware Error Status)!'
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  2, Model: 'XL430-W250', Joint Name: 'shoulder'.
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  8, Model: 'XL430-W250', Joint Name: 'gripper'.
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  3, Model: 'XL430-W250', Joint Name: 'shoulder_shadow'.
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  1, Model: 'XL430-W250', Joint Name: 'waist'.
[xs_sdk-1] [FATAL] Failed to find all motors. Shutting down...
[xs_sdk-1] [FATAL] [1678296033.622916332] [interbotix_xs_sdk.xs_sdk]: InterbotixDriverXS initialization failed: 'Failed to find all motors.'.
[xs_sdk-1] [FATAL] [1678296033.623321044] [interbotix_xs_sdk.xs_sdk]: For troubleshooting, please see 'https://docs.trossenrobotics.com/interbotix_xsarms_docs/troubleshooting.html'.
[ERROR] [xs_sdk-1]: process has died [pid 3086, exit code 1, cmd '/home/jinyong2/interbotix_ws/install/interbotix_xs_sdk/lib/interbotix_xs_sdk/xs_sdk --ros-args -r __node:=xs_sdk -r __ns:=/px150 --params-file /tmp/launch_params_4i7zeysa'].

```

LSinterbotix, 2023-03-08:
If the LED on the elbow motor is still flashing, a simple power cycle should solve the `Hardware error` issue.

closed 2023-04-07 by LSinterbotix