Interbotix/interbotix_ros_manipulators #70: [Bug]: No module named 'interbotix_perception_modules'

Finn-Land, 2023-03-08:
### What happened?

Good morning,
I am currently trying to run the simulation of the wx200 robot. However when launching, I get an error stating that there's no module named 'interbotix_perception_modules'. I am using a raspberry pi model 4, just installed ubuntu 20.04.

### Robot Model

WX200, no real robot, just simulation.

### Operating System

Ubuntu 20.04

### ROS Distro

ROS 2 Galactic

### Steps To Reproduce

I just installed a new ubuntu image on a raspberry pi model 4. I did the steps as mentioned in the guide:
```
sudo apt install curl
curl 'https://raw.githubusercontent.com/Interbotix/interbotix_ros_manipulators/main/interbotix_ros_xsarms/install/rpi4/xsarm_rpi4_install.sh' > xsarm_rpi4_install.sh
chmod +x xsarm_rpi4_install.sh
./xsarm_rpi4_install.sh -d galactic
```
When installed I ran:
```
source /opt/ros/galactic/setup.bash
source ~/interbotix_ws/install/setup.bash
```
I also built the workspace of course.

### Relevant log output

```shell
ubuntu@ubuntu:~$ source /opt/ros/galactic/setup.bash
ubuntu@ubuntu:~$ source ~/interbotix_ws/install/setup.bash
ubuntu@ubuntu:~$ ros2 launch interbotix_xsarm_descriptions xsarm_description.launch.py robot_model:=wx200 use_joint_pub_gui:=true
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2023-03-08-11-41-14-362098-ubuntu-22333
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py:226> exception=InvalidLaunchFileError('py')>
Traceback (most recent call last):
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 53, in get_launch_description_from_any_launch_file
    return loader(launch_file_path)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/python_launch_file_utilities.py", line 62, in get_launch_description_from_python_launch_file
    launch_file_module = load_python_launch_file_as_module(python_launch_file_path)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/python_launch_file_utilities.py", line 37, in load_python_launch_file_as_module
    loader.exec_module(mod)
  File "<frozen importlib._bootstrap_external>", line 848, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/ubuntu/interbotix_ws/install/interbotix_xsarm_descriptions/share/interbotix_xsarm_descriptions/launch/xsarm_description.launch.py", line 29, in <module>
    from interbotix_xs_modules.xs_common import (
  File "/home/ubuntu/interbotix_ws/install/interbotix_xs_modules/lib/python3.8/site-packages/interbotix_xs_modules/__init__.py", line 38, in <module>
    from . import xs_robot
  File "/home/ubuntu/interbotix_ws/install/interbotix_xs_modules/lib/python3.8/site-packages/interbotix_xs_modules/xs_robot/__init__.py", line 40, in <module>
    from . import locobot
  File "/home/ubuntu/interbotix_ws/install/interbotix_xs_modules/lib/python3.8/site-packages/interbotix_xs_modules/xs_robot/locobot.py", line 39, in <module>
    from interbotix_perception_modules.armtag import InterbotixArmTagInterface
ModuleNotFoundError: No module named 'interbotix_perception_modules'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/actions/include_launch_description.py", line 130, in execute
    launch_description = self.__launch_description_source.get_launch_description(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_source.py", line 84, in get_launch_description
    self._get_launch_description(self.__expanded_location)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_description_source.py", line 53, in _get_launch_description
    return get_launch_description_from_any_launch_file(location)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 56, in get_launch_description_from_any_launch_file
    raise InvalidLaunchFileError(extension, likely_errors=exceptions)
launch.invalid_launch_file_error.InvalidLaunchFileError: Caught exception when trying to load file of format [py]: No module named 'interbotix_perception_modules'
```

LSinterbotix, 2023-03-08:
Please see if the changes in interbotix/interbotix_ros_toolboxes@ae10f9f solve this issue.

Finn-Land, 2023-03-09:
Good morning,
This fixed my problem, thanks!

closed 2023-03-09 by Finn-Land