Interbotix/interbotix_ros_manipulators #71: [Question]: Randomly be ignored to set a velocity(moving_time) in python3

phillip1998, 2023-03-15:
### Question

Hi, I'm still studying the manipulation with python base on example codes.
and I found that when I set the moving_time, sometimes the command was ignored and arm moves with the previous speed.
Bigger problem is, even though it is a same code,  sometimes it works and other times it doesn't.
I tried find the problem but I don't know yet what's going on in rospy. So I stucked in here

I felt that the command is ignored more frequently when the self.moving_time is set small (fast)
and the ignore always cause for the first request to set moving_time if self.moving_time is not 2.0 at start. 

I wanted to add the link what I'm watching but I failed to find where the code is in github, sorry :(

below is the test code to find when the ignore cause

```

def main():
    bot = InterbotixManipulatorXS(
        robot_model='px150',
        group_name='arm',
        gripper_name='gripper'
    )
    
    # bot.arm.set_trajectory_time_force(moving_time=1)
         # 'set_trajectory_time_force' is customized function to force a request, and it worked well with this code

    print(bot.arm.moving_time)
    bot.arm.set_ee_pose_components(x=0.1,z=0.35, moving_time=2)
    print(bot.arm.moving_time)
    # If I run this python file twice, first one works well. But in second, the argument 'moving_time=2' is ignored    

    bot.arm.go_to_sleep_pose(moving_time=1)
    
    print(bot.arm.moving_time)
    bot.arm.set_ee_pose_components(x=0.1,z=0.35, moving_time=3)
    print(bot.arm.moving_time)
    bot.arm.go_to_sleep_pose(moving_time=1)
    
    bot.shutdown()

if __name__ == '__main__':
    main()

```

### Robot Model

px150

### Operating System

Ubuntu 22.04

### ROS Version

ROS 2 Humble

LSinterbotix, 2023-04-18:
Could you run the xs_driver in DEBUG mode when launching the xsarm_control package? Provide the output corresponding to you running your script.

LSinterbotix, 2023-06-13:
Closing due to lack of activity. Feel free to reopen if necessary.

closed 2023-06-13 by LSinterbotix