Interbotix/interbotix_ros_manipulators #73: [Question]: Go to a point with all possible orientation

sekisushai, 2023-03-20:
### Question

Hello,
I would like to bring the end effector of a `wx250s` robot to an "optimal" point `x,y,z` where a maximum number of orientation `yaw, pitch, roll` are reachable.

Do you have any idea where this point would be ?

To do this, I use the Python API and especially the function `set_ee_pose_components`.

I notice that the IK solver can not reach positions that nevertheless seem reachable:

For example in the following script:

```
from interbotix_xs_modules.xs_robot.arm import InterbotixManipulatorXS
import numpy as np
from time import sleep

bot = InterbotixManipulatorXS(
        robot_model='wx250s',
        group_name='arm',
        gripper_name='gripper'
    )

bot.arm.go_to_home_pose()
for i in np.arange(0, 1, 0.1):
    print(i)
    bot.arm.set_ee_pose_components(x=0.2, y=0.1, z=0.4, roll =i*np.pi, pitch=0)
    sleep(1)
```
The IK solver does not find solution for `i=0.7` and `i=0.8` while when I am using the function 
`bot.arm.set_single_joint_position(joint_name='wrist_rotate', position=0.8*np.pi)` from `i=0.7` position, it is working.

Thank you for your help.

### Robot Model

wx250s

### Operating System

Ubuntu 22.04

### ROS Version

ROS 2 Humble

LSinterbotix, 2023-04-07:
The kinematics solver in the Python-ROS API is not particularly powerful. If your goal position is nearby in the joint-space, you can provide the current joint positions to the method as a [`custom_guess`](https://github.com/Interbotix/interbotix_ros_toolboxes/blob/7a0852d4ab3a891c2bffeb8c903b6d358e6a5d28/interbotix_xs_toolbox/interbotix_xs_modules/interbotix_xs_modules/xs_robot/arm.py#L546) to help the solver.

LSinterbotix, 2023-05-03:
Do you have any further questions on this? Otherwise, this issue will be closed.

sekisushai, 2023-05-04:
Thank you for your answer. I'm helping the solver for the next pose with the joint position of the current pose with `custom_guess` and it seems to work. I believe I should use a reachability map to have an idea of the point I'm looking for, such as http://wiki.ros.org/reuleaux

LSinterbotix, 2023-05-30:
Have you found success in using a reachability map?

LSinterbotix, 2023-06-13:
Closing due to lack of activity. Feel free to reopen if necessary.

closed 2023-06-13 by LSinterbotix