Interbotix/interbotix_ros_manipulators #75: [Question]: Help for customizing a wx250s robot

sekisushai, 2023-04-04:
### Question

Hello,
I would like to customize a new robot from `wx250s` to `wx250s_custom` by:
- Removing the gripper servo + bar and fingers
- Defining the end-effector frame that is collinear with the `gripper_link` frame, i.e. moving the current `ee_gripper_link` frame in RViz to another position

I would like to use the Python API. 
According to its documentation I first run RViz with the following command:
```
$ ros2 launch interbotix_xsarm_control xsarm_control.launch.py robot_model:=wx250s_custom
```
To make this command work, from what I understood in the source code but also from issue #20, here is what I have done so far:
1) Add `wx250_custom` to the `_XSARM_MODELS` tuple in the file `~/interbotix_ws/install/interbotix_xsarm_control/share/interbotix_xsarm_control/config/xs_common.py`
2) Create the file `wx250s_custom.urdf.xacro` in the folder 
`~/interbotix_ws/install/interbotix_xsarm_descriptions/share/interbotix_xsarm_descriptions/urdf/` and set `false` for all that concern the gripper. Here is the file:
https://pastebin.com/L0fDFhTz 
3) Create a new file `wx250s_custom.yaml` in `~/interbotix_ws/install/interbotix_xsarm_control/share/interbotix_xsarm_control/config/` where I have removed all that concern the gripper. Here is my file:
https://pastebin.com/1QVAi1qd
4) copy the `wx250s` folder to `wx250s_custom` in the `~/interbotix_ws/install/interbotix_xsarm_descriptions/share/interbotix_xsarm_descriptions/meshes/` folder
5) Comment out the following lines in the `modes.yaml` file in the `~/interbotix_ws/install/interbotix_xsarm_control/share/interbotix_xsarm_control/config/` folder:
```
#singles:
# gripper:
# operating_mode: pwm
# torque_enable: true
```
This removes the error `[ERROR] The 'gripper' joint/group does not exist. Was it added to the motor config file?` when launching rviz.
7) Create a class in file `~/interbotix_ws/install/interbotix_xs_modules/lib/python3.10/site-packages/interbotix_xs_modules/xs_robot/mr_descriptions.py` as mentionned in #20.

Currently, this is what I get in RViz :

 https://imgur.com/a/N5wA0eB

Modifying the `wx250s_custom` class in the `mr_descriptions.py` file` doesn't seem to change anything, in particular the ee_gripper_link does not move.

Thus, my questions are what am I doing wrong in the previous steps and what would be the procedure to get it right?

Thanks for your help.

### Robot Model

wx250s

### Operating System

Ubuntu 22.04

### ROS Version

ROS 2 Humble

LSinterbotix, 2023-04-07:
Note that all changes should be made in the `src` directory, not `install` or `build` as changes may be overwritten on subsequent builds. To have changes in `src` take effect, rebuild the workspace, or link the contents of your `src` directory to the install space by using the `--symlink-install` flag when building.

I believe your steps are mostly correct. To move the location of the end-effector frame, you would have to edit the position of the `ee_gripper_link` frame by modifying the origin of the `ee_gripper` joint.

Once changes to your URDF have been made you can then edit mr_descriptions. mr_descriptions.py describes the kinematic properties of the arm to the Python-ROS API for FK/IK. To get accurate values for `M` and `Slist`, use the [kinematics_from_description](https://github.com/Interbotix/kinematics_from_description) tool.

sekisushai, 2023-05-04:
Thank you for your answer. I've managed to customize the arm for my needs for now.

sekisushai, 2023-06-29:
> Note that all changes should be made in the `src` directory, not `install` or `build` as changes may be overwritten on subsequent builds. To have changes in `src` take effect, rebuild the workspace, [...]

How do you rebuild the workspace ?

LSinterbotix, 2023-06-29:
You can (re)build a workspace using the command `colcon build` in its root. The `--symlink-install` flag symbolically links your src directory to the install space, allowing you to edit non-compiled source files without having to rebuild to apply changes.

sekisushai, 2023-07-05:
I removed the gripper in my `.urdf.xacro` file. However, when I try to use the simulated arm I've got the following error:
```
$ ros2 launch interbotix_xsarm_control xsarm_control.launch.py robot_model:=wx250s_custom use_sim:=true
[INFO] [launch]: All log files can be found below /home/sekisushai/.ros/log/2023-07-05-14-19-30-381034-nancy-22662
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [xs_sdk_sim.py-1]: process started with pid [22671]
[INFO] [robot_state_publisher-2]: process started with pid [22673]
[INFO] [rviz2-3]: process started with pid [22675]
[xs_sdk_sim.py-1] [INFO] [1688559571.319761164] [interbotix_xs_sdk.xs_sdk_sim]: Loaded motor configs from `/home/sekisushai/interbotix_ws/install/interbotix_xsarm_control/share/interbotix_xsarm_control/config/wx250s_custom.yaml`.
[xs_sdk_sim.py-1] [INFO] [1688559571.321002996] [interbotix_xs_sdk.xs_sdk_sim]: Loaded mode configs from `/home/sekisushai/interbotix_ws/install/interbotix_xsarm_control/share/interbotix_xsarm_control/config/modes.yaml`.
[xs_sdk_sim.py-1] [INFO] [1688559571.321316300] [interbotix_xs_sdk.xs_sdk_sim]: The operating mode for the 'all' group was changed to 'position'.
[xs_sdk_sim.py-1] [INFO] [1688559571.321609863] [interbotix_xs_sdk.xs_sdk_sim]: The operating mode for the 'arm' group was changed to 'position'.
[xs_sdk_sim.py-1] Traceback (most recent call last):
[xs_sdk_sim.py-1]   File "/home/sekisushai/interbotix_ws/install/interbotix_xs_sdk/lib/interbotix_xs_sdk/xs_sdk_sim.py", line 855, in <module>
[xs_sdk_sim.py-1]     main()
[xs_sdk_sim.py-1]   File "/home/sekisushai/interbotix_ws/install/interbotix_xs_sdk/lib/interbotix_xs_sdk/xs_sdk_sim.py", line 836, in main
[xs_sdk_sim.py-1]     node = InterbotixRobotXS()
[xs_sdk_sim.py-1]   File "/home/sekisushai/interbotix_ws/install/interbotix_xs_sdk/lib/interbotix_xs_sdk/xs_sdk_sim.py", line 130, in __init__
[xs_sdk_sim.py-1]     if not self.robot_get_motor_configs():
[xs_sdk_sim.py-1]   File "/home/sekisushai/interbotix_ws/install/interbotix_xs_sdk/lib/interbotix_xs_sdk/xs_sdk_sim.py", line 304, in robot_get_motor_configs
[xs_sdk_sim.py-1]     self.robot_set_joint_operating_mode(
[xs_sdk_sim.py-1]   File "/home/sekisushai/interbotix_ws/install/interbotix_xs_sdk/lib/interbotix_xs_sdk/xs_sdk_sim.py", line 393, in robot_set_joint_operating_mode
[xs_sdk_sim.py-1]     self.motor_map[name]['mode'] = mode
[xs_sdk_sim.py-1] KeyError: 'gripper'
[ERROR] [xs_sdk_sim.py-1]: process has died [pid 22671, exit code 1, cmd '/home/sekisushai/interbotix_ws/install/interbotix_xs_sdk/lib/interbotix_xs_sdk/xs_sdk_sim.py --ros-args -r __node:=xs_sdk_sim -r __ns:=/wx250s_custom --params-file /tmp/launch_params_0lrptm_t']
```
Any ideas how to fix this ?

LSinterbotix, 2023-07-05:
Please provide both your motor and mode config files.

Make sure you have removed the gripper joint from the modes config `singles`.

sekisushai, 2023-07-06:
I have removed the gripper joint from the modes config `singles` and it is working now. Thank you.

closed 2023-07-06 by sekisushai