Interbotix/interbotix_ros_manipulators #76: [Question]:

PizzzzaGuy, 2023-04-18:
### Question

i am having issues with the installation of the ROS2 (using the recommended script here https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros_interface/ros2/software_setup.html) on ubuntu 22.04 OS installed on RPI4 (8gb)

This is the truncated message i got on the terminal after an installation attempt

Summary: 30 packages finished [10min 54s]
  5 packages had stderr output: dynamixel_workbench_toolbox interbotix_common_modules interbotix_perception_modules interbotix_xs_modules moveit_visual_tools
Interbotix Arm ROS Packages built successfully!
Setting up Environment Variables...
Installation complete, took 3493 seconds in total.
NOTE: Remember to reboot the computer before using the robot!

[log file command prompt.txt](https://github.com/Interbotix/interbotix_ros_manipulators/files/11261548/log.file.command.prompt.txt)

### Robot Model

wx250s

### Operating System

Ubuntu 22.04

### ROS Version

ROS 2 Humble

LSinterbotix, 2023-04-18:
It looks like you're using a virtual environment, is this correct? ROS does not play with those. We recommend working directly with your Python installation.

PizzzzaGuy, 2023-04-18:
Hello,
i am using RPI4 8gb running a ubuntu mate 22.04 OS. 
No virtual environments being used.

LSinterbotix, 2023-04-18:
On line 446 of your log, you run the command 

```
ust-ra6d0f@ustra6d0f-desktop:~$ python3 -m venv tutorial-env
```

which activates the virtual environment.

Based on the output of the installation script, it seems like everything worked properly. Did you try going through the [ROS 2 Quickstart Guide](https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros_interface/ros2/quickstart.html)?

PizzzzaGuy, 2023-04-18:
i havent actually proceeded to the quickstart guide after reading the below messages. if there is nothing wrong with the gazebo install, i would like to go through the quickstart guide and try using the physical arm. 

WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y ros-humble-gazebo-ros] failed
  apt: command [sudo -H apt-get install -y ros-humble-gazebo-ros2-control] failed
  apt: Failed to detect successful installation of [ros-humble-gazebo-ros]
  apt: Failed to detect successful installation of [ros-humble-gazebo-ros2-control]
Starting >>> interbotix_xs_msgs
Starting >>> dynamixel_workbench_toolbox
Starting >>> interbotix_common_modules
Starting >>> interbotix_common_sim                                                                     
--- stderr: interbotix_common_modules                                   
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.

closed 2023-05-10 by LSinterbotix