Interbotix/interbotix_ros_manipulators #77: [Question]: ImportError

KSaiAkshit, 2023-04-26:
### Question

Hello,
I am trying to run bartender.py with the ReactorX150 (rx150), following as mentioned in the docs.
The module does exist inside the toolbox. I have installed the packages using the script given in the [Documentation](https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros_interface/ros1/software_setup.html). I believe it has installed properly given how I can manipulate the arm in rviz and Gazebo. 

I have made sure to source the setup script. Output of `printenv PYTHONPATH`:
`/home/akshit/interbotix_ws/devel/lib/python3/dist-packages:/home/akshit/catkin_ws/devel/lib/python3/dist-packages:/opt/ros/noetic/lib/python3/dist-packages`

### Robot Model

rx150

### Operating System

Other (Describe in "Additional Info")

### ROS Version

ROS 1 Noetic

### Additional Info

I use WSL2 with Windows 12

KSaiAkshit, 2023-04-26:
Turns out the `interbotix_ux_module` is not in `dist-packages`. The `catkin_python_setup()` line is uncommented in the CMakeList of `interbotix_ux_module` and the `setup.py` file is in the root directory of the package. The workspaces builds just fine with `catkin build`.
It still doesn't work though. I still get a ModuleNotFoundError when I run `bartender.py`

KSaiAkshit, 2023-04-26:
**Update**

The `catkin build` command can't find any of the `interbotix_ux_*` packages in the workspace. The cmakelist and package.xml are fine as far as I can tell.

LSinterbotix, 2023-04-26:
The rx150 is an Interbotix X-Series arm, not a UFactory xArm. It is compatible with the [bartender.py script](https://github.com/Interbotix/interbotix_ros_manipulators/blob/main/interbotix_ros_xsarms/examples/python_demos/bartender.py) in the interbotix_ros_xsarms Python demos using the [interbotix_xs_modules](https://github.com/Interbotix/interbotix_ros_toolboxes/tree/main/interbotix_xs_toolbox/interbotix_xs_modules) Python package.

closed 2023-04-27 by KSaiAkshit