Interbotix/interbotix_ros_manipulators #78: [Question]: Packages not found

jakepdconway, 2023-04-26:
### Question

Hello everyone,

I am trying to use the ReactorX 200 with ROS2 Humble on Ubuntu 22.04. I have followed the standard software setup guide from https://www.trossenrobotics.com/docs/interbotix_xsarms/ros_interface/ros2_software_setup.html .

It all went well and the final checkup command worked and I moved to the ROS2 quickstart guide. I sourced and ran the first command as seen below:
```
source /opt/ros/humble/setup.bash
ros2 launch interbotix_xsarm_descriptions xsarm_description.launch.py robot_model:=rx200 use_joint_pub_gui:=true
```
But I end up getting this error:

> Package 'interbotix_xsarm_descriptions' not found: "package 'interbotix_xsarm_descriptions' not found, searching: ['/opt/ros/humble']"

I tried cloning the repo and building the interbotix_xsarm_description package, but the error persisted.
Do you have any suggestions to how I can fix this?

Thank you

### Robot Model

rx200

### Operating System

Ubuntu 22.04

### ROS Version

ROS 2 Humble

LSinterbotix, 2023-04-27:
You should also source the built workspace:

```
source /opt/ros/humble/setup.bash
source ~/interbotix_ws/install/setup.bash # <--
ros2 launch interbotix_xsarm_descriptions xsarm_description.launch.py robot_model:=rx200 use_joint_pub_gui:=true
```

Alternatively, you can add the workspace sourcing command to your `~/.bashrc` file so every new terminal's environment is configured properly (and the distro sourcing command if it's not already there).

You are using an older version of the documentation as it says at the top of the page. Please refer to the [correct location](https://docs.trossenrobotics.com/interbotix_xsarms_docs/index.html) in the future.

jakepdconway, 2023-04-27:
Hi,

Thank you this worked very well and fixed my issue.

I now have a new issue where when i launch the code to move the arm itself nothing is happening.

`ros2 launch interbotix_xsarm_ros_control xsarm_ros_control.launch.py robot_model:=rx200 use_joint_pub_gui:=true use_rviz:=true`

When RViz is launched along side the robot, the image of the robot starts to flick between the initial position of the arm (in which it is stuck with motors on) and the desired positions set in the controller. I have checked the arguments on the website below:

https://www.trossenrobotics.com/docs/interbotix_xsarms/ros2_packages/ros_control.html

As well as this when I change the values in the window the arm doesn't move or react but the motors are all working fine. If the arm is slightly wiggled the said wiggle is visable in RViz in the initial position it is stuck in.

Would you know what causes the issue or how to fix it?
Thank you.

LSinterbotix, 2023-04-27:
What do you expect to happen when the `xsarm_ros_control` package is launched? I noticed that you're using the `use_joint_pub_gui` launch arg which is not included in the package's launch file. Please provide a screencapture showing the flickering behavior you mention.

The `xsarm_ros_control` package isn't meant to be used by itself — rather it is typically launched by the [MoveIt configuration package](https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros2_packages/moveit_motion_planning_configuration.html) when the actual hardware is used, as it configures the arm and gripper JointTrajectoryControllers. Refer to that package instead for usage info.

closed 2023-06-07 by LSinterbotix