Interbotix/interbotix_ros_manipulators #79: [Question]: No module named 'interbotix_xs_modules'

Zyonia, 2023-04-28:
### Question

Hello everyone,

I am using ROS2 Humble on WSL Ubuntu 22.04. 
I have followed the standard software setup guide from https://www.trossenrobotics.com/docs/interbotix_xsarms/ros_interface/ros2_software_setup.html , everything seems ok, no error.
And then I entered the checkup code `ls /dev | grep ttyDXL` but there is noting show up.
i tried 
`source ~/interbotix_ws/install/setup.bash`
`ros2 launch interbotix_xsarm_moveit xsarm_moveit.launch.py`
here is the output
`[INFO] [launch]: All log files can be found below /home/zyonia/.ros/log/2023-04-28-16-44-41-871055-zhao-720
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught exception when trying to load file of format [py]: No module named 'interbotix_xs_modules'
`
What should i do?

### Operating System

Other (Describe in "Additional Info")

### ROS Version

ROS 2 Humble

### Additional Info

WSL UBUNTU 22.04

LSinterbotix, 2023-05-10:
WSL isn't a platform we support so you may have to do some additional configuration to get access to external hardware devices like the U2D2. [This guide](https://learn.microsoft.com/en-us/windows/wsl/connect-usb) from Microsoft might be a good start.

For the missing modules issue, provide the output of the following commands:

```bash
source /opt/ros/humble/setup.bash
source ~/interbotix_ws/install/setup.bash
ros2 pkg list | grep interbotix
```

LSinterbotix, 2023-06-13:
Do you have any further questions on this? Otherwise, this issue will be closed.

Zyonia, 2023-06-14:
```bash
$ ros2 pkg list | grep interbotix
interbotix_ros_xsarms
interbotix_ros_xsarms_examples
interbotix_ros_xseries
interbotix_xs_driver
interbotix_xs_msgs
interbotix_xs_sdk
interbotix_xsarm_control
interbotix_xsarm_descriptions
interbotix_xsarm_joy
interbotix_xsarm_moveit
interbotix_xsarm_moveit_interface
interbotix_xsarm_perception
interbotix_xsarm_ros_control
interbotix_xsarm_sim
```
Hello, here is the output, hope my reply won't be too late :-)

Zyonia, 2023-06-16:
here is the output of 
```bash
cd ~/interbotix_ws && colcon build --symlink-install
```
```bash
# zyonia @ zhao in ~/interbotix_ws [10:15:33]
$ colcon build --symlink-install
Starting >>> interbotix_xs_driver
Starting >>> interbotix_xs_msgs
Starting >>> interbotix_xsarm_descriptions
Starting >>> moveit_visual_tools
Finished <<< interbotix_xs_driver [4.27s]
--- stderr: interbotix_xs_msgs
failed to create symbolic link '/home/zyonia/interbotix_ws/build/interbotix_xs_msgs/ament_cmake_python/interbotix_xs_msgs/interbotix_xs_msgs' because existing path cannot be removed: Is a directory
gmake[2]: *** [CMakeFiles/ament_cmake_python_symlink_interbotix_xs_msgs.dir/build.make:70: CMakeFiles/ament_cmake_python_symlink_interbotix_xs_msgs] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:421: CMakeFiles/ament_cmake_python_symlink_interbotix_xs_msgs.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< interbotix_xs_msgs [5.15s, exited with code 2]
Aborted  <<< interbotix_xsarm_descriptions [5.32s]
Aborted  <<< moveit_visual_tools [20.6s]

Summary: 1 package finished [20.8s]
  1 package failed: interbotix_xs_msgs
  2 packages aborted: interbotix_xsarm_descriptions moveit_visual_tools
  2 packages had stderr output: interbotix_xs_msgs moveit_visual_tools
  11 packages not processed
```

LSinterbotix, 2023-06-16:
Try removing the `build` and `install` directories before building.

Zyonia, 2023-06-28:
```bash
# zyonia @ zhao in ~/interbotix_ws/src [9:35:26]
$ tree
.
├── interbotix_ros_core
│   ├── LICENSE
│   ├── README.md
│   ├── images
│   │   ├── core_banner.png
│   │   ├── core_irros_structure.png
│   │   └── core_repo_structure.png
│   └── interbotix_ros_xseries
│       ├── README.md
│       ├── dynamixel_workbench_toolbox
│       ├── interbotix_ros_xseries
│       │   ├── CMakeLists.txt
│       │   ├── README.md
│       │   └── package.xml
│       ├── interbotix_xs_driver
│       │   ├── CMakeLists.txt
│       │   ├── CONTRIBUTING.md
│       │   ├── LICENSE
│       │   ├── README.md
│       │   ├── cmake
│       │   │   ├── Version.cmake
│       │   │   └── version.hpp.in
│       │   ├── include
│       │   │   └── interbotix_xs_driver
│       │   │       ├── xs_common.hpp
│       │   │       ├── xs_driver.hpp
│       │   │       └── xs_logging.hpp
│       │   ├── package.xml
│       │   └── src
│       │       ├── xs_driver.cpp
│       │       └── xs_logging.cpp
│       ├── interbotix_xs_msgs
│       │   ├── CMakeLists.txt
│       │   ├── README.md
│       │   ├── msg
│       │   │   ├── ArmJoy.msg
│       │   │   ├── HexJoy.msg
│       │   │   ├── JointGroupCommand.msg
│       │   │   ├── JointSingleCommand.msg
│       │   │   ├── JointTemps.msg
│       │   │   ├── JointTrajectoryCommand.msg
│       │   │   ├── LocobotJoy.msg
│       │   │   └── TurretJoy.msg
│       │   ├── package.xml
│       │   └── srv
│       │       ├── MotorGains.srv
│       │       ├── OperatingModes.srv
│       │       ├── Reboot.srv
│       │       ├── RegisterValues.srv
│       │       ├── RobotInfo.srv
│       │       └── TorqueEnable.srv
│       └── interbotix_xs_sdk
│           ├── 99-interbotix-udev.rules
│           ├── CMakeLists.txt
│           ├── README.md
│           ├── TROUBLESHOOTING.md
│           ├── config
│           │   ├── mode_configs_template.yaml
│           │   └── motor_configs_template.yaml
│           ├── images
│           │   ├── wizard_options.png
│           │   └── xs_sdk_banner.png
│           ├── include
│           │   └── interbotix_xs_sdk
│           │       └── xs_sdk_obj.hpp
│           ├── package.xml
[... 489 lines cut ...]
    │       ├── imarker_end_effector.h
    │       ├── imarker_robot_state.h
    │       └── moveit_visual_tools.h
    ├── launch
    │   ├── demo.rviz
    │   └── demo_rviz.launch.py
    ├── mainpage.dox
    ├── moveit_visual_tools.repos
    ├── package.xml
    ├── resources
    │   ├── demo.png
    │   ├── demo_mesh.stl
    │   ├── rrbot.srdf
    │   ├── rrbot.urdf
    │   └── screenshot.png
    └── src
        ├── imarker_end_effector.cpp
        ├── imarker_robot_state.cpp
        ├── moveit_visual_tools.cpp
        └── moveit_visual_tools_demo.cpp

120 directories, 455 files

```
There are 4 `COLCON_IGNORE` in the xxx/example/

LSinterbotix, 2023-06-28:
Looks like git failed to clone the interbotix_ros_toolboxes repo to your workspace. To fix this: 

1. Clone the branch matching your ROS distro to the `~/interbotix_ws/src` directory
2. Remove any `COLCON_IGNORE` files in packages you'd like to build (like the one in interbotix_perception_toolbox)
3. Run rosdep to install any missing dependencies
4. Rebuild your workspace.

```bash
cd ~/interbotix_ws/src
git clone -b humble https://github.com/Interbotix/interbotix_ros_toolboxes.git
rm interbotix_ros_toolboxes/interbotix_perception_toolbox/COLCON_IGNORE
cd ~/interbotix_ws
rosdep install --from-paths src --ignore-src -r -y
colcon build --symlink-install
```

Zyonia, 2023-07-06:
Alright, I succeed! Thanks a lot!

closed 2023-07-06 by LSinterbotix