Interbotix/interbotix_ros_manipulators #80: [Question]:

walkergrimes5, 2023-05-04:
### Question

Working my way through the installation guide and have made it to the check steps. I can verify the connection with the U2D2, but when I try to confirm the package installation, nothing shows up. I input the following commands:
$ source /opt/ros/humble/setup.bash
$ source ~/interbotix_ws/install/setup.bash
$ ros2 pkg list | grep interbotix

This issue is corroborated when I try to move onto the quick start guide. When I enter:
$ source /opt/ros/humble/setup.bash
$ source ~/interbotix_ws/install/setup.bash
$ ros2 launch interbotix_xsarm_descriptions xsarm_description.launch.py robot_model:=wx250 use_joint_pub_gui:=true

I get the following error:
Package 'interbotix_xsarm_descriptions' not found: "package 'interbotix_xsarm_descriptions' not found, searching: ['/opt/ros/humble']"

I checked the installation printout and see confirmation that the packages are installed. I am new to Linux, so I'm sure I'm making a silly mistake somewhere, but I'm not sure what it is.

Thanks!

### Robot Model

wx250

### Operating System

Ubuntu 22.04

### ROS Version

ROS 2 Humble

LSinterbotix, 2023-05-10:
Did you see any errors in the terminal when running the install script? What happens when you run the command `source /opt/ros/humble/setup.bash && colcon build --symlink-install` in the `~/interbotix_ws` directory?

walkergrimes5, 2023-05-10:
No errors in the in terminal. Ended up resolving this issue earlier today by reverting to ROS1 and Noetic. Same set of install commands worked perfectly.

closed 2023-05-10 by walkergrimes5