Interbotix/interbotix_ros_manipulators #81: [Question]: Waiting for controller manager node to exist.

AbishekNP, 2023-05-23:
### Question

I'm getting this error when I try to execute the "ros2 launch interbotix_xsarm_moveit xsarm_moveit.launch.py robot_model:=px100 hardware_type:=actual" command.
The output is as follows:

[move_group-1] You can start planning now!
[move_group-1] 
[INFO] [spawn_entity.py-5]: process has finished cleanly [pid 158050]
[INFO] [spawner-7]: process started with pid [158196]
[gzclient-4] gzclient: /usr/include/boost/smart_ptr/shared_ptr.hpp:728: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = gazebo::rendering::Camera; typename boost::detail::sp_member_access<T>::type = gazebo::rendering::Camera*]: Assertion `px != 0' failed.
[ERROR] [gzclient-4]: process has died [pid 158048, exit code -6, cmd 'gzclient'].
[spawner-7] [INFO] [1684874563.018613795] [vx300s.joint_state_broadcaster_spawner]: Waiting for '/vx300s/controller_manager' node to exist
[spawner-7] [INFO] [1684874565.026315674] [vx300s.joint_state_broadcaster_spawner]: Waiting for '/vx300s/controller_manager' node to exist
[spawner-7] [INFO] [1684874567.034041841] [vx300s.joint_state_broadcaster_spawner]: Waiting for '/vx300s/controller_manager' node to exist
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[move_group-1] [INFO] [1684874567.818492195] [rclcpp]: signal_handler(signum=2)
[spawner-7] Traceback (most recent call last):
[spawner-7]   File "/opt/ros/humble/lib/controller_manager/spawner", line 33, in <module>
[spawner-7]     sys.exit(load_entry_point('controller-manager==2.25.2', 'console_scripts', 'spawner')())
[spawner-7]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/controller_manager/spawner.py", line 182, in main
[spawner-7]     if not wait_for_controller_manager(node, controller_manager_name,
[spawner-7]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/controller_manager/spawner.py", line 100, in wait_for_controller_manager
[spawner-7]     node_and_namespace = wait_for_value_or(
[spawner-7]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/controller_manager/spawner.py", line 61, in wait_for_value_or
[spawner-7]     time.sleep(0.2)
[spawner-7] KeyboardInterrupt
[move_group-1] [INFO] [1684874567.837646210] [moveit.ros_planning_interface.moveit_cpp]: Deleting MoveItCpp
[move_group-1] [INFO] [1684874567.838333792] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Stopped publishing maintained planning scene.
[move_group-1] [INFO] [1684874567.838834914] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Stopping world geometry monitor
[move_group-1] [INFO] [1684874567.839016928] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Stopping planning scene monitor
[move_group-1] Warning: class_loader.ClassLoader: SEVERE WARNING!!! Attempting to unload library while objects created by this loader exist in the heap! You should delete your objects before attempting to unload the library or destroying the ClassLoader. The library will NOT be unloaded.
[move_group-1]          at line 127 in ./src/class_loader.cpp
[move_group-1] Stack trace (most recent call last):
[move_group-1] #16   Object "", at 0xffffffffffffffff, in 
[ERROR] [spawner-7]: process has died [pid 158196, exit code -2, cmd '/opt/ros/humble/lib/controller_manager/spawner -c controller_manager joint_state_broadcaster --ros-args -r __node:=joint_state_broadcaster_spawner -r __ns:=/vx300s --params-file /tmp/launch_params_19mlagqw'].
[move_group-1] #15   Object "/opt/ros/humble/lib/moveit_ros_move_group/move_group", at 0x55c681114784, in 
[INFO] [robot_state_publisher-6]: process has finished cleanly [pid 158053]
[move_group-1] #14   Source "../csu/libc-start.c", line 392, in __libc_start_main_impl [0x7fb07fe29e3f]
[ERROR] [rviz2-2]: process has died [pid 158044, exit code -11, cmd '/opt/ros/humble/lib/rviz2/rviz2 -d /home/robotarm/interbotix_ws/install/interbotix_xsarm_moveit/share/interbotix_xsarm_moveit/rviz/xsarm_moveit.rviz -f world --ros-args -r __node:=rviz2 --params-file /tmp/launch_params_dtj2nivy --params-file /tmp/launch_params_f_gsutze --params-file /tmp/launch_params_17x6e75z --params-file /home/robotarm/interbotix_ws/install/interbotix_xsarm_moveit/share/interbotix_xsarm_moveit/config/kinematics.yaml --params-file /tmp/launch_params_9_8p8q7x -r vx300s/get_planning_scene:=/vx300s/get_planning_scene -r /arm_controller/follow_joint_trajectory:=/vx300s/arm_controller/follow_joint_trajectory -r /gripper_controller/follow_joint_trajectory:=/vx300s/gripper_controller/follow_joint_trajectory'].
[move_group-1] #13   Source "../sysdeps/nptl/libc_start_call_main.h", line 58, in __libc_start_call_main [0x7fb07fe29d8f]
[move_group-1] #12   Object "/opt/ros/humble/lib/moveit_ros_move_group/move_group", at 0x55c6811136bb, in 
[move_group-1] #11   Object "/opt/ros/humble/lib/moveit_ros_move_group/move_group", at 0x55c681115769, in 
[move_group-1] #10   Object "/opt/ros/humble/lib/libmoveit_cpp.so.2.5.4", at 0x7fb080afba46, in moveit_cpp::MoveItCpp::~MoveItCpp()
[move_group-1] #9    Object "/opt/ros/humble/lib/libmoveit_cpp.so.2.5.4", at 0x7fb080af9999, in 
[move_group-1] #8    Object "/opt/ros/humble/lib/libmoveit_trajectory_execution_manager.so.2.5.4", at 0x7fb0800d15b5, in trajectory_execution_manager::TrajectoryExecutionManager::~TrajectoryExecutionManager()
[move_group-1] #7    Object "/opt/ros/humble/lib/libmoveit_trajectory_execution_manager.so.2.5.4", at 0x7fb0800e41a9, in 
[move_group-1] #6    Object "/opt/ros/humble/lib/librclcpp.so", at 0x7fb080763cec, in rclcpp::Node::~Node()
[move_group-1] #5    Object "/opt/ros/humble/lib/librclcpp.so", at 0x7fb080763cbe, in rclcpp::Node::~Node()
[move_group-1] #4    Object "/opt/ros/humble/lib/librclcpp.so", at 0x7fb08073f7c9, in 
[move_group-1] #3    Object "/opt/ros/humble/lib/librclcpp.so", at 0x7fb08076d2d9, in 
[move_group-1] #2    Object "/opt/ros/humble/lib/librclcpp.so", at 0x7fb08076d220, in rclcpp::node_interfaces::NodeBase::~NodeBase()
[move_group-1] #1    Object "/opt/ros/humble/lib/librclcpp.so", at 0x7fb08073f7c9, in 
[move_group-1] #0    Object "/opt/ros/humble/lib/librclcpp.so", at 0x7fb080744511, in rclcpp::CallbackGroup::~CallbackGroup()
[move_group-1] Segmentation fault (Address not mapped to object [0x7fb0742867a8])
[INFO] [gzserver-3]: process has finished cleanly [pid 158046]
[ERROR] [move_group-1]: process has died [pid 158042, exit code -11, cmd '/opt/ros/humble/lib/moveit_ros_move_group/move_group --ros-args --params-file /tmp/launch_params_mijosql8 --params-file /tmp/launch_params_vc0uynes --params-file /tmp/launch_params_t8y5qcl5 --params-file /home/robotarm/interbotix_ws/install/interbotix_xsarm_moveit/share/interbotix_xsarm_moveit/config/kinematics.yaml --params-file /tmp/launch_params_gts_pq6z --params-file /tmp/launch_params_kv_bole4 --params-file /tmp/launch_params_efs5a46m --params-file /tmp/launch_params_rt1zt_nu --params-file /tmp/launch_params__0_qz3gh --params-file /tmp/launch_params_hb2fpehe -r vx300s/get_planning_scene:=/vx300s/get_planning_scene -r /arm_controller/follow_joint_trajectory:=/vx300s/arm_controller/follow_joint_trajectory -r /gripper_controller/follow_joint_trajectory:=/vx300s/gripper_controller/follow_joint_trajectory'].
[move_group-1] 

### Robot Model

vx300s

### Operating System

Ubuntu 22.04

### ROS Version

ROS 2 Humble

LSinterbotix, 2023-05-23:
Please see if the advice in [this ROS Answers comment](https://answers.ros.org/question/358847/cannot-launch-gzclient-on-a-launch-file-results-in-shared_ptr-assertion-error/?answer=387418#post-id-387418) solves your issue.

AbishekNP, 2023-05-23:
Hey. The solution you provided launches the robot arm both in rviz and in a gazebo world. But, it still throws the same "waiting for controller manager node to exist" error. The arm does not move in the desired trajectory when I move it to the desired position and click on Plan and Execute. It just stays still.

AbishekNP, 2023-05-23:
This is only occurring when I use the hardware_type to gz_classic. The MoveIt launch command works fine when I use the "actual" hardware type.

LSinterbotix, 2023-05-23:
Are your local repos up-to-date? A similar (if not the same) issue was fixes in the recent commit https://github.com/Interbotix/interbotix_ros_manipulators/commit/3ba0e5e55edabb4a116396e9af96e9228409ac88. Try to see if pulling this change solves your issue.

AbishekNP, 2023-05-23:
Yeah. They're up-to-date. Still it doesn't solve the issue.

AbishekNP, 2023-05-25:
robotarm@gorgon:~$ ros2 launch interbotix_xsarm_moveit xsarm_moveit.launch.py robot_model:=vx300s hardware_type:=gz_classic
[INFO] [launch]: All log files can be found below /home/robotarm/.ros/log/2023-05-24-17-42-30-478850-gorgon-183258
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [183262]
[INFO] [rviz2-2]: process started with pid [183264]
[INFO] [gzserver-3]: process started with pid [183266]
[INFO] [gzclient-4]: process started with pid [183268]
[INFO] [spawn_entity.py-5]: process started with pid [183280]
[INFO] [robot_state_publisher-6]: process started with pid [183287]
[move_group-1] [WARN] [1684975351.993410010] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-1] [INFO] [1684975351.996475022] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0 seconds
[move_group-1] [INFO] [1684975351.996502382] [moveit_robot_model.robot_model]: Loading robot model 'vx300s'...
[move_group-1] [INFO] [1684975351.996509615] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[move_group-1] [INFO] [1684975352.166692505] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-1] [INFO] [1684975352.166846467] [moveit.ros_planning_interface.moveit_cpp]: Listening to '/vx300s/joint_states' for joint states
[move_group-1] [INFO] [1684975352.167426465] [moveit_ros.current_state_monitor]: Listening to joint states on topic '/vx300s/joint_states'
[move_group-1] [INFO] [1684975352.167803286] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[move_group-1] [INFO] [1684975352.167816805] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-1] [INFO] [1684975352.168083586] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[move_group-1] [INFO] [1684975352.168095086] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-1] [INFO] [1684975352.168379960] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-1] [INFO] [1684975352.168654266] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-1] [WARN] [1684975352.169010232] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[move_group-1] [ERROR] [1684975352.169026666] [moveit.ros.occupancy_map_monitor.middleware_handle]: No sensor plugin specified for octomap updater ; ignoring.
[move_group-1] [INFO] [1684975352.169038258] [moveit.ros.occupancy_map_monitor.middleware_handle]: Skipping octomap updater plugin ''
[move_group-1] [INFO] [1684975352.200669218] [moveit.ros_planning_interface.moveit_cpp]: Loading planning pipeline 'move_group'
[move_group-1] [INFO] [1684975352.213741193] [moveit.ros_planning.planning_pipeline]: Using planning interface 'OMPL'
[move_group-1] [INFO] [1684975352.216202621] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.path_tolerance' was not set. Using default value: 0.100000
[move_group-1] [INFO] [1684975352.216220060] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.resample_dt' was not set. Using default value: 0.100000
[move_group-1] [INFO] [1684975352.216224814] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.min_angle_change' was not set. Using default value: 0.001000
[move_group-1] [INFO] [1684975352.216237276] [moveit_ros.fix_workspace_bounds]: Param 'move_group.default_workspace_bounds' was not set. Using default value: 10.000000
[move_group-1] [INFO] [1684975352.216251072] [moveit_ros.fix_start_state_bounds]: Param 'move_group.start_state_max_bounds_error' was set to 0.100000
[move_group-1] [INFO] [1684975352.216256213] [moveit_ros.fix_start_state_bounds]: Param 'move_group.start_state_max_dt' was not set. Using default value: 0.500000
[move_group-1] [INFO] [1684975352.216266519] [moveit_ros.fix_start_state_collision]: Param 'move_group.start_state_max_dt' was not set. Using default value: 0.500000
[move_group-1] [INFO] [1684975352.216271262] [moveit_ros.fix_start_state_collision]: Param 'move_group.jiggle_fraction' was not set. Using default value: 0.020000
[move_group-1] [INFO] [1684975352.216276009] [moveit_ros.fix_start_state_collision]: Param 'move_group.max_sampling_attempts' was not set. Using default value: 100
[move_group-1] [INFO] [1684975352.216286395] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Add Time Optimal Parameterization'
[move_group-1] [INFO] [1684975352.216290621] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Workspace Bounds'
[move_group-1] [INFO] [1684975352.216294144] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State Bounds'
[move_group-1] [INFO] [1684975352.216320411] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State In Collision'
[... cut ...] mp/launch_params_uazyt1s3 --params-file /tmp/launch_params_w084dwby --params-file /tmp/launch_params_on5_kll6 --params-file /tmp/launch_params_8r8amnds --params-file /tmp/launch_params_mjo7vjnl --params-file /tmp/launch_params__9lfkhts -r vx300s/get_planning_scene:=/vx300s/get_planning_scene -r /arm_controller/follow_joint_trajectory:=/vx300s/arm_controller/follow_joint_trajectory -r /gripper_controller/follow_joint_trajectory:=/vx300s/gripper_controller/follow_joint_trajectory'].
[move_group-1]

LSinterbotix, 2023-05-25:
> [gzclient-4] gzclient: /usr/include/boost/smart_ptr/shared_ptr.hpp:728: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = [...]

Looks like Gazebo is still crashing. This will prevent the gazebo_ros2_control plugin from loading, causing the `Waiting for '/vx300s/controller_manager' node to exist` issue.

Add the line below to your `~/.bashrc` file to prevent this from happening in the future, open a new terminal, and try again.

```
. /usr/share/gazebo/setup.sh
```

AbishekNP, 2023-05-25:
Still getting error.

robotarm@gorgon:~/interbotix_ws$ ros2 launch interbotix_xsarm_moveit xsarm_moveit.launch.py robot_model:=vx300s hardware_type:=gz_classic
[INFO] [launch]: All log files can be found below /home/robotarm/.ros/log/2023-05-24-18-15-26-464349-gorgon-185242
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [185246]
[INFO] [rviz2-2]: process started with pid [185248]
[INFO] [gzserver-3]: process started with pid [185250]
[INFO] [gzclient-4]: process started with pid [185252]
[INFO] [spawn_entity.py-5]: process started with pid [185254]
[INFO] [robot_state_publisher-6]: process started with pid [185256]
[move_group-1] [WARN] [1684977328.090983964] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-1] [INFO] [1684977328.093963644] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0 seconds
[move_group-1] [INFO] [1684977328.093989065] [moveit_robot_model.robot_model]: Loading robot model 'vx300s'...
[move_group-1] [INFO] [1684977328.093997093] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[move_group-1] [INFO] [1684977328.364899582] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-1] [INFO] [1684977328.365043357] [moveit.ros_planning_interface.moveit_cpp]: Listening to '/vx300s/joint_states' for joint states
[move_group-1] [INFO] [1684977328.366112999] [moveit_ros.current_state_monitor]: Listening to joint states on topic '/vx300s/joint_states'
[move_group-1] [INFO] [1684977328.369686220] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[move_group-1] [INFO] [1684977328.369711874] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-1] [INFO] [1684977328.370510530] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[move_group-1] [INFO] [1684977328.370525009] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-1] [INFO] [1684977328.371290367] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-1] [INFO] [1684977328.371851141] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-1] [WARN] [1684977328.372181594] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[move_group-1] [ERROR] [1684977328.372196702] [moveit.ros.occupancy_map_monitor.middleware_handle]: No sensor plugin specified for octomap updater ; ignoring.
[move_group-1] [INFO] [1684977328.372208034] [moveit.ros.occupancy_map_monitor.middleware_handle]: Skipping octomap updater plugin ''
[move_group-1] [INFO] [1684977328.408984008] [moveit.ros_planning_interface.moveit_cpp]: Loading planning pipeline 'move_group'
[move_group-1] [INFO] [1684977328.422241016] [moveit.ros_planning.planning_pipeline]: Using planning interface 'OMPL'
[move_group-1] [INFO] [1684977328.425093360] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.path_tolerance' was not set. Using default value: 0.100000
[move_group-1] [INFO] [1684977328.425436250] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.resample_dt' was not set. Using default value: 0.100000
[move_group-1] [INFO] [1684977328.425681405] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.min_angle_change' was not set. Using default value: 0.001000
[move_group-1] [INFO] [1684977328.425926904] [moveit_ros.fix_workspace_bounds]: Param 'move_group.default_workspace_bounds' was not set. Using default value: 10.000000
[move_group-1] [INFO] [1684977328.426176110] [moveit_ros.fix_start_state_bounds]: Param 'move_group.start_state_max_bounds_error' was set to 0.100000
[move_group-1] [INFO] [1684977328.426493490] [moveit_ros.fix_start_state_bounds]: Param 'move_group.start_state_max_dt' was not set. Using default value: 0.500000
[move_group-1] [INFO] [1684977328.426750020] [moveit_ros.fix_start_state_collision]: Param 'move_group.start_state_max_dt' was not set. Using default value: 0.500000
[move_group-1] [INFO] [1684977328.426986650] [moveit_ros.fix_start_state_collision]: Param 'move_group.jiggle_fraction' was not set. Using default value: 0.020000
[move_group-1] [INFO] [1684977328.427248846] [moveit_ros.fix_start_state_collision]: Param 'move_group.max_sampling_attempts' was not set. Using default value: 100
[move_group-1] [INFO] [1684977328.427488021] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Add Time Optimal Parameterization'
[move_group-1] [INFO] [1684977328.427722684] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Workspace Bounds'
[move_group-1] [INFO] [1684977328.428004568] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State Bounds'
[move_group-1] [INFO] [1684977328.428323322] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State In Collision'
[... cut ...] mp/launch_params_qen8vmvh --params-file /tmp/launch_params_djskthb4 --params-file /tmp/launch_params_ajueolsa --params-file /tmp/launch_params_pwmx6wu6 --params-file /tmp/launch_params_1lo94ss0 --params-file /tmp/launch_params_n52lszzf -r vx300s/get_planning_scene:=/vx300s/get_planning_scene -r /arm_controller/follow_joint_trajectory:=/vx300s/arm_controller/follow_joint_trajectory -r /gripper_controller/follow_joint_trajectory:=/vx300s/gripper_controller/follow_joint_trajectory'].
[move_group-1]

AbishekNP, 2023-05-25:
No I haven't made any changes to the gazebo world. I get the same "waiting for controller" error even when I launch the xsarm_ros_control package with hardware type as gz_classic. But both packages work fine when the hardware type is actual. 

This is the output when I set the default verbose value as true and launched the Moveit package again. 

robotarm@gorgon:~/interbotix_ws$ ros2 launch interbotix_xsarm_moveit xsarm_moveit.launch.py robot_model:=vx300s hardware_type:=gz_classic
[INFO] [launch]: All log files can be found below /home/robotarm/.ros/log/2023-05-25-11-40-39-526027-gorgon-4719
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [4723]
[INFO] [rviz2-2]: process started with pid [4725]
[INFO] [gzserver-3]: process started with pid [4727]
[INFO] [gzclient-4]: process started with pid [4729]
[INFO] [spawn_entity.py-5]: process started with pid [4744]
[INFO] [robot_state_publisher-6]: process started with pid [4747]
[move_group-1] [WARN] [1685040041.070079097] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-1] [INFO] [1685040041.072874289] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0 seconds
[move_group-1] [INFO] [1685040041.072902641] [moveit_robot_model.robot_model]: Loading robot model 'vx300s'...
[move_group-1] [INFO] [1685040041.072909728] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[move_group-1] [INFO] [1685040041.227584120] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-1] [INFO] [1685040041.227705686] [moveit.ros_planning_interface.moveit_cpp]: Listening to '/vx300s/joint_states' for joint states
[move_group-1] [INFO] [1685040041.228343795] [moveit_ros.current_state_monitor]: Listening to joint states on topic '/vx300s/joint_states'
[move_group-1] [INFO] [1685040041.228757720] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[move_group-1] [INFO] [1685040041.228773261] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-1] [INFO] [1685040041.229093599] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[move_group-1] [INFO] [1685040041.229105799] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-1] [INFO] [1685040041.229444341] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-1] [INFO] [1685040041.229904973] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-1] [WARN] [1685040041.230262130] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[move_group-1] [ERROR] [1685040041.230279478] [moveit.ros.occupancy_map_monitor.middleware_handle]: No sensor plugin specified for octomap updater ; ignoring.
[move_group-1] [INFO] [1685040041.230290651] [moveit.ros.occupancy_map_monitor.middleware_handle]: Skipping octomap updater plugin ''
[move_group-1] [INFO] [1685040041.261252367] [moveit.ros_planning_interface.moveit_cpp]: Loading planning pipeline 'move_group'
[move_group-1] [INFO] [1685040041.274184999] [moveit.ros_planning.planning_pipeline]: Using planning interface 'OMPL'
[move_group-1] [INFO] [1685040041.276736124] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.path_tolerance' was not set. Using default value: 0.100000
[move_group-1] [INFO] [1685040041.276756759] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.resample_dt' was not set. Using default value: 0.100000
[move_group-1] [INFO] [1685040041.276761560] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.min_angle_change' was not set. Using default value: 0.001000
[move_group-1] [INFO] [1685040041.276774736] [moveit_ros.fix_workspace_bounds]: Param 'move_group.default_workspace_bounds' was not set. Using default value: 10.000000
[move_group-1] [INFO] [1685040041.276788471] [moveit_ros.fix_start_state_bounds]: Param 'move_group.start_state_max_bounds_error' was set to 0.100000
[move_group-1] [INFO] [1685040041.276793632] [moveit_ros.fix_start_state_bounds]: Param 'move_group.start_state_max_dt' was not set. Using default value: 0.500000
[move_group-1] [INFO] [1685040041.276803483] [moveit_ros.fix_start_state_collision]: Param 'move_group.start_state_max_dt' was not set. Using default value: 0.500000
[move_group-1] [INFO] [1685040041.276808075] [moveit_ros.fix_start_state_collision]: Param 'move_group.jiggle_fraction' was not set. Using default value: 0.020000
[move_group-1] [INFO] [1685040041.276812839] [moveit_ros.fix_start_state_collision]: Param 'move_group.max_sampling_attempts' was not set. Using default value: 100
[move_group-1] [INFO] [1685040041.276822885] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Add Time Optimal Parameterization'
[move_group-1] [INFO] [1685040041.276827358] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Workspace Bounds'
[... cut ...]
[move_group-1] [WARN] [1685040068.009112395] [moveit_ros.current_state_monitor]: No state update received within 100ms of system clock
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT) again, ignoring...
[move_group-1] [INFO] [1685040068.483424111] [rclcpp]: signal_handler(signum=2)
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT) again, ignoring...
[move_group-1] [INFO] [1685040068.947468583] [rclcpp]: signal_handler(signum=2)
[move_group-1] [WARN] [1685040069.010759514] [moveit_ros.current_state_monitor]: No state update received within 100ms of system clock

LSinterbotix, 2023-05-30:
> ```
> [ros2_control_node-1] what(): According to the loaded plugin descriptions the class gazebo_ros2_control/GazeboSystem with base class type hardware_interface::SystemInterface does not exist. Declared types are fake_components/GenericSystem interbotix_xs_ros_control/XSHardwareInterface mock_components/GenericSystem test_hardware_components/TestSystemCommandModes test_hardware_components/TestTwoJointSystem test_system
> ```

You may not have reinstalled the `gazebo_ros2_control` package. Please run the command below in the root of the Interbotix workspace to install all dependencies, including the gazebo_ros2_control package:

```
rosdep install --from-paths src --ignore-src -r -y
```

AbishekNP, 2023-05-31:
Still doesn't work...

LSinterbotix, 2023-06-02:
You might try to manually install it instead:

```
sudo apt-get install ros-humble-gazebo-ros2-control
```

Check that it is installed using dpkg:

```
dpkg --list | grep gazebo-ros2-control
```

Make sure to close and reopen your terminal or source the ROS distro setup file before running anything after a package installation.

Try to launch just the Gazebo sim package and set Gazebo's output to verbose:

```
ros2 launch interbotix_xsarm_sim xsarm_gz_classic.launch.py robot_model:=vx300s verbose:=true
```

AbishekNP, 2023-06-02:
Still the same error:
I installed ros-humblegazebo-ros2-control and sourced install/setup.bash as well.

Output 1:

robotarm@gorgon:~$ dpkg --list | grep gazebo-ros2-control
ii  ros-humble-gazebo-ros2-control                     0.4.3-1jammy.20230523.211003            amd64        gazebo_ros2_control

Output 2:

obotarm@gorgon:~/interbotix_ws$ ros2 launch interbotix_xsarm_sim xsarm_gz_classic.launch.py robot_model:=vx300s verbose:=true
[INFO] [launch]: All log files can be found below /home/robotarm/.ros/log/2023-06-02-12-29-28-693078-gorgon-767831
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [767841]
[INFO] [gzclient-2]: process started with pid [767843]
[INFO] [spawn_entity.py-3]: process started with pid [767845]
[INFO] [robot_state_publisher-4]: process started with pid [767847]
[INFO] [rviz2-5]: process started with pid [767857]
[gzclient-2] Gazebo multi-robot simulator, version 11.10.2
[gzclient-2] Copyright (C) 2012 Open Source Robotics Foundation.
[gzclient-2] Released under the Apache 2 License.
[gzclient-2] http://gazebosim.org
[gzclient-2] 
[gzserver-1] Gazebo multi-robot simulator, version 11.10.2
[gzserver-1] Copyright (C) 2012 Open Source Robotics Foundation.
[gzserver-1] Released under the Apache 2 License.
[gzserver-1] http://gazebosim.org
[gzserver-1] 
[gzserver-1] [INFO] [1685734171.767444060] [gazebo_ros2_control]: Loading gazebo_ros2_control plugin
[gzserver-1] [INFO] [1685734171.770377964] [vx300s.gazebo_ros2_control]: Starting gazebo_ros2_control plugin in namespace: /vx300s
[gzserver-1] [INFO] [1685734171.770632613] [vx300s.gazebo_ros2_control]: Starting gazebo_ros2_control plugin in ros 2 node: gazebo_ros2_control
[gzserver-1] [INFO] [1685734171.770679755] [vx300s.gazebo_ros2_control]: Loading parameter files /home/robotarm/interbotix_ws/install/interbotix_xsarm_sim/share/interbotix_xsarm_sim/config/trajectory_controllers/vx300s_trajectory_controllers.yaml
[gzserver-1] [INFO] [1685734171.773805869] [vx300s.gazebo_ros2_control]: connected to service!! robot_state_publisher
[gzserver-1] [INFO] [1685734171.774292133] [vx300s.gazebo_ros2_control]: Recieved urdf from param server, parsing...
[gzserver-1] [INFO] [1685734171.786753646] [vx300s.gazebo_ros2_control]: Loading joint: waist
[gzserver-1] [INFO] [1685734171.786814641] [vx300s.gazebo_ros2_control]: 	State:
[gzserver-1] [INFO] [1685734171.786824763] [vx300s.gazebo_ros2_control]: 		 position
[gzserver-1] [INFO] [1685734171.786833893] [vx300s.gazebo_ros2_control]: 	Command:
[gzserver-1] [INFO] [1685734171.786841170] [vx300s.gazebo_ros2_control]: 		 position
[gzserver-1] [INFO] [1685734171.786885857] [vx300s.gazebo_ros2_control]: Loading joint: shoulder
[gzserver-1] [INFO] [1685734171.786905087] [vx300s.gazebo_ros2_control]: 	State:
[gzserver-1] [INFO] [1685734171.786911905] [vx300s.gazebo_ros2_control]: 		 position
[gzserver-1] [INFO] [1685734171.786918917] [vx300s.gazebo_ros2_control]: 	Command:
[gzserver-1] [INFO] [1685734171.786925446] [vx300s.gazebo_ros2_control]: 		 position
[gzserver-1] [INFO] [1685734171.786941044] [vx300s.gazebo_ros2_control]: Loading joint: elbow
[gzserver-1] [INFO] [1685734171.786949536] [vx300s.gazebo_ros2_control]: 	State:
[gzserver-1] [INFO] [1685734171.786956313] [vx300s.gazebo_ros2_control]: 		 position
[gzserver-1] [INFO] [1685734171.786963124] [vx300s.gazebo_ros2_control]: 	Command:
[gzserver-1] [INFO] [1685734171.786969664] [vx300s.gazebo_ros2_control]: 		 position
[gzserver-1] [INFO] [1685734171.786984142] [vx300s.gazebo_ros2_control]: Loading joint: forearm_roll
[gzserver-1] [INFO] [1685734171.786992329] [vx300s.gazebo_ros2_control]: 	State:
[gzserver-1] [INFO] [1685734171.786998941] [vx300s.gazebo_ros2_control]: 		 position
[gzserver-1] [INFO] [1685734171.787005392] [vx300s.gazebo_ros2_control]: 	Command:
[gzserver-1] [INFO] [1685734171.787021662] [vx300s.gazebo_ros2_control]: 		 position
[gzserver-1] [INFO] [1685734171.787034988] [vx300s.gazebo_ros2_control]: Loading joint: wrist_angle
[gzserver-1] [INFO] [1685734171.787052748] [vx300s.gazebo_ros2_control]: 	State:
[gzserver-1] [INFO] [1685734171.787059279] [vx300s.gazebo_ros2_control]: 		 position
[gzserver-1] [INFO] [1685734171.787065979] [vx300s.gazebo_ros2_control]: 	Command:
[gzserver-1] [INFO] [1685734171.787072759] [vx300s.gazebo_ros2_control]: 		 position
[gzserver-1] [INFO] [1685734171.787086150] [vx300s.gazebo_ros2_control]: Loading joint: wrist_rotate
[gzserver-1] [INFO] [1685734171.787094104] [vx300s.gazebo_ros2_control]: 	State:
[gzserver-1] [INFO] [1685734171.787100555] [vx300s.gazebo_ros2_control]: 		 position
[gzserver-1] [INFO] [1685734171.787106950] [vx300s.gazebo_ros2_control]: 	Command:
[gzserver-1] [INFO] [1685734171.787113391] [vx300s.gazebo_ros2_control]: 		 position
[gzserver-1] [INFO] [1685734171.787126315] [vx300s.gazebo_ros2_control]: Loading joint: gripper
[gzserver-1] [INFO] [1685734171.787133902] [vx300s.gazebo_ros2_control]: 	State:
[gzserver-1] [INFO] [1685734171.787140470] [vx300s.gazebo_ros2_control]: 		 position
[gzserver-1] [INFO] [1685734171.787146966] [vx300s.gazebo_ros2_control]: 	Command:
[gzserver-1] [INFO] [1685734171.787153368] [vx300s.gazebo_ros2_control]: 		 position
[gzserver-1] [INFO] [1685734171.787162513] [vx300s.gazebo_ros2_control]: Loading joint: left_finger
[... 22 lines cut ...]
[... cut ...]
[ERROR] [gzclient-2]: process has died [pid 767843, exit code -2, cmd 'gzclient --verbose'].
[gzclient-2] 
[gzclient-2] 
[ERROR] [spawner-6]: process has died [pid 767982, exit code -2, cmd '/opt/ros/humble/lib/controller_manager/spawner -c controller_manager joint_state_broadcaster --ros-args -r __node:=joint_state_broadcaster_spawner -r __ns:=/vx300s --params-file /tmp/launch_params_i77mzq1m'].
[INFO] [robot_state_publisher-4]: process has finished cleanly [pid 767847]
[INFO] [rviz2-5]: process has finished cleanly [pid 767857]

LSinterbotix, 2023-06-13:
> ```
> [gzserver-1] [INFO] [1685734171.787663761] [vx300s.gazebo_ros2_control]: Loading controller_manager
> ...
> [ERROR] [gzserver-1]: process has died [pid 767841, exit code -11, cmd 'gzserver /home/robotarm/interbotix_ws/install/interbotix_common_sim/share/interbotix_common_sim/worlds/interbotix.world --verbose -slibgazebo_ros_init.so -slibgazebo_ros_factory.so -slibgazebo_ros_force_system.so'].
> ```

It might be failing when trying to load the controller manager. As I am unable to replicate this issue even on a fresh install, I would encourage you to further troubleshoot this issue yourself.

Are you able to run the command in the error?

```console
gzserver /home/robotarm/interbotix_ws/install/interbotix_common_sim/share/interbotix_common_sim/worlds/interbotix.world --verbose -slibgazebo_ros_init.so -slibgazebo_ros_factory.so -slibgazebo_ros_force_system.so
```

AbishekNP, 2023-06-13:
This is the output when I try to run that command:

robotarm@gorgon:~/interbotix_ws$ gzserver /home/robotarm/interbotix_ws/install/interbotix_common_sim/share/interbotix_common_sim/worlds/interbotix.world --verbose -slibgazebo_ros_init.so -slibgazebo_ros_factory.so -slibgazebo_ros_force_system.so
Gazebo multi-robot simulator, version 11.10.2
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 171.65.72.41
[Wrn] [ModelDatabase.cc:340] Getting models from[http://models.gazebosim.org/]. This may take a few seconds.
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://TrossenRoboticsOfficeBuilding]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://TrossenRoboticsOfficeBuilding]
Error Code 12 Msg: Unable to find uri[model://TrossenRoboticsOfficeBuilding]
[Msg] Loading world file [/home/robotarm/interbotix_ws/install/interbotix_common_sim/share/interbotix_common_sim/worlds/interbotix.world]

I've already installedL: ros-humble-ros2-control, ros-humble-ros2-controllers and ros-humble-gazebo-ros2-control packages. Is there any additional packages or source command required to load the controller manager?

RodrigoMA170393, 2023-10-13:
> > [gzclient-4] gzclient: /usr/include/boost/smart_ptr/shared_ptr.hpp:728: typename boost::detail::sp_member_access::type boost::shared_ptr::operator->() const [with T = gazebo::rendering::Camera; typename boost::detail::sp_member_access::type = gazebo::rendering::Camera*]: Assertion `px != 0' failed.
> 
> Looks like Gazebo is still crashing. This will prevent the gazebo_ros2_control plugin from loading, causing the `Waiting for '/vx300s/controller_manager' node to exist` issue.
> 
> Add the line below to your `~/.bashrc` file to prevent this from happening in the future, open a new terminal, and try again.
> 
> ```
> . /usr/share/gazebo/setup.sh
> ```

Thank you, it worked

closed 2024-05-31 by lukeschmitt-tr